CadQuery/cadquery-freecad-workbench #43: No module named QtNetwork

tobltobs, 2015-02-03:
I am using using FreeCAD 0.14, Libs: 0.14R3702 (Git) on Kubuntu 14. 
After extracting the current version of cadquery-freecad-module into freecads Mod directory I get an "No module named QtNetwork" error when I try to start the CadQuery Workbench in FreeCAD.

jmwright, 2015-02-03:
Thanks for all the version info. I was able to reproduce the error on a Kubuntu 14.04 64-bit VM. I'm a little surprised that nobody has mentioned this error before because FreeCAD from the stock Ubuntu repository and from the daily PPA give the same error. I'll dig into this more and let you know.

jmwright, 2015-02-03:
@tobltobs Please run the following line in a terminal. It should fix your issue. It did for me running on the Kubuntu VM.
```
sudo apt-get install python-pyside.qtnetwork
```

closed 2015-02-03 by jmwright

tobltobs, 2015-02-03:
Great. Problem solved.

jmwright, 2015-02-03:
Glad to hear it. Thanks for reporting this. I've added a [Troubleshooting section](https://github.com/jmwright/cadquery-freecad-module/wiki/User-Documentation#troubleshooting) to the wiki that covers this issue.

TT44bb, 2018-01-28:
I have the same issue, but on Debian testing (2018-01-28), Freecad 0.17 AppImage (2018-01-27), CadQuery 1.0.0 installed via Tools/Addon manager.

`python-pyside.qtnetwork` is installed and when I run a Python 2 shell I can `from PySide.QtNetwork import *` without error.

`CadQuery 1.0.0
CadQuery is a parametric scripting API for creating and traversing CAD models
Author: David Cowden
License: Apache-2.0
Website: https://github.com/dcowden/cadquery
Running the Python command 'CadQueryNewScript' failed:
Traceback (most recent call last):
  File "/home/t4b/.FreeCAD/Mod/cadquery_module/CQGui/Command.py", line 248, in Activated
    ImportCQ.open(os.path.join(templ_dir_path, 'script_template.py'))
  File "/home/t4b/.FreeCAD/Mod/cadquery_module/CQGui/ImportCQ.py", line 50, in open
    from pyqode.core.modes import FileWatcherMode
  File "/home/t4b/.FreeCAD/Mod/cadquery_module/Libs/pyqode/core/modes/__init__.py", line 6, in <module>
    from .autocomplete import AutoCompleteMode
  File "/home/t4b/.FreeCAD/Mod/cadquery_module/Libs/pyqode/core/modes/autocomplete.py", line 5, in <module>
    from pyqode.core.api import TextHelper
  File "/home/t4b/.FreeCAD/Mod/cadquery_module/Libs/pyqode/core/api/__init__.py", line 6, in <module>
    from .code_edit import CodeEdit
  File "/home/t4b/.FreeCAD/Mod/cadquery_module/Libs/pyqode/core/api/code_edit.py", line 18, in <module>
    from pyqode.core.managers import BackendManager
  File "/home/t4b/.FreeCAD/Mod/cadquery_module/Libs/pyqode/core/managers/__init__.py", line 15, in <module>
    from .backend import BackendManager
  File "/home/t4b/.FreeCAD/Mod/cadquery_module/Libs/pyqode/core/managers/backend.py", line 7, in <module>
    from pyqode.core.api.client import JsonTcpClient, BackendProcess
  File "/home/t4b/.FreeCAD/Mod/cadquery_module/Libs/pyqode/core/api/client.py", line 13, in <module>
    from pyqode.qt import QtCore, QtNetwork
  File "/home/t4b/.FreeCAD/Mod/cadquery_module/Libs/pyqode/qt/QtNetwork.py", line 15, in <module>
    from PySide.QtNetwork import *

No module named QtNetwork`

Any idea what I could do to find the problem?

jmwright, 2018-01-28:
@TT44bb Can you do the import from the Python console inside FreeCAD to see if it works?

TT44bb, 2018-01-29:
@jmwright It doesn't work from there either. I also tried FreeCAD-0.16.6712.glibc2.17-x86_64.AppImage, can't find that module either.

FreeCAD from Debian repos is currently somewhat broken (which is why I was using AppImages in the first place), but I managed to get a version of 0.16 from the stable repos running and within that FreeCAD the manual import works and consequently the whole of CadQuery does.

So maybe some problem with the AppImages? Are those supposed to work/tested?

jmwright, 2018-01-29:
@TT44bb I've never tried to run CadQuery inside a FreeCAD AppImage (didn't know that was an option). There might be something weird about how it handles says.path or something else similar which prevents it from finding QtNetwork.

TT44bb, 2018-01-29:
@jmwright 
In 0.17 AppImage:
```
>>> from sys import path
>>> for p in path:
... 	print p
... 
/tmp/.mount_FreeCAyFol3G/usr/lib/freecad-daily/Mod/Drawing
/tmp/.mount_FreeCAyFol3G/usr/lib/freecad-daily/Mod/Inspection
/tmp/.mount_FreeCAyFol3G/usr/lib/freecad-daily/Mod/Idf
/tmp/.mount_FreeCAyFol3G/usr/lib/freecad-daily/Mod/AddonManager
/tmp/.mount_FreeCAyFol3G/usr/lib/freecad-daily/Mod/Robot
/tmp/.mount_FreeCAyFol3G/usr/lib/freecad-daily/Mod/Arch
/tmp/.mount_FreeCAyFol3G/usr/lib/freecad-daily/Mod/Part
/tmp/.mount_FreeCAyFol3G/usr/lib/freecad-daily/Mod/Path
/home/t4b/.FreeCAD/Mod/cadquery-freecad-module/Libs
/home/t4b/.FreeCAD/Mod/cadquery-freecad-module/Libs/cadquery
/home/t4b/.FreeCAD/Mod/cadquery-freecad-module
/tmp/.mount_FreeCAyFol3G/usr/lib/freecad-daily/Mod/Mesh
/tmp/.mount_FreeCAyFol3G/usr/lib/freecad-daily/Mod/Raytracing
/tmp/.mount_FreeCAyFol3G/usr/lib/freecad-daily/Mod/Material
/tmp/.mount_FreeCAyFol3G/usr/lib/freecad-daily/Mod/Web
/tmp/.mount_FreeCAyFol3G/usr/lib/freecad-daily/Mod/Complete
/tmp/.mount_FreeCAyFol3G/usr/lib/freecad-daily/Mod/Tux
/tmp/.mount_FreeCAyFol3G/usr/lib/freecad-daily/Mod/TechDraw
/tmp/.mount_FreeCAyFol3G/usr/lib/freecad-daily/Mod/Import
/tmp/.mount_FreeCAyFol3G/usr/lib/freecad-daily/Mod/Test
/tmp/.mount_FreeCAyFol3G/usr/lib/freecad-daily/Mod/Draft
/tmp/.mount_FreeCAyFol3G/usr/lib/freecad-daily/Mod/Points
/tmp/.mount_FreeCAyFol3G/usr/lib/freecad-daily/Mod/Start
/tmp/.mount_FreeCAyFol3G/usr/lib/freecad-daily/Mod/Spreadsheet
/tmp/.mount_FreeCAyFol3G/usr/lib/freecad-daily/Mod/Fem
/tmp/.mount_FreeCAyFol3G/usr/lib/freecad-daily/Mod/OpenSCAD
/tmp/.mount_FreeCAyFol3G/usr/lib/freecad-daily/Mod/MeshPart
/tmp/.mount_FreeCAyFol3G/usr/lib/freecad-daily/Mod/Plot
/tmp/.mount_FreeCAyFol3G/usr/lib/freecad-daily/Mod/Ship
/tmp/.mount_FreeCAyFol3G/usr/lib/freecad-daily/Mod/Measure
/tmp/.mount_FreeCAyFol3G/usr/lib/freecad-daily/Mod/PartDesign
/tmp/.mount_FreeCAyFol3G/usr/lib/freecad-daily/Mod/Surface
/tmp/.mount_FreeCAyFol3G/usr/lib/freecad-daily/Mod/Image
/tmp/.mount_FreeCAyFol3G/usr/lib/freecad-daily/Mod/Show
/tmp/.mount_FreeCAyFol3G/usr/lib/freecad-daily/Mod/ReverseEngineering
/tmp/.mount_FreeCAyFol3G/usr/lib/freecad-daily/Mod/Sketcher
/tmp/.mount_FreeCAyFol3G/usr/lib/freecad-daily/Mod
/tmp/.mount_FreeCAyFol3G/usr/lib/freecad-daily/lib64
/tmp/.mount_FreeCAyFol3G/usr/lib/freecad-daily/lib
/tmp/.mount_FreeCAyFol3G/usr/lib/freecad-daily/Ext

/tmp/.mount_FreeCAyFol3G/usr
/tmp/.mount_FreeCAyFol3G/usr/lib/python2.7
/tmp/.mount_FreeCAyFol3G/usr/lib/python2.7/plat-x86_64-linux-gnu
/tmp/.mount_FreeCAyFol3G/usr/lib/python2.7/lib-tk
/tmp/.mount_FreeCAyFol3G/usr/lib/python2.7/lib-old
/tmp/.mount_FreeCAyFol3G/usr/lib/python2.7/lib-dynload
/tmp/.mount_FreeCAyFol3G/usr/lib/python2.7/dist-packages
/home/t4b/.FreeCAD/Macro/
/tmp/.mount_FreeCAyFol3G/usr/lib/freecad-daily/Macro
```

So this seems not to include anything installed on my system, only things from the AppImage and from FreeCad config.

I don't really know what sys.path is, but I assume it's like Bash $PATH, then I guess this is the reason it won't work.

jmwright, 2018-01-29:
You can actually manipulate the path where Python will look for libs using the sys module. It's what I do so that FreeCAD can find the embedded libs in the CadQuery workbench. I'm not sure if an AppImage will let you access the filesystem outside of it, but if it does, you can do something like this in FreeCAD's Python console.
```
import sys
qtLibPath = '/path/to/qtnetwork/lib'
sys.path.insert(0, qtLibPath)
import QtNetwork
```

TT44bb, 2018-01-29:
Basically what you suggested works, but I needed to reload PySide from the new path.

```
import sys
# add is sys.path from my native Python install
add=['', '/usr/lib/python2.7', '/usr/lib/python2.7/plat-x86_64-linux-gnu', '/usr/lib/python2.7/lib-tk', '/usr/lib/python2.7/lib-old', '/usr/lib/python2.7/lib-dynload', '/usr/local/lib/python2.7/dist-packages', '/usr/lib/python2.7/dist-packages', '/usr/lib/python2.7/dist-packages/PILcompat', '/usr/lib/python2.7/dist-packages/gtk-2.0']
sys.path=add+sys.path
PySide=reload(PySide)
from PySide.QtNetwork import *
```

Thank you.

jmwright, 2018-01-29:
@TT44bb Is that a true fix for you, or just an ugly workaround?

I'm a little bit hesitant to put that path manipulation into the workbench without some careful planning because I have to make sure that it works across platforms, and installation locations differ even between Linux distros.

TT44bb, 2018-01-29:
@jmwright 
Well it's good enough for me, but it's a workaround, yeah. I'll switch back to native package management as soon as the packaging problems with FreeCad are solved in Debian (some dependency issue related to phasing out QT4).

I don't know how it could be fixed properly. It seems to me that AppImages just aren't suited well to programs which have addons which can have additional dependencies. Maybe petition FreeCad to incude QtNetwork in the AppImage.

TT44bb, 2018-04-01:
I built an AppImage compatible with cadquery and thought I'd share my process.

Prerequisites: git, docker
```
git clone https://www.github.com/AppImage/AppImages.git
cd AppImages
cp recipes/FreeCAD-nightly.yml recipes/FreeCADquery-nightly.yml
# add python-pyside.qtnetwork to the package section of this file
./pkg2appimage-with-docker -c recipes/FreeCADquery-nightly.yml
```
The resulting appimage is in the out directory and can install cadquery using the plugin manager in FreeCAD.

jmwright, 2018-04-02:
@TT44bb Awesome, thanks for sharing. Tagging @dcowden and @adam-urbanczyk to make sure they see your process.