CadQuery/cadquery-freecad-workbench #45: Error when launch CADQuery Workbench

wood-galaxy, 2015-03-08:
When i launch the workbench i get this window message error :
/usr/lib/freecad/lib/Drawing.so: undefined symbol: _ZNK3App17PropertyContainer7getNameEPKNS_8PropertyE

And in the report view I get this interesting information :

/usr/lib/freecad/lib/Drawing.so: undefined symbol: _ZNK3App17PropertyContainer7getNameEPKNS_8PropertyE
Traceback (most recent call last):
  File "<string>", line 66, in Activated
  File "/home/jo/.FreeCAD/Mod/CadQuery/Libs/cadquery/__init__.py", line 4, in <module>
    from .freecad_impl import exporters
  File "/home/jo/.FreeCAD/Mod/CadQuery/Libs/cadquery/freecad_impl/exporters.py", line 29, in <module>
    import Drawing

It seem that manually import Drawing workbench by typing "import Drawing" in the python console before launch CadQuery workbench make the workbench working good (at least it is launch without error)

OS: Ubuntu 14.04.2 LTS
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.15.4639 (Git)
Branch: master
Hash: f3922e54117a0ea663018fa7ebd2054380325069
Python version: 2.7.6
Qt version: 4.8.6
Coin version: 4.0.0a
OCC version: 6.8.0.oce-0.17-dev

jmwright, 2015-03-08:
Thanks for reporting this @wood-galaxy 

I run a 64-bit Ubuntu 14.04 setup too and have not seen this error. I'll have to dig into it some more.

wood-galaxy, 2015-03-08:
Ok, I have just discover this workbench after a french user come to the freecadweb forum and say he have problem with this workbench. At the moment I don't know if his problem is related to this issue.

jmwright, 2015-03-08:
Where is FreeCAD installed on this Ubuntu 14.04 system?

wood-galaxy, 2015-03-08:
Ha, indeed this freecad is not "installed" it is just compiled from sources in a folder of my home folder.
Indeed I have not see that the path in the error message point on the traditional installation location.

jmwright, 2015-03-08:
Is the easiest way for you to fix this issue right now with symlinks then?

jmwright, 2015-03-10:
A long-term fix for this would be to add an array of common FreeCAD paths for each major OS to the config file. Then users would be able to add non-standard locations easily.

wood-galaxy, 2015-03-18:
About the symlinks, I don't try. Because I still want to have both current freecad version and stable version. My knowledge about library usage are too small to understand what i need to do.
About FreeCAD common paths, it's perhaps the way to go but there is a lot of possibility (i think).

jmwright, 2015-04-02:
@wood-galaxy I've created a feature request to make it easier for users like yourself to enter new FreeCAD installation paths. I'm going to close this issue because of that.

closed 2015-04-02 by jmwright