CadQuery/cadquery-freecad-workbench #46: CadQuery workbench broke the gui config

wood-galaxy, 2015-03-08:
Hi,
Launching CadQuery workbench "break" the current FreeCAD gui configuration of the user. I found it too much invasive...
For example I have normally Combo View, Python console, Report View and Selection View enable. After launching the CadQuery WB all this views are deactivate (except Report View) and the CadQuery Code view is still enable after switching to other workbench.

jmwright, 2015-03-08:
This is something that I actually tried to fix when I first wrote the CQ workbench. I kept track of what was open when the CQ workbench was first selected, and then tried to reshow those things when another workbench was selected again. However, I ran into a problem that I was never able to get an answer for on the FreeCAD forums. I do agree though that it's much too intrusive the way it is now.

I'm going to go ahead and mark this as a bug.

jmwright, 2015-05-12:
I've been thinking about this, and I believe the best way to address this is to put the CadQuery script editor in the same location as the current built-in macro editor. It would be a document in a tab, and the module could tile the script document and the 3D view document to get the same side-by-side look. Then the module wouldn't break the previous dock widget layout, and it also wouldn't interfere with the normal FreeCAD workflow. Another advantage is that it would give users flexibility in how they use the module. They could be working in multiple script windows and multiple 3D views at the same time. I've created a mock-up of what this would look like using the FreeCAD macro editor and the Tile documents menu option.
[image]
I'm working on the CadQuery library itself right now, and it may be awhile before I get back to this. However, I thought I'd try to get some feedback on this idea now.

jmwright, 2015-06-10:
There's a bug in versions of FreeCAD prior to 0.16 which makes it crash with a segfault when I implement this new layout. I think that rather than have the layout be inconsistent between versions, I'll stick with the current layout. However, I have made some changes that are coming in the next release that won't break the user's FreeCAD GUI setup when opening the CadQuery workbench.

jmwright, 2015-06-11:
Addressed in commit 73f644e091822ab483f545be4c813b661b67f8af

closed 2015-06-11 by jmwright