CadQuery/cadquery-freecad-workbench #47: Failed to load CadQuery WB on Windows7 and FC 0.14.37000

wood-galaxy, 2015-03-13:
Hi,
A french user of FreeCAD have a problem on loading CadQuery workbench. His FreeCAD data are :
OS: Windows 7
Word size: 32-bit
Version: 0.14.3700 (Git)
Branch: releases/FreeCAD-0-14
Hash: 32f5aae0a64333ec8d5d160dbc46e690510c8fe1
Python version: 2.6.2
Qt version: 4.5.2
Coin version: 3.1.0
SoQt version: 1.4.1
OCC version: 6.5.1

He got the message :
"atelier défaillant"
('invalid syntax', ('C:\\Program Files (x86)\\FreeCAD 0.14\\Mod\\CadQuery\\Libs\\pyqode\\core\\modes\\autocomplete.py', 23, 48, " SELECTED_QUOTES_FORMATS = {key: '%s%s%s' for key in MAPPING.keys()}\n"))

And the report view say :
('invalid syntax', ('C:\\Program Files (x86)\\FreeCAD 0.14\\Mod\\CadQuery\\Libs\\pyqode\\core\\modes\\autocomplete.py', 23, 48, " SELECTED_QUOTES_FORMATS = {key: '%s%s%s' for key in MAPPING.keys()}\n"))
Running the Python command 'CadQueryNewScript' failed:
Traceback (most recent call last):
File "C:\Program Files (x86)\FreeCAD 0.14\Mod\CadQuery\Gui\Command.py", line 201, in Activated
CadQueryCloseScript().Activated()
File "C:\Program Files (x86)\FreeCAD 0.14\Mod\CadQuery\Gui\Command.py", line 66, in Activated
if len(cqCodePane.file.path) == 0:

'NoneType' object has no attribute 'file'Running the Python command 'CadQueryExecuteScript' failed:
Traceback (most recent call last):
File "C:\Program Files (x86)\FreeCAD 0.14\Mod\CadQuery\Gui\Command.py", line 164, in Activated
tempFile.write(cqCodePane.toPlainText())

'NoneType' object has no attribute 'toPlainText'Ex001_Simple_Block.py
Running the Python command '%m/%d/%Y %H:%M:%S' failed:
Traceback (most recent call last):
File "C:\Program Files (x86)\FreeCAD 0.14\Mod\CadQuery\Gui\Command.py", line 123, in Activated
CadQueryCloseScript().Activated()
File "C:\Program Files (x86)\FreeCAD 0.14\Mod\CadQuery\Gui\Command.py", line 66, in Activated
if len(cqCodePane.file.path) == 0:

'NoneType' object has no attribute 'file'Ex001_Simple_Block.py
Running the Python command '%m/%d/%Y %H:%M:%S' failed:
Traceback (most recent call last):
File "C:\Program Files (x86)\FreeCAD 0.14\Mod\CadQuery\Gui\Command.py", line 123, in Activated
CadQueryCloseScript().Activated()
File "C:\Program Files (x86)\FreeCAD 0.14\Mod\CadQuery\Gui\Command.py", line 66, in Activated
if len(cqCodePane.file.path) == 0:

'NoneType' object has no attribute 'file'Running the Python command 'CadQueryNewScript' failed:
Traceback (most recent call last):
File "C:\Program Files (x86)\FreeCAD 0.14\Mod\CadQuery\Gui\Command.py", line 201, in Activated
CadQueryCloseScript().Activated()
File "C:\Program Files (x86)\FreeCAD 0.14\Mod\CadQuery\Gui\Command.py", line 66, in Activated
if len(cqCodePane.file.path) == 0:

'NoneType' object has no attribute 'file'Ex001_Simple_Block.py
Running the Python command '%m/%d/%Y %H:%M:%S' failed:
Traceback (most recent call last):
File "C:\Program Files (x86)\FreeCAD 0.14\Mod\CadQuery\Gui\Command.py", line 123, in Activated
CadQueryCloseScript().Activated()
File "C:\Program Files (x86)\FreeCAD 0.14\Mod\CadQuery\Gui\Command.py", line 66, in Activated
if len(cqCodePane.file.path) == 0:

'NoneType' object has no attribute 'file'Running the Python command 'CadQueryNewScript' failed:
Traceback (most recent call last):
File "C:\Program Files (x86)\FreeCAD 0.14\Mod\CadQuery\Gui\Command.py", line 201, in Activated
CadQueryCloseScript().Activated()
File "C:\Program Files (x86)\FreeCAD 0.14\Mod\CadQuery\Gui\Command.py", line 66, in Activated
if len(cqCodePane.file.path) == 0:

'NoneType' object has no attribute 'file'Ex002_Block_With_Bored_Center_Hole.py
Running the Python command 'Return the per-file header as a string.' failed:
Traceback (most recent call last):
File "C:\Program Files (x86)\FreeCAD 0.14\Mod\CadQuery\Gui\Command.py", line 123, in Activated
CadQueryCloseScript().Activated()
File "C:\Program Files (x86)\FreeCAD 0.14\Mod\CadQuery\Gui\Command.py", line 66, in Activated
if len(cqCodePane.file.path) == 0:

'NoneType' object has no attribute 'file'Running the Python command 'CadQueryNewScript' failed:
Traceback (most recent call last):
File "C:\Program Files (x86)\FreeCAD 0.14\Mod\CadQuery\Gui\Command.py", line 201, in Activated
CadQueryCloseScript().Activated()
File "C:\Program Files (x86)\FreeCAD 0.14\Mod\CadQuery\Gui\Command.py", line 66, in Activated
if len(cqCodePane.file.path) == 0:

'NoneType' object has no attribute 'file'Running the Python command 'CadQueryOpenScript' failed:
Traceback (most recent call last):
File "C:\Program Files (x86)\FreeCAD 0.14\Mod\CadQuery\Gui\Command.py", line 241, in Activated
CadQueryCloseScript().Activated()
File "C:\Program Files (x86)\FreeCAD 0.14\Mod\CadQuery\Gui\Command.py", line 66, in Activated
if len(cqCodePane.file.path) == 0:

'NoneType' object has no attribute 'file'

Is it a bug ? or what can I advise him to fix this ?

jmwright, 2015-03-13:
I asked about this in a PyCode issue and they said the problem is Python 2.6. There are no plans to support Python 2.6 in PyQode. FreeCAD 0.14.3700 works fine for me in Windows 8 64-bit with Python 2.7.9 installed. I was thinking the Windows version of FreeCAD uses the embedded 2.6 interpreter always, but maybe not. Installing 2.7.9 may be worth a try. I would be sure to let the installer make all the registry and environment variable changes.

jmwright, 2015-03-17:
@wood-galaxy Did the above information help you at all?

wood-galaxy, 2015-03-18:
Hi @jmwright ! Sorry for the radio silence. Yes it help me and the user i help.
He first update his python version, then, still having trouble, he get the development snapshot of FreeCAD available on Sourceforge. Now it seem that it's running fine. thank you.

jmwright, 2015-04-02:
@wood-galaxy I am unable to reproduce this error on Windows 7 64-bit with FreeCAD 0.14.3700 64-bit installed using the installer downloaded from freecadweb.org. Since the user you're helping is up and running, I'm going to close this issue for now.

closed 2015-04-02 by jmwright