CadQuery/cadquery-freecad-workbench #50: Add Recent Scripts Submenu in CadQuery Menu

jmwright, 2015-04-04:
It would be nice to easily access the scripts that you've worked with recently without having to navigate back to them through the Open File dialog.

easyw, 2015-08-09:
Yes ... I see you've already seen the script...
we are using cadquery to generate a parametric lib of ICs models for kicad...
may be you could also consider to merge hyOzd fork that seems to have some enhancement...
https://github.com/hyOzd/cadquery
thank you
Maurice

jmwright, 2018-12-24:
The menu structure is determined when the workbench starts up, which prevents this feature from being terribly useful. The menu needs to be dynamically populated as files are opened without a FreeCAD/workbench restart. I'm going to close this, and someone can re-open it later if they have a good idea on how to make the menu dynamic without some really messy dynamic Qt code.

closed 2018-12-24 by jmwright