CadQuery/cadquery-freecad-workbench #51: polyline.mirrorX doesn't work as expected

hyOzd, 2015-04-07:
I was trying to draw a t-slot profile by drawing only a quarter then mirroring. But mirrorX doesn't seem to work. Please try below snippet:

```python
from cadquery import Workplane
from Helpers import show

r = 20
d1 = 5
d2 = 8
s = 7
t = 1.5
H = 10

points = [
    (0, t/2),
    (r/2-1.5*t, r/2-t),
    (s/2, r/2-t),
    (s/2, r/2),
    (r/2, r/2),
    (r/2, s/2),
    (r/2-t, s/2),
    (r/2-t, r/2-1.5*t),
    (t/2, 0)
    ]

result = Workplane("front").polyline(points).mirrorX()

show(result)
```

It behaves like this in my case.

[image]

But mirrorY works as I expect:

[image]

jmwright, 2015-04-07:
I get the same behavior with your code. I'm not sure yet why this is happening.

It's interesting that if I execute the following, mirrorX and mirrorY both work as expected.
```python
from cadquery import Workplane
from Helpers import show

points = [(0.0, 0.0), (2.5, 0.0), (2.5, 5.0), (5.0, 5.0), (5.0, 7.5), (0.0, 7.5)]

result = Workplane("XY").polyline(points).mirrorY()

show(result)
```
But, if I break the symmetry about the axes at all, I can break one or the other. The following change of the X in the first points tuple breaks mirrorY.
```python
from cadquery import Workplane
from Helpers import show

points = [(1.75, 0.0), (2.5, 0.0), (2.5, 5.0), (5.0, 5.0), (5.0, 7.5), (0.0, 7.5)]

result = Workplane("XY").polyline(points).mirrorY()

show(result)
```
[image]

I suspect this is an issue with the underlying CadQuery library or FreeCAD itself. I'll cross-post this as a bug on the CQ project's issue tracker.

jmwright, 2015-04-10:
@hyOzd I did some experimenting tonight, and it looks (on the surface) like this is a FreeCAD problem. The following FreeCAD script yields the same results as CadQuery does.
```python
import Part, FreeCAD, math
from FreeCAD import Base

r = 20
d1 = 5
d2 = 8
s = 7
t = 1.5
H = 10

points = [
    Base.Vector(0, t/2),
    Base.Vector(r/2-1.5*t, r/2-t),
    Base.Vector(s/2, r/2-t),
    Base.Vector(s/2, r/2),
    Base.Vector(r/2, r/2),
    Base.Vector(r/2, s/2),
    Base.Vector(r/2-t, s/2),
    Base.Vector(r/2-t, r/2-1.5*t),
    Base.Vector(t/2, 0)
    ]

aWire = Part.makePolygon(points)

aTrsf=Base.Matrix()
aTrsf.rotateX(math.pi) # rotate around the x-axis

aMirroredWire = aWire.transformGeometry(aTrsf)

myWireProfile = Part.Wire([aWire, aMirroredWire])

#Part.show(aWire)
#Part.show(aMirroredWire)
Part.show(myWireProfile)
```
I'm going to post to the FreeCAD forums to see if we're just handling the wires wrong or something.

jmwright, 2015-04-10:
FreeCAD forum topic [here](http://forum.freecadweb.org/viewtopic.php?f=10&t=10474#wrap).

hyOzd, 2015-04-10:
Yay. I figured it out. It's the order of points (vertexes).

This works:

```
import Part, FreeCAD, math
from FreeCAD import Base

r = 20
d1 = 5
d2 = 8
s = 7
t = 1.5
H = 10

points = [
    Base.Vector(0, t/2),
    Base.Vector(r/2-1.5*t, r/2-t),
    Base.Vector(s/2, r/2-t),
    Base.Vector(s/2, r/2),
    Base.Vector(r/2, r/2),
    Base.Vector(r/2, s/2),
    Base.Vector(r/2-t, s/2),
    Base.Vector(r/2-t, r/2-1.5*t),
    Base.Vector(t/2, 0)
    ]

mirrored_points = [(p[0], -p[1], p[2]) for p in points]
mirrored_points.reverse() # !! same issue will happen without this line

aWire = Part.makePolygon(points)
aMirroredWire = Part.makePolygon(mirrored_points)

myWireProfile = Part.Wire([aWire, aMirroredWire])

#Part.show(aWire)
#Part.show(aMirroredWire)
Part.show(myWireProfile)
```

jmwright, 2015-04-10:
Interesting. Thanks for posting this find to the FrreeCAD forum topic too.

jmwright, 2015-04-10:
Got a couple of responses on the forum about how to address the root of the problem.

http://forum.freecadweb.org/viewtopic.php?f=10&t=10474&p=84383#p84383

It may be next week, but I'll look into using that info to make CadQuery's mirror functions more robust.

dcowden, 2015-04-10:
thats some great sleuthing!

jmwright, 2015-04-13:
Ok, so the following changes, based on the response from wmayer in the FreeCAD forum topic, fix the problem.
```python
import Part, FreeCAD, math
from FreeCAD import Base

r = 20
d1 = 5
d2 = 8
s = 7
t = 1.5
H = 10

points = [
    Base.Vector(0, t/2),
    Base.Vector(r/2-1.5*t, r/2-t),
    Base.Vector(s/2, r/2-t),
    Base.Vector(s/2, r/2),
    Base.Vector(r/2, r/2),
    Base.Vector(r/2, s/2),
    Base.Vector(r/2-t, s/2),
    Base.Vector(r/2-t, r/2-1.5*t),
    Base.Vector(t/2, 0)
    ]

aWire = Part.makePolygon(points)

aTrsf=Base.Matrix()
aTrsf.rotateX(math.pi) # rotate around the x-axis

aMirroredWire = aWire.transformGeometry(aTrsf)

aEdges = aWire.Edges
aEdges.extend(aMirroredWire.Edges)
comp = Part.Compound(aEdges)
wire = comp.connectEdgesToWires(False).Wires[0]

myWireProfile = Part.Wire([aWire, wire])

Part.show(myWireProfile)
```
[image]

Now, I need to dig into the FreeCAD documentation a little bit to understand what exactly those functions do. After that, I should be able to integrate that fix into CadQuery.

jmwright, 2015-04-13:
I've almost got this fix working on a branch. However, it breaks the OCC bottle example even though the mirrored outline looks the same before extrusion. The only difference is the line weight that's used when the outlines are displayed in a document. I'm going to check into whether or not this means something.

jmwright, 2015-04-14:
I'm having trouble with the last 10% of this fix. I certainly think it's worth the time to get it right though, especially since I believe this will also allow us to chain the mirrorX and mirrorY functions. That's something that's been broken for a long time.

I'll keep hacking on it, but I'll be lucky to find the time to have a fix committed this week.

jmwright, 2015-04-17:
It looks like I have this fixed, but I need to check a few more things and add the appropriate tests. One side benefit of this is that you can now chain mirrorX().mirrorY() and it seems to work correctly.

My work is on the [mirrorfix branch](https://github.com/dcowden/cadquery/tree/mirrorfix) of CadQuery if anyone's interested.

dcowden, 2015-04-17:
sweet! you are a machine!

jmwright, 2015-04-23:
@hyOzd - The fix for the issue you were seeing above is in the latest commit of this module: d8eb92cfc6af4029fc013b4839896734ad780478

The issue of chaining mirrorX and mirrorY is still not resolved though. It's a different problem due to the underlying FreeCAD library or something strange in our interaction with it, so I'm going to continue to work on that part. The problem that caused you to file this issue has been fixed though.

dcowden, 2015-04-23:
man, Jeremy you are a machine! Great work!

jmwright, 2015-05-19:
I'm closing this since the bug with chaining mirrorX and mirrorY is recorded in the main CadQuery issue tracker.

closed 2015-05-19 by jmwright