CadQuery/cadquery-freecad-workbench #53: Change of workingplace failed

moeb, 2015-05-08:
Hi, i just installed freecad and the cadquery-freecad-module.
Unfortunately it errors when i change the workbench to CadQuery. In fact it errors on all changes of the workbench, if CadQuery is in the Mod directory.

I attached a screenshot:
[image]

My system is Archlinux.
Freecad version is 0.15 revision 4671, installed from the official archlinux repositories.

BTW. i wrote a PKGBUILD for installing/uninstalling this module with the packagemanager.
When this works and you are positive, i would like to push it to AUR (Archlinux user repository)

Best, bmoe

jmwright, 2015-05-08:
Hi @moeb What version of Python is FreeCAD using on that system?

> BTW. i wrote a PKGBUILD for installing/uninstalling this module with the packagemanager.

Very cool, thanks for doing that. You're the first Arch user that's reported in. I'll set up an Arch VM as soon as I get a chance to see if I can diagnose this.

moeb, 2015-05-08:
Hi @jmwright, it uses /usr/bin/python2 wich is python2.7

I wrote that the error occurs on all workbench changes, but i just found out, it only occurs after i changed one time to CadQuery. After that, it happens on all changes.

I don't know how to attach the PKGBUILD here, it's pretty simple and short though, so i just post it :)

```
pkgname=cadquery-freecad-module
pkgver=0.1.8
pkgrel=2
arch=('i686' 'x86_64')
pkgdesc="cadquery module for freecad"
url="https://github.com/jmwright/cadquery-freecad-module"
provides=('cadquery-freecad-module')
depends=('freecad')
license=('LGPL3')
source=('https://github.com/jmwright/cadquery-freecad-module/archive/v0.1.8.tar.gz')
md5sums=('422a73c39d01694ac66bf44b38469018')
_moddir=usr/share/freecad/Mod/

package() {
    cd $srcdir/${pkgname}-${pkgver}
    mkdir -p $pkgdir/$_moddir
    cp -r CadQuery $pkgdir/$_moddir
}
```
If you don't have no experience with archlinux, and got some problems, feel free to pm me :)

jmwright, 2015-05-08:
> If you don't have no experience with archlinux, and got some problems, feel free to pm me :)

Thanks for the offer. I really like Arch, but it's been a long time since I've run it. Is there a quick way for me to get up and running with Arch quickly to troubleshoot this, or should I work through the ground-up setup process on the Arch Wiki?

One other thing to try while I'm getting set up is to open the FreeCAD Python console and try to import cadquery to see if it gives any errors.
```python
import sys
sys.path.insert(0, '/path/to/freecad/install/Mod/CadQuery/Libs')
import cadquery
```

moeb, 2015-05-09:
It should be all the same, until you reboot into the system. I don't think, there is something like a downloadable clean snapshot for arch; at least i don't know of one.

>One other thing to try while I'm getting set up is to open the FreeCAD Python console and try to import cadquery to see if it gives any errors.

It's no Problem to import CadQuery

jmwright, 2015-05-10:
I'm starting to set up an Arch 64-bit system. Are you running 64 or 32 bit?

moeb, 2015-05-10:
Thats great !
I'm running 64 bit

jmwright, 2015-05-11:
What window manager/desktop environment(s) are you running FreeCAD in?

moeb, 2015-05-11:
Atm i'm running gnome 3
Do you think that makes a difference ? O.o

jmwright, 2015-05-11:
I doubt that it does, but I want to make sure.

jmwright, 2015-05-11:
I set up Arch and installed FreeCAD from the official repos. I installed the released v0.1.8 CadQuery module, and I cannot reproduce your error. It runs without errors and displays the first example. I extracted the .tar.gz file and manually copied the `CadQuery` directory into `/usr/share/freecad/Mod`. Is that what you did?

moeb, 2015-05-11:
I did it again by hand just now, with the same result as with the package. After that i reinstalled freecad with the same results. So what is the difference ?? O.o

moeb, 2015-05-11:
I got a little success, but i'm a little confused nevertheless.
The error occures in /usr/share/freecad/Mods/Libs/pygments/plugin.py
When i replace this:
```
try:
    import pkg_resources
except ImportError:
    pkg_resources = None
```

with this:
```
try:
    import pkg_resources
except:
    pkg_resources=None
```

I get the first example running :)
I'm a little irritated, that the other error that occurs is not printed though.

jmwright, 2015-05-11:
Thanks for the troubleshooting work. That gives us a starting point.

PyQode and Pygments have trouble with Python 2.6, I think. We should probably start out by verifying that FreeCAD is not using 2.6 behind the scenes. You can verify this by entering the following in FreeCAD's Python console.
```python
import sys
sys.version_info
```
Should output something like...
```python
sys.version_info(major=2, minor=7, micro=6, releaselevel='final', serial=0)
```
Major needs to be 2, and minor needs to be 7. If it's running 2.7, I think we'll need to get the PyQode developers involved.

moeb, 2015-05-11:
Its 2.7.9 definitly
i caught the exceptin like this
```
except Exception as exception:
    print(exception.__class__.__name__)
```

and it is a TypeError.

>I think we'll need to get the PyQode developers involved.

I came to PyQode first, because it imports Pygments somwhere internaly, so i'm not sure, if it has anything to do with the error.

I'll try to look into pkg_resources, and perhaps read about it a little
https://pythonhosted.org/setuptools/pkg_resources.html

My normal python2.7 version seems to be the same as the one, that freecad uses.
I compared the sys.path of both, and the sys.path of the `cmdline` version is contained within the freecad version.

I was curios, wether there might be two version of pkg_resources, but locate tells me otherwise.
Tomorrow i will narrow down the exception in pkg_resources.

jmwright, 2015-05-11:
@ColinDuquesnoy has a suggestion of something to look at [here](https://github.com/pyQode/pyQode/issues/40#issuecomment-101026432).

moeb, 2015-05-13:
Perhaps it is because my site-packages are broken, but those tests with pip simply worked for me.
So i did a little more debugging and found the Problem: :P

`_fc_path()` inside `freecad_impl/__init__.py` returns None, wich is inserted into sys.path
Later pkg_resources is imported by another project, wich works with the strings inside sys.path
and so produces and TypeError when it find None inside sys.path. Now i will try to find out, why _fc_path yields None on my system.

moeb, 2015-05-13:
Well after looking at it, its became obvious to me
```
if sys.platform.startswith('linux'):
        #Make some dangerous assumptions...
        for _PATH in [
                os.path.join(os.path.expanduser("~"), "lib/freecad/lib"),
                "/usr/local/lib/freecad/lib",
                "/usr/lib/freecad/lib",
                "/opt/freecad/lib/",
                "/usr/bin/freecad/lib",
                ]:
            print(_PATH)
            if os.path.exists(_PATH):
                return _PATH
```

the freecad libs are in /usr/lib/freecad on archlinux xD

jmwright, 2015-05-13:
Does the FREECAD_LIB environment variable exist on your system?

moeb, 2015-05-13:
no it doesnt

jmwright, 2015-05-13:
> the freecad libs are in /usr/lib/freecad on archlinux xD

I wonder why it worked for me then?

moeb, 2015-05-13:
lol xD

jmwright, 2015-05-13:
Nope, 0.15 rev 4671 from the main repos. I tried to match your setup as close as I could.

moeb, 2015-05-13:
what did you do exactly ?

jmwright, 2015-05-13:
I'm not in front of my computer right now, but...
```
pacman -S freecad
```
Then I double checked the version in FreeCAD's About dialog.

jmwright, 2015-05-13:
I can add that path in the CadQuery code, but I'd like to understand why it works for me first. This also seems to be related to a feature request I filed awhile back.

https://github.com/jmwright/cadquery-freecad-module/issues/48

moeb, 2015-05-13:
ok, i can reproduce the error on the cmdline now !
go to the folder above cadquery and then start up python2.7
```
import sys
sys.path.append("/usr/lib/freecad")
import cadquery
import pkg_resources
```

jmwright, 2015-05-13:
Ok, I'll try that when I get back in front of a keyboard here in a bit.

jmwright, 2015-05-13:
> go to the folder above cadquery and then start up python2.7

Here's what I did.

```python
cd /usr/share/freecad/Mod/CadQuery/Libs/
python2.7
>>> import sys
>>> sys.path.append("/usr/lib/freecad")
>>> import cadquery
>>> import pkg_resources
```
That gives me an import error for pkg_resources. Do I need to install the `python2-setuptools` package?

moeb, 2015-05-13:
>Do I need to install the python2-setuptools package?

Lol that seems to be the case xD
I verified it with uninstalling+installing, just to be sure :)

jmwright, 2015-05-13:
Ok, I get the same error. Now the CadQuery module is broken for me inside of FreeCAD. It gives the `must be string, not None` error. If I uninstall `python2-setuptools`, it starts working again.

jmwright, 2015-05-14:
I just verified that adding `/usr/lib/freecad` to the list of search paths in `__init__.py` does fix the problem, even with python2-setuptools installed. I can't explain why the module worked fine before the installation of the python2-setuptools package though.

@moeb Can you verify that the module will start up for you if you uninstall python2-setuptools before you start FreeCAD?

moeb, 2015-05-14:
>@moeb Can you verify that the module will start up for you if you uninstall python2-setuptools before you start FreeCAD?

Yes it starts up for me when setuptools is not installed too.

I believe it works without setuptools, for two reasons:
1. ImportError is caught when trying to import pkg_resources, while TypeError is not.
2. freecad seems to either import it's modules, or insert their path into sys.path.

imho best would be, to detect, if cadquery is loaded by freecad or not, before inserting the
path into sys.path (or not) and to raise an exception with some meaningful msg like "cadquery was not able to find the freecad library path" at the end of `_fc_path` function.

moeb, 2015-05-15:
I filed a pull-request that fixes this :)
Best moeb

jmwright, 2015-05-15:
Awesome, thank you for contributing. I updated CadQuery's changelog to give you credit for the fix. The latest commit of this module has the updates, and runs fine for me on Ubuntu and Arch. Is it acceptable for you to use the zip of the latest commit for your package? The latest commit is basically just v0.1.8 with updates to the CadQuery library.

closed 2015-05-19 by jmwright