CadQuery/cadquery-freecad-workbench #55: is cadquery-freecad-module 0.18 up to date?

easyw, 2015-08-15:
Hi JM,
could you please check if the latest 0.18 release 
downloadable at
https://github.com/jmwright/cadquery-freecad-module
includes the latest cadquery Lib?

if I download your master at
https://github.com/jmwright/cadquery-freecad-module/archive/master.zip
and I put CadQuery dir in Mod dir I get the latest, but if I download 0.18 release from the previous link, I think I will not get the latest

Thank you
Maurice

jmwright, 2015-08-16:
Hi Maurice,

What's missing that makes you think that the latest version of CadQuery isn't bundled?

jmwright, 2015-08-16:
I double-checked manually. The "add simplify" changes made by @hyOzd were the last ones made to CadQuery. Those changes are in the version of CadQuery that's bundled in the current master, which you should get if you click the "Download ZIP" button on the repository's landing page.

easyw, 2015-08-16:
Hi JM,
in the master that is getting from "Download ZIP" the changes are present, but they are not in the link present at the page
https://github.com/jmwright/cadquery-freecad-module
https://github.com/jmwright/cadquery-freecad-module/releases/tag/v0.1.8

When I first installed "cadquery module" I just downloaded it from that page at the suggested link
"Download the latest released version" which is not aligned.

thank you
Maurice

hyOzd, 2015-08-16:
Hi @easyw , here is a tip: I don't use the cadquery shipped with the freecad-module. I've removed it from freecad-module folder. Instead I have placed a symbolic link to cadquery repository folder. It worked fine for me until now. That way, if I update my cadquery repo, freecad-module automatically uses the latest version. 

I use linux though, not sure if it would work on windows. Note that windows also has symbolic links. I've used them in the past for similar tasks.

jmwright, 2015-08-16:
@easyw and @hyOzd Whenever there's a commit to the CadQuery repository I always deploy that to the module. So, if you're not always getting the latest commit bundled with the module, something's wrong.

I can see doing it the way @hyOzd is during development, but regular users shouldn't have to jump through any hoops. Let me know if the CadQuery library is not getting updated properly. Everything looks fine from my end, so I don't know where to start.

easyw, 2015-08-16:
Hi @jmwright, sorry for the confusion.

I'm trying to explain better:
I checked 
https://github.com/jmwright/cadquery-freecad-module/archive/v0.1.8.zip
and it is not updated to cadquery latest...
I followed the wiki and downloaded from the link
https://github.com/jmwright/cadquery-freecad-module/archive/v0.1.8.zip
then I tried to load a script that we are developing and there is an error related to:
[loft() got an unexpected keyword argument 'ruled']
but if I get cadquery Lib from your zip master here
https://codeload.github.com/hyOzd/cadquery/zip/master
I don't have any problem...
I think that the 0.18 zip file does not have the latest cadquery lib
I'm doing a howto guide to use our scripts and I just followed your wiki to add CadQuery Module and the Lib inside the zip is not the latest.
Please let me know if I'm wrong.
here there is a code that is not working in 0.18 zip file, but is working in master zip file

import cadquery as cq
from Helpers import show
import FreeCAD, FreeCADGui
result = cq.Workplane(cq.Plane.XY()).workplane(offset=2). \
        rect(4.67, 3.67).workplane(offset=5).rect(4.9, 3.9). \
        workplane(offset=3).rect(4.9, 3.9).rect(4.75, 3.75). \
        workplane(offset=0.725).rect(4.52,3.52).loft(ruled=True).faces(">Z")
# Rendering
show(result)
FreeCADGui.SendMsgToActiveView("ViewFit")

Thank you
Maurice

jmwright, 2015-08-16:
@easyw I think I understand now. You are correct that v0.1.8 doesn't have the latest CadQuery library. It's a release and has version 0.1.8 of CadQuery bundled with it. To get the latest CadQuery library bundled, you'll need to download the latest mater zip, not the tagged, release version's zip. Be sure to get your zip [here]:(https://github.com/jmwright/cadquery-freecad-module/archive/master.zip).

jmwright, 2015-08-16:
One other thing that can potentially cause confusion: The version for this module is locked to the version of CadQuery. Every time there's a release of CadQuery, there will be a matching release of this module. In-between releases, the master of this module is kept in sync with the master of CadQuery.

easyw, 2015-08-16:
ok, thanks for clarifications...
I will detail to users in my howto for the scripts using cadquery module to download the master zip version

closed 2015-08-16 by easyw

jmwright, 2015-08-16:
@easyw We're at the point we're we really could justify doing another CadQuery release. I was hoping to squeeze some more unit tests in before the next release, but they're not a must. Would that help your users if we go ahead and do the release tomorrow? It will be v0.2.0 since we're fixing our versioning to align properly with symantic versioning.

easyw, 2015-08-16:
@jmwright that would be perfect!
At the moment also kicad (to which we are developing these scripts) is releasing a new version.
So everything would be in sink...
when the cq module release will be out, I'll update doc and links...
thank you
Maurice

jmwright, 2015-08-17:
@easyw - v0.2.0 of both the CadQuery lib and this FreeCAD module have been tagged/released. v0.2.0 of the CadQuery library hasn't made it into PyPi yet though, but it will soon.

Also, we've just set up a Google Group for CadQuery users. I wanted to make sure that you and @hyOzd knew about it, as well as anyone else who reads this thread later. https://groups.google.com/forum/#!forum/cadquery

easyw, 2015-08-17:
Hi @jmwright,
thank you for your very fast updating...
Please just have a check  to the links at the main page and wiki page, they still point to 0.18 version
main page https://github.com/jmwright/cadquery-freecad-module
version 0.2.0 still 
to https://github.com/jmwright/cadquery-freecad-module/releases/tag/v0.1.8
Download the latest released version 
to https://github.com/jmwright/cadquery-freecad-module/releases/tag/v0.1.8
wiki page https://github.com/jmwright/cadquery-freecad-module/wiki
Download the archive for the current release
to https://github.com/jmwright/cadquery-freecad-module/releases/tag/v0.1.8
thank you again
Maurice

jmwright, 2015-08-17:
@easyw - Thanks for reporting those, I completely forgot to update them. They should be fixed now. Please let me know if anything else has been missed.

easyw, 2015-08-17:
@jmwright 
the link at the green version button seems still wrong
[image]

jmwright, 2015-08-17:
That's fixed now. The CadQuery library's badge also had the wrong link, so I fixed that too. Thanks for the heads-up. I appreciate your attention to detail!

easyw, 2015-08-17:
@jmwright 
thanks to you for your cadquery freecad module!
I found it very useful to develop the scripts for STEP parametric 3D modules I'm doing with @hyOzd 
:)

jmwright, 2015-08-18:
@easyw Awesome! I'm excited to hear that my code is being used and that it's helping you with your project.