CadQuery/cadquery-freecad-workbench #56: problems for PhoenixContact's "DIN rail" creation

huskier, 2015-09-24:
Dear Sir, 

I try to create a PhoenixContact's "DIN rail" with CADQuery inside FreeCAD, and I've successfully got the basic model. However, when I try to punch holes on the basic model, I've got error hints which I do not know how to resolve.

result = cq.Workplane("front").moveTo(-15,10).lineTo(-10,10).lineTo(-10,0).lineTo(10,0)\
    .lineTo(10,10).lineTo(15,10).lineTo(15,8).lineTo(12,8).lineTo(12,-2).lineTo(-12,-2)\
        .lineTo(-12,8).lineTo(-15,8).close().extrude(200)

#Problem statement......
#result = result.faces("XZ").workplane().circle(5.0).cutThruAll()     

show(result)

Could you please give a hand?

Thanks in advance!

Youbao

jmwright, 2015-09-24:
Hi Youbao, and welcome. When you select a face like that, you need to give CadQuery an axis instead of defining a plane. Try the following.

```python
import cadquery as cq
from Helpers import show

result = cq.Workplane("front").moveTo(-15,10).lineTo(-10,10).lineTo(-10,0).lineTo(10,0)\
.lineTo(10,10).lineTo(15,10).lineTo(15,8).lineTo(12,8).lineTo(12,-2).lineTo(-12,-2)\
.lineTo(-12,8).lineTo(-15,8).close().extrude(200)

result = result.faces(">Y").workplane().circle(5.0).cutThruAll()

show(result)
```

[Here's](http://parametricparts.com/docs/selectors.html?highlight=selector) some documentation on string selectors.

huskier, 2015-09-24:
Dear Wright,

"<Y" selector works for me. Thanks. One more question, in the following code, polyline statement works just what I want; however, my own sketch (moveTo, threePointArc, lineTo, etc.) does not work properly. It's very weird. 

import cadquery as cq
from Helpers import show

result = cq.Workplane("front").moveTo(-15,10).lineTo(-10,10).lineTo(-10,0).lineTo(10,0)\
.lineTo(10,10).lineTo(15,10).lineTo(15,8).lineTo(12,8).lineTo(12,-2).lineTo(-12,-2)\
.lineTo(-12,8).lineTo(-15,8).close().extrude(200)

pts = [(5,0),(-5,0),(-5,10),(5,10),(5,0)]

result = result.faces("<Y").workplane().center(0,-90).polyline(pts).cutThruAll()    #This works fine

#result = result.faces("<Y").workplane().center(0,-90).moveTo(5,0).threePointArc((0,-5),(-5,0)).lineTo(5,0).close().cutThruAll()        #My own sketch, do not work properly

show(result)

jmwright, 2015-09-25:
Looks like the close function is acting improperly and causing this problem. Take a look at the result of this code.

```python
import cadquery as cq
from Helpers import show

result = cq.Workplane("front").moveTo(-15,10).lineTo(-10,10).lineTo(-10,0).lineTo(10,0)\
.lineTo(10,10).lineTo(15,10).lineTo(15,8).lineTo(12,8).lineTo(12,-2).lineTo(-12,-2)\
.lineTo(-12,8).lineTo(-15,8).close().extrude(200)

pts = [(5,0),(-5,0),(-5,10),(5,10),(5,0)]

#result = result.faces("<Y").workplane().center(0,-90).polyline(pts).cutThruAll() #This works fine

result = result.faces("<Y").workplane().center(0,-90).moveTo(5,0).threePointArc((0,-5),(-5,0)).close()#.cutThruAll() #My own sketch, do not work properly

show(result)
```

huskier, 2015-09-25:
YES, Wright, I tried many times and never find a solution to this. Then, how to solve this problem?

Thought for 2D sketch:

First create a 2D sketch object:
Obj_2DSketch = Workplane().lineTo(20,10).threePointArc((30,20),(18,28)).lineTo(40,50).close()

and then, assume there is a function like "polyline", let's call it draw2DSketch which take a parameters like 2D sketch object. and after that, we can write the following code:

result = result.faces("<Y").workplane().center(0,-90).draw2DSketch(Obj_2DSketch).cutThruAll()

Is the above thought possible?

Thanks.

dcowden, 2015-09-25:
Your concept here is perfect, and in fact you can do just that by writing a
cadquery plugin.

Have a look at this page:

http://parametricparts.com/docs/extending.html#

There is a simple example of a plugin there.

You can also look at CQ.polygon.. it is written in 'plugin style' also.

Sorry for being lazy and not giving a code example.. I'm flailing around
between commitments but trying to stay up to speed on CQ...

huskier, 2015-09-26:
Thanks for your help, Dave. I could try to write a plugin later.

However, now, I want to deal with the "close function" first.

I could create a "DIN rail" by mouse and keyboard; but I like the way CADQuery does: scripting the 3D model. BTW, to some extent, I am a coder.

dcowden, 2015-09-26:
Understood. Yeah it does appear that close needs some fixing. Unfortunately
I have like no time these days, my day job is crazy. If you're willing we
would gladly accept a pull request...

huskier, 2015-09-28:
I'd like to contribute to the project, but firstly I need to understand the CADQuery code. I'm trying......

Thanks for you guys excellent work!

dcowden, 2015-09-28:
haha-- wait, you're holding yourself to a very high standard here!

I'm not sure i understand it all, and i wrote a lot of it ;) dont worry,
there are test cases!

huskier, 2015-09-28:
Thanks. I will do my best.

huskier, 2015-09-28:
I put the following print code into the close function, and it returns my very first point (-15,10)
"print self.ctx.firstPoint.x, self.ctx.firstPoint.y"

I think that every time we finished a 3D operation, such as extrude, cutThruAll, hole, etc., we need to make the ctx.firstPoint none. Otherwise, we've got weird results.

After I put the code "self.ctx.firstPoint=None" very before the return statement in the function extrude, my problem solved.

jmwright, 2015-09-30:
@huskier Shouldn't the firstPoint be set back to the center of the workplane rather than None for future operations to work correctly (like drawing another polyline)? I'm guessing off the top of my head, so please correct me if I'm wrong.

jmwright, 2015-09-30:
@huskier Well, I just put your change into the close function on my laptop and it didn't break any of the tests, so that's a step in the right direction.

jmwright, 2015-10-12:
@huskier I have committed the firstPoint=None change to the CadQuery library, and deployed it to this module. Thanks for tracking that problem down. You've been given credit in the CadQuery changes.md file.

closed 2015-10-12 by jmwright

huskier, 2015-10-19:
Thanks Wright!