CadQuery/cadquery-freecad-workbench #58: Global variables are not deleted

galou, 2015-10-02:
If I run the script
```Python
a = 3
print a
```
then, in the same session,
```Python
print a
```
The second script does not fail as expected, probably because the global namespace still contains `a`.

jmwright, 2015-10-02:
@galou That's a good catch, thanks. 

I had to use the [imp module](https://docs.python.org/2/library/imp.html#imp.load_module) to load and execute the script in the editor, and FreeCAD must be sharing information between executes. That could lead to problems in certain situations, but I'm not quite sure how to fix it yet.

jmwright, 2015-10-02:
Looks like PyRun_File() might be the answer.

http://www.freecadweb.org/tracker/view.php?id=528

jmwright, 2015-10-12:
There a runFile function available in the C++ codebase for FreeCAD that should handle this, but I can't figure out how to call it. I've asked for help on the FreeCAD forum.

http://forum.freecadweb.org/viewtopic.php?f=22&t=12794

jmwright, 2017-08-20:
I'm not completely sure yet, but I believe the move to CadQuery Gateway Interface (CQGI) compatibility will resolve this issue. That move is underway.

jmwright, 2017-08-27:
This does seem to be fixed with the way the CQGI implementation works. I don't have it ready to merge into master yet, but I'll hopefully do that in the next few weeks. I'll release a new version when I get it merged and at least one other community member has tested it. It's a fairly significant change.

https://github.com/jmwright/cadquery-freecad-module/tree/cqgi

The only thing that will be required for this fix will be to swap the show() function call for build_object(). The Helpers import line will no longer be needed either after making that change.

jmwright, 2017-10-04:
This was fixed as of commit 6bfe99eac70ac77866a5a517aa80ef00bc8b114d

jmwright, 2017-10-04:
I guess I should add that users have to use the CadQuery Gateway Interface (CQGI) for this to work. The only thing that has to be done to make that happen is to use 'show_object()' instead of 'show()', and use the options argument instead of passing rgba the same way as before. The script template has an explanation of what the difference is.

```python
# This is a CadQuery script template
# Add your script code below
import cadquery as cq

# The show function will still work, but is outdated now
# Use the following to render your model with grey RGB and no transparency
# show(my_model, (204, 204, 204, 0.0))

# New method to render script results using the CadQuery Gateway Interface
# Use the following to render your model with grey RGB and no transparency
# show_object(result, options={"rgba":(204, 204, 204, 0.0)})
```

closed 2017-10-04 by jmwright