CadQuery/cadquery-freecad-workbench #63: object has no attribute 'write'

triplus, 2015-10-18:
Hi.

As described here:

http://forum.freecadweb.org/viewtopic.php?f=9&t=3914&start=80#p103014

When i install CadQuery FreeCAD module (Ubuntu 14.04) and run one of the unrelated macros i get mentioned error (without switching to CadQuery WB first). I hope it will be obvious what could be causing the issue.

jmwright, 2015-10-18:
Can you please post the steps to reproduce your error? I don't see any errors while doing WebGL export using the latest commit of the CadQuery workbench in FreeCAD 0.16 on Ubuntu 15.04.

triplus, 2015-10-18:
Hi.

If you will look in the FreeCAD forum thread there is some new information in it. And i adapted code snippet from here:

https://wiki.qt.io/PySideTutorials_Simple_Dialog

``` python
import sys
from PySide.QtCore import *
from PySide.QtGui import *

class Form(QDialog):

    def __init__(self, parent = None):
        super(Form, self).__init__(parent)
        self.setWindowTitle("My Form")
        self.edit = QLineEdit("Write my name here")
        self.button = QPushButton("Show Greetings")
        layout = QVBoxLayout()
        layout.addWidget(self.edit)
        layout.addWidget(self.button)
        self.setLayout(layout)
        self.button.clicked.connect(self.greetings)

    def greetings(self):
        print ("Hello : " + self.edit.text())

if __name__ == '__main__':
    form = Form()
    form.show()
```

If cadquery-freecad-module is installed in Mod directory print statement will have mentioned issue. If removed it will work fine.

Hopefully this gives the idea of what could be going wrong.

jmwright, 2015-10-20:
Your code snipped doesn't throw an error for me.

Could you try the latest commit of this module, please? I removed a PrintMessage call that didn't use the result of QtGui.QApplication.translate like the others.

triplus, 2015-10-21:
Hi. I still get:

AttributeError: 'list' object has no attribute 'write'

jmwright, 2015-10-21:
What version of FreeCAD are you running, and what version of Windows/Linux/MacOS?

triplus, 2015-10-22:
Ubuntu 14.04
FreeCAD installed from Daily PPA

jmwright, 2015-10-22:
I've got the same setup except for running 15.04. Do you have any other custom plugins installed?

jmwright, 2015-10-22:
Sorry, I mean workbenches, not plugins.

jmwright, 2015-10-26:
@triplus I have set up a fresh Ubuntu 14.04 virtual machine and installed FreeCAD from the daily PPA, installed the latest master of this CadQuery module, and then ran the macro that you pasted above. I do not get any errors. We need to figure out what differences there are between your installation and mine. Have you customized your Python installation at all? I assume it's possible that there's a conflict between this module and another on your system, so a list of custom workbenches that you have installed might help.

triplus, 2015-10-26:
Hi.

I didn't see your answer and yes i do use more 3rd Party modules. I will remove one by one from Mod directory and see if there is any difference. I will provide more feedback in about 2 hours.

triplus, 2015-10-26:
I did some tests and i think i narrowed it down to a clash between cadquery-freecad-module and FreeCAD_assembly2:

https://github.com/hamish2014/FreeCAD_assembly2

Can you confirm the same behaviour?

jmwright, 2015-10-27:
No, I don't see any errors as long as I make sure python-numpy is installed. Without python-numpy, I get an error that says `module has no attribute file`.

Do you have one of the two workbenches selected when you try to run the macro from above?

triplus, 2015-10-27:
No i don't touch the mentioned workbenches after FreeCAD starts and i run the macro. But now i tested what happens if i do and if i load CadQuery WB before i run the macro i don't get the error.

Therefore did you try to set Start up WB in Preferences to for example Part Design and after starting FreeCAD run the macro again directly. That should get you to mentioned issue. If not i am not sure what could be the difference.

Probably unrelated but when i start CadQuery WB i get:

Cannot find icon: :/icons/macro-execute.svg
CadQuery 0.1.8
CadQuery is a parametric scripting language for creating and traversing CAD models
Author: David Cowden
License: LGPL
Website: https://github.com/dcowden/cadquery

And this:

Opened /home/user/.FreeCAD/Mod/CadQuery/Examples/Ex000_Introduction.py
RuntimeWarning: Parent module 'temp' not found while handling absolute import
import cadquery

RuntimeWarning: Parent module 'temp' not found while handling absolute import
from Helpers import show

jmwright, 2015-10-27:
Ok, so I have Assembly2 and CadQuery installed, and I've set the "Part Design" workbench to come up by default. I restart FreeCAD, go up to Macro->Macros... and click on the test macro (that you posted above), then click the Execute button. The dialog comes up, and I'm able to type text. I don't get any errors.

Here's what I'm running:
FreeCAD 0.16, Revision number 5838
Ubuntu 14.04.3, 64-bit

triplus, 2015-10-27:
Yes and after you type in the text press on the button and look in FreeCAD Report view window. I tested this further on why i might get the mentioned issue and you can't reproduce it.

Go in Edit -> Preferences... -> General -> Output window and enable both options under Python interpreter section and apply the changes.

Restart FreeCAD and after test the macro again. Hopefully this time you will be able to reproduce the issue.

jmwright, 2015-10-27:
Ok, so we've got a little bit of progress now. I now get a segfault when I follow your instructions. There's no mention of the write error though.
```
*** Abort *** an exception was raised, but no catch was found.
	... The exception is:SIGSEGV 'segmentation violation' detected. Address 0
```

jmwright, 2015-10-27:
Ok, I think I found it. I have the code redirecting the stdout to the report view, and that was causing the problem. I don't think that code is even needed in the newer versions of FreeCAD, so I just commented it out.

Please try the latest commit of the module to see if that fixes the problem for you.

triplus, 2015-10-27:
I tested the latest revision and it is working without reported issue now.

P.S. Just wondering:

1.) Why did the Assembly2 module had an affect on cadquery-freecad-module behaviour.
2.) Do you get the same messages in Report view about the icon missing and RuntimeWarnings when switching to CadQuery WB?

jmwright, 2015-10-27:
> 1.) Why did the Assembly2 module had an affect on cadquery-freecad-module behaviour.

Both modules may have been trying to redirect stdout, and were conflicting. I haven't looked through the codebase of Assembly2 to verify that yet though.

> 2.) Do you get the same messages in Report view about the icon missing and RuntimeWarnings when switching to CadQuery WB?

Yes. I'm actually looking into those right now. It looks like the macro-execute SVG file has been removed in FreeCAD 0.16+, but I can't figure out what the replacement is. I'm not sure what the temp errors are in relation to. I've never seen them error before the latest daily updates of FreeCAD.

jmwright, 2015-10-27:
Ok, so the icon is fixed now. I think that the macro-execute SVG was basically the same as the media-playback-start SVG, so they just combined them.

jmwright, 2015-10-27:
@triplus - And thanks for working through this with me, I appreciate it.

triplus, 2015-10-27:
No problem. 

P.S. I looked and indeed the mentioned icon just got renamed in FreeCAD and therefore the icon is still the same just uses different name now.

jmwright, 2015-10-28:
The Runtime Warnings are now fixed (commit 3f1bd3b70f1e8b7145e060f2be5233fe9cb4deb7). I was using `imp.load_module` with a period in the `name` argument, which causes the function to expect a package-module relationship. I changed the period to an underscore and the problem went away.

closed 2015-10-28 by jmwright