CadQuery/cadquery-freecad-workbench #65: Automating scripted Box with editable covers and panels

huskier, 2015-11-22:
We want to create a paramatic box with editable covers and panels with CADQuery. However, the following code give us a whole box as an integraged part, and we cannot edit(e.g. hide or show covers and panels individually) the six faces individually. We're not sure whether CADQuery can do this thing. We know that it creates whole parts powerfully. If not, do we have other scripting options for this application? Thanks.

```
# This example is meant to be used from within the CadQuery module of FreeCAD.
import cadquery as cq
from Helpers import show

# The dimensions of the model. These can be modified rather than changing the
# object's code directly.
width = 40
height = 20
thickness = 2

# Create a plate with two polygons cut through it
top_prt = cq.Workplane("XY").box(width, height, thickness).translate((0,0,10))
bottom_prt = cq.Workplane("XY").box(width, height, thickness).translate((0,0,-10))
left_prt = cq.Workplane("XZ").box(width, height, thickness).translate((0,-10,0))
right_prt = cq.Workplane("XZ").box(width, height, thickness).translate((0,10,0))
front_prt = cq.Workplane("YZ").box(height, height, thickness).translate((20,0,0))
back_prt = cq.Workplane("YZ").box(height, height, thickness).translate((-20,0,0))

result = top_prt.union(bottom_prt).union(left_prt).union(right_prt).union(front_prt).union(back_prt)

print result.all()
# Render the solid
show(result)
```

dcowden, 2015-11-22:
what, more specifically, do you mean by 'hide/show the panels
individually'? Are you referring to the ability to script ability to
optionally hide some of the panels? This is very easy-- you simply put
conditional logic into your script, like

if show_top:
     //create the top

Or is there something else you'd like to do?

jmwright, 2015-11-22:
@huskier Here are also a couple of examples to look at along the lines of what @dcowden is talking about.

[Parametric enclosure](https://github.com/jmwright/cadquery-freecad-module/blob/master/CadQuery/Examples/Ex023_Parametric_Enclosure.py)

[Remote enclosure](https://github.com/jmwright/cadquery-freecad-module/blob/master/CadQuery/Examples/Ex027_Remote_Enclosure.py)

Notice how the second example uses separate show statements at the bottom so that different parts of the enclosure can be displayed differently (hidden, shown, have different color and/or opacity, etc).

huskier, 2015-11-23:
By "editable covers and panels" and "hide/show", I mean that we could edit the covers and panels (or i.e. parts) after exported to STEP file then opened in a CAD environment.

The problems solved once I removed the "union" and "combine" operations, and just leave the parts alone, and show them all. The reminder of "uses separate show statements" is the key to the problem. Thanks.

However, the part names appeared in the ComboView's Model tree are not meaningful, such as Shape0_623164280712, Shape0_551131371949, etc. Although we could rename them in the FreeCAD GUI environment, it is much better than we could name them in the scripting environment. Could we manipulate the part names appeared in the Model tree?

jmwright, 2015-11-23:
@huskier Currently, the CadQuery module just assigns a random ID because you can't have two shapes with the same name. Would it help if I could add the variable name to the shape name in the combo view?

huskier, 2015-11-23:
@jmwright 

>Would it help if I could add the variable name to the shape name in the combo view?

That would be helpful if the variable name appear in the combo view instead of random IDs. But it would be much better if we could assign a name there through a property of the variable, such as top_part.name = "top cover".

Thanks.

jmwright, 2015-11-23:
I'll look into it, but we need to be careful not to tightly couple the CadQuery library to the CadQuery module for FreeCAD. The library needs to stay stand-alone so that someone doesn't need the module to use CadQuery.

jmwright, 2015-11-24:
@dcowden What do you think about adding an instance attribute of "name" to the freecad_impl.Shape class? I could then reference that name when creating the shape ID in FreeCAD.

huskier, 2015-11-24:
In the FreeCAD's Combo View, it is called "Label". [Ref] https://github.com/huskier/images/blob/master/FreeCADComboView.png

Just a piece of information, NO preference at all.

dcowden, 2015-11-24:
that's ok with me. so is 'label'

jmwright, 2015-11-24:
@huskier I have added a "label" instance attribute to all of the shapes in the freecad_impl module in the CadQuery library. That commit of the library has been included in this module's latest commit 70949df5ba391ef2f80799363fa6146d0909fb48

To use it, you would add the following code right before the call to Helpers.show in your code above.
```python
result.val().label = "Test"
```
When you execute the script you'll see that label show up in the combo view. The code just trusts the user to ensure that they have unique labels for all solids. FreeCAD will automatically append something like 001 on the end of the label if it's a duplicate.

If you don't provide a label, the shape will get a random ID like it was before.

Does that work for you?

huskier, 2015-11-25:
@jmwright  Thank you very much for your effort.
Still some questions:
1. Why do we need the ".val()" part? I think 
     result.label = "Test"
     is much simple!

2. label support problem: [Ref] 
    a. for the top_prt, I label it as "Top Part", but it appears in the ComboView as "Top_Part". There is an extra "_" character. I tried  that the SPACE character is supported in FreeCAD, like the name in the image "Bottom Part".
    b. About Chinese support: I label front_prt as "前面部分(Chinese)", but it appears as "______" in FreeCAD.  I Changed the "LeftPart" into "左边"，it's OK. So Chinese label name is supported in FreeCAD.

3. How do I export the six parts without the CadQuery-FreeCAD-module support? For some applications, it is quite useful.

Thanks!

jmwright, 2015-11-25:
@huskier The val() part is needed because of the FreeCAD abstraction layer. It's designed to let us swap other CAD kernels in for FreeCAD/OpenCASCADE. Every CQ object can wrap freecad_impl.shapes, which in turn can wrap FreeCAD shapes. So, val() gets you the wrapped freecad_impl shape, and calling wrapped on that gives you the FreeCAD shape.

I'm not sure where that extra underscore character comes from. I'm not adding that, so FreeCAD's scripting interface must be doing it.

As far as the Chinese characters, the module does not do a good job of localizing languages. That's something that I need to fix.

What do you mean by exporting? Do you mean that you want to automatically export 6 parts to STEP files or something?

huskier, 2015-11-25:
>What do you mean by exporting? Do you mean that you want to automatically export 6 parts to STEP files or >something?

YES, I want to export the 6 parts into a single STEP file. Actually it is a one target object, i.e. a box.

jmwright, 2015-11-25:
I'm not sure how the exportStep function will work with multiple solids/compounds, but there are multiple examples in the CadQuery unit tests.

https://github.com/dcowden/cadquery/blob/master/tests/TestCadQuery.py#L79

Will that line of code do what you need?

huskier, 2015-11-26:
>Will that line of code do what you need?

Not really. The exportStep() function only exports one whole part, not workable for multiple parts.

However, when we use 
```
result = top_prt.union(bottom_prt)
result.val(),exportStep("box.step")
```

There are two label items in the STEP file. But if we union another one part, there is only one label item available in the STEP file. Very weired!!! Also, the label name is like "Open CASCADE STEP translator 6.8", not meaningful as well.

jmwright, 2015-11-26:
You might want to file an issue on CadQuery for this. Sounds like we may need to dig into STEP export, and evaluate whether there's a viable alternative to the FreeCAD/OCE method, or if we can contribute a fix.

https://github.com/dcowden/cadquery/issues

huskier, 2015-11-26:
@jmwright  OK, Thanks. I will fire an issue there.

Thank you very much for your help.

closed 2015-11-26 by huskier

jmwright, 2015-11-26:
I've filed an issue to improve the internationalization/localization support in this module too.