CadQuery/cadquery-freecad-workbench #70: Troubles breaking a project into modules

noraef, 2015-12-18:
Hi,

I'm starting with the CadQuery FreeCAD module, but would like the main .py module I load from FreeCAD to import such things as dimensions from another .py module, so to reuse those dimensions in other parts of the project, but am having two troubles:

1) the path of the .py file loaded from FreeCAD doesn't seem loaded into the Python interpreter, and there doesn't even seem to be a way to automatically find it from this main module itself; I'm not proficient with Python at all, but all the methods have tried seemed to point to the interpreter believing this main module resides in /tmp... so I have to rely to such a thing as:

    import sys
    sys.path.append("/path/to/project/dir")
    import dim

which will be a bugger if I decide to change the directory I store the project in. Naively, I'd expect the directory of the .py file I load in FreeCAD to automatically be appended to the interpreter path, so I can call other modules easily instead of having to hardcode storage locations.

2) speaking of which: when making a change to this dim.py file, and then executing the main script from FreeCAD again, the change of the variables values is not reflected in the project until I restart FreeCAD. So, if I've defined 
`w = 10`
in the dim.py, when I load the main module in FreeCAD, which calls for
`dim.w`
but I then changed it to
`w=20`
in dim.py, and I then execute the script again in FreeCAD, it all behaves as if dim.w was still equal to 10, not to 20. I have to exit FreeCAD, then restart it, and load the main module again before the change to dim.w is reflected in the drawn part.

Don't know if I'm doing something wrong... as said, I'm not proficient with Python at all...

jmwright, 2015-12-18:
Hi @noraef and welcome.

A couple of things come to mind to try. 
1. Have you tried a [relative import](https://docs.python.org/2.5/whatsnew/pep-328.html) of dim.py? I think that should work.
2. You could also set your main .py and dim.py files up in as being part of a package with an __init__.py (the double underscores are being converted to bold by Github) file at the directory level. Then you should be able to use package import lines like you see in the example [here](https://opendesignengine.net/projects/yavin-thruster/repository/entry/trunk/tests/chamber/test_cad.py#L3).

Do either of those help you at all?

noraef, 2015-12-19:
One thing helped me a lot: I renamed the main file from (actually) draft.py to main.py, and now, I can simply call
`import dim`
and it now works alright without having to add an absolute path.

Have had no luck with relative import, though. When calling for such a thing as
`from . import dim`
FreeCAD Python console ends up outputting
`ValueError: Attempted relative import in non-package`

Don't get why, as I've had an (empty) init.py (with the two double underscores) in the project folder all along...

Also, still no luck with modifications to the dim.py file being reflected in FreeCAD unless I exit and then enter FreeCAD back...

FWIW, I'm using Fedora 22, FreeCAD 0.15 and Python 2.7.10.

jmwright, 2015-12-22:
> Also, still no luck with modifications to the dim.py file being reflected in FreeCAD unless I exit and then enter FreeCAD back...

Try the following line to see if your dim.py updates are loaded.
```python
reload(dim)
```
http://stackoverflow.com/questions/437589/how-do-i-unload-reload-a-python-module

noraef, 2015-12-23:
Ha, thanks! It works great with reload. I really need to get deeper into Python... though it seems quite weird to me in some aspects (eg a package import doesn't easily import its functions by default...). I also pretty much have resolved my problems with imports, revamping my work folders.

Though it seems one has to be extra-careful with imports, seeing as anything imported stays in FreeCAD environment, without easy ways to clean it apart from exiting FreeCAD, and can still influence it (guess it's nothing you can do about).

closed 2015-12-23 by noraef

jmwright, 2015-12-28:
Issue #58 covers the problem of variables sticking around between executions of the script. That's a big problem, but not one that I've been able to figure out yet. It seems like I'm fairly limited in the FreeCAD environment on setting up something like a virtual environment.