CadQuery/cadquery-freecad-workbench #73: duplicate name

fayrez, 2016-02-02:
this is small one (and no practical problem), but for now and for the future Work's sake... 
please check how addon uses _PillowBlock.PNG_ and _pillowblock.png_ files in 

> cadquery-freecad-module/CadQuery/Libs/cadquery-lib/doc/_static/

because in Windows OS  they are same names and cannot be in the same dir.

jmwright, 2016-03-01:
Thanks @fayrez

jmwright, 2016-04-26:
@dcowden It looks like the lower case pillowblock.png is the only one that's being used. Does that mean that we can just delete the other one?

dcowden, 2016-04-26:
probably. Times were fast and loose back when i made those samples, and i
was still developing on windows.

jmwright, 2016-04-27:
CadQuery pull request [145](https://github.com/dcowden/cadquery/pull/145) includes a name change for the file causing the conflict.

Thanks again to @fayrez for taking the time to point this out.

closed 2016-04-27 by jmwright