CadQuery/cadquery-freecad-workbench #75: Certain Characters In Filename Creates New Doc Each Execution

jmwright, 2016-06-27:
This is a reference to an issue that was opened on the main CadQuery repo.

https://github.com/dcowden/cadquery/issues/149

Coding-Badly, 2016-06-28:
@jmwright, what do use to debug problems like this one?

jmwright, 2016-06-28:
I suspect the problem is in this code here: https://github.com/jmwright/cadquery-freecad-module/blob/master/CadQuery/Gui/Command.py#L142

I'll probably just start debugging with print statements since it should be a fairly easy problem to track down. It's a little difficult to debug the code directly since the module is only designed to run within FreeCAD.

jmwright, 2016-07-29:
It appears to be the FreeCAD.getDocument() function that is causing the problem. I think about the only thing I can do is translate disallowed characters into something harmless (like an underscore) before trying to set the document up for execution. The filename would stay the same, but the tab name wouldn't match it exactly.

jmwright, 2016-08-01:
This was fixed in commit c319b4f7f0e87e891f18f8ebe5a41744f7b7979d

It is available by downloading/cloning master for the time being, and will be part of a version release at some point.

@Coding-Badly Thanks for reporting this.

closed 2016-08-01 by jmwright