CadQuery/cadquery-freecad-workbench #77: Edge direction tests fails in FreeCAD 0.17

noir-soleil, 2016-12-14:
For example this code:
`result = cadquery.Workplane("XY").box(3, 3, 0.5).edges("|Z").fillet(0.125)`
from example Ex020 fails, complaining that edges must be selected for fillet function.

After some discovery i have found that it happens because straight edges now has the type _FreeCADPart.LineSegment_ instead of _FreeCADPart.Line_

The simple addition of
`FreeCADPart.LineSegment: 'LINE',`
in _edgetypes_ of _class Edge_ fixed the issue for me, but it causes crash with FreeCAD 0.16 and earlier.

jmwright, 2016-12-15:
@noir-soleil Thanks for reporting this. I see the same behavior where FreeCAD stable (0.16) works but FreeCAD daily (0.17) doesn't. It seems the FreeCAD team has made a breaking change in their underlying line objects.

Are you running the FreeCAD daily PPA on Ubuntu, compiling 0.17 yourself, or something else? I wonder if this will be fixed before 0.17 is officially released.

noir-soleil, 2016-12-15:
@jmwright: I am using prebuilt binaries for 64bit windows:
OS: Windows 7
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.17.9175 (Git)
Build type: Release
Branch: master
Hash: 26e1a24060df702816e9561b8473bd3fb41825e2
Python version: 2.7.8
Qt version: 4.8.7
Coin version: 4.0.0a
OCC version: 7.0.0

jmwright, 2017-03-14:
@dcowden I'm interested in fixing this, but I'm not thrilled about adding more conditional code to program around FreeCAD version differences. Thoughts?

easyw, 2017-04-19:
@jmwright @dcowden @noir-soleil
Here a solution commonly  adopted that is working both for FC 0.16 & FC0.17
https://forum.freecadweb.org/viewtopic.php?t=17742&start=20#p147436

jmwright, 2017-04-19:
@easyw Thanks for this. I was looking through class documentation yesterday trying to figure out how to handle it. I like this solution better than anything I had come up with so far.

dcowden, 2017-04-19:
I may try to have a look at this...

dcowden, 2017-04-19:
@easyw @noir-soleil 

Ok so i think the issue is that in this code from cadquery.Edge:

```
        self.edgetypes = {
            FreeCADPart.Line: 'LINE',
            FreeCADPart.ArcOfCircle: 'ARC',
            FreeCADPart.Circle: 'CIRCLE'
        }
```
for FC > 0.16, we need to have FreeCADPart.LineSegment. But then i guess that FreeCADPart.Line nullpointers on FC <= 0.16?

So the fix is to change the code here:

```
    def geomType(self):
        t = type(self.wrapped.Curve)
        if self.edgetypes.has_key(t):
            return self.edgetypes[t]
        else:
            return "Unknown Edge Curve Type: %s" % str(t)
```
do use hasattr before calling FreeCADPart.<LINE|LINESEGMENT>

Is that right?

easyw, 2017-04-19:
also 'makeLine' has to be reviewed

jmwright, 2017-04-22:
@dcowden How far did you get on this? Should I start a branch incorporating your changes and starting on makeLine, or have you already done this?

dcowden, 2017-04-22:
Hi Jeremy, I have not made any changes in the repo yet. Ive been busy in
the first robotics championship... Expecting to be able to work on it next
week.

dcowden, 2017-04-23:
Ok, the latest CQ master branch has the recommended fix.  All tests pass on FreeCAD 0.16, while these fixes support FreeCAD 0.17.

However,9 tests still fail for 0.17 when running against the latest daily build.

I started looking into them, but I think we should wait till FreeCAD 0.17 is stable-- otherwise, i think we could waste a lot of time.

jmwright, 2017-04-24:
Looks like 0.17 is scheduled for these at the end of next month. https://www.freecadweb.org/tracker/roadmap_page.php

dcowden, 2017-04-24:
ok thanks. I think we'll table this until that's finished then.  With luck, things will 'just work'. If not, we'll have to dive in to see whats' going on.

jmwright, 2017-04-30:
Ex020 runs with the latest master. @dcowden has opened an issue to follow up on any remaining FreeCAD 0.17 incompatibilities when it is stable.

https://github.com/dcowden/cadquery/issues/183

I'm going to go ahead and close this since I think we have all our bases covered.

closed 2017-04-30 by jmwright