CadQuery/cadquery-freecad-workbench #79: Try to Get this Module Into FreeCAD-addons

jmwright, 2017-03-01:
The genesis of this is issue #78 

https://github.com/FreeCAD/FreeCAD-addons

Easier installs are always good, and this would at the very least give users another path to get this module installed and working.

jmwright, 2017-03-01:
I've asked for help getting this module included here: http://forum.freecadweb.org/viewtopic.php?f=22&t=21014

jmwright, 2017-03-03:
I'm concerned that restructuring the files in this module for the FreeCAD-addons repo will force users who download the ZIP to go through an extra step. I believe they'll need to create a "CadQuery" directory and then extract the ZIP into that. Not a huge deal, but it's a change from how we've been doing it all this time.

jmwright, 2017-04-15:
@easyw Our conversations on the other issues today reminded me of this. Would this help your users at all? If I remember right you point your users to this module's releases. I'd have to take the top layer out of the directory structure which means that people who download the zip would have to create the CadQuery directory in *FreeCAD/Mod manually and then copy the contents of the zip into that. What are your thoughts?

easyw, 2017-04-16:
@jmwright 
probably to let the git be complaint with the FC add-on plugin you could have the repo organized as following:

> .gitignore
> changes.md
> LICENSE
> README.md

at the same level of the content of the actual CadQuery folder (no more a CQ folder inside the zip)
Then you zip file will be:
cadquery-freecad-module-master
that can be moved to the right directory when managed by the FC add-on plugin

For people downloading the git zip file the only operation to do would be copying the folder 'cadquery-freecad-module-master' to the right place and **optionally** rename it to CadQuery
I noticed that in FC when you put a Mod in the right place, it will work even though the name is the used WB name
You can check it just manually doing these operations
Maurice

jmwright, 2017-04-16:
@easyw Thanks, I'll experiment with that l.

jmwright, 2017-04-17:
I've restructured the repo and asked to have the FreeCAD-addon commit link to the repo updated.

easyw, 2017-04-17:
the add-on manager already  installs it and works like a charm!
thx

jmwright, 2017-04-17:
@easyw Awesome, thanks. I'll add a subsection on this installation method in the readme.

jmwright, 2017-04-17:
The FreeCAD-Addons installation method has been documented in the Readme, and I've confirmed that it works on Windows too. Yay for easier installs and updates!

closed 2017-04-17 by jmwright