CadQuery/cadquery-freecad-workbench #81: Object not visible in R1.0.0 but visible in R0.5.2

hackscribble, 2017-04-14:
I am seeing a difference in behaviour between 1.0.0 and 0.5.2 with the following test program.  I'm using FreeCAD 0.16 on Ubuntu 16.10.

```
import cadquery as cq
from Helpers import show

a = cq.Workplane("XY").rect(5, 5).extrude(1)
b = cq.Workplane("XY").rect(3, 3).extrude(2)

show(a)
show(b)
```
In 0.5.2, this shows ...

[image]

In 1.0.0, I see this ...

[image]

Adding a union gives the expected result.

Is this a bug or expected change under 1.0.0?

Thanks

jmwright, 2017-04-15:
I don't think there was an intentional change, but I have sometimes had some strange behavior like that when dealing with nested solids. I'm not sure if this helps, but this is a more idiomatic way to do the same thing in CadQuery.

```python
import cadquery as cq
from Helpers import show

a = cq.Workplane("XY").rect(5, 5).extrude(1) \
                      .faces(">Z").workplane() \
                      .rect(3, 3).extrude(1)
                      
show(a)
```

The code now uses the max face in the Z direction to establish a new workplane, and builds the new solid extrusion from there. I don't know that I've ever had problems with solids doing it this way. In short, I'm not sure why 1.0.0 is doing that and 0.5.2 doesn't, but I try to avoid building nested, intersecting solids because of the strangeness it has caused for me in the past.

Does that help? If you need to do nested solids for some reason we can dig into it a little more.

hackscribble, 2017-04-15:
Thanks @jmwright There are a bunch of scripts at https://github.com/easyw/kicad-3d-models-in-freecad that will break in 1.0.0, I think.  

The scripts generate an electronic component as separate solids, rather than one fused one, so that the individual parts can be coloured differently in a subsequent export to STEP and WRL.

I have tweaked my example so that it looks more like a 1-pin pin header.  The pin and body solids would be coloured differently in the later export process.

```
import cadquery as cq
from Helpers import show

body = cq.Workplane("XY").rect(4, 4).extrude(2)
pins = cq.Workplane("XY").workplane(offset=-2).rect(0.5, 0.5).extrude(10)

show(body)
show(pins)
```

[image]

The different behaviour is also there with non-intersecting objects.  In 0.5.2 this script shows two solids. 
In 1.0.0 I can only see one or the other, by alternately commenting out the show() statements.

```
import cadquery as cq
from Helpers import show

a = cq.Workplane("XY").rect(5, 5).extrude(1)
b = cq.Workplane("XY").center(10,0).rect(3, 3).extrude(10)

show(a)
show(b)
```

[image]

easyw, 2017-04-15:
@hackscribble @jmwright
All these issues seem related only to the new 'show' function in Helpers.py 
Reverting to the old function one (0.5.2) everything is working fine

easyw, 2017-04-15:
ok this is related to:
`Shared.clearActiveDocument()`
in 'Helpers.py'
and there there is also this issue, compared to release 0.5.2
https://github.com/jmwright/cadquery-freecad-module/issues/82
We can manage these on our scripts, but are those changes really needed?
Thx 
Maurice

jmwright, 2017-04-15:
Ok, thanks guys. Sorry, I dropped the ball on troubleshooting these two issues. The tabbed interface changes were significant, but I'm hoping these growing pains will be worth it. I'll take another look based on what @easyw found. I don't want users to have to deal with a broken interface.

@easyw Have you never had trouble with the nested solids like that pin example? It makes me wonder if I was doing something else wrong when I've had trouble before.

easyw, 2017-04-15:
@jmwright 
> Have you never had trouble with the nested solids like that pin example?

not really a part from similar problems you can have with FC when making a union of parts were coplanar faces are involved...
Thx for going into this and for your CQ plugin! We are doing so much with it 😸

jmwright, 2017-04-15:
@easyw If Shared.clearActiveDocument is not called somewhere in the execution chain, then you have duplicate objects stack up in the 3D view. I think the best thing to do here is to move the clearing of the document into the higher level execution code so that you can call Helpers.show as many times as you want. As you two have noted, that's the way it used to work, and that makes the most sense.

jmwright, 2017-04-15:
@easyw Please try the latest commit to master to see if it fixes your issue.

I would still like to understand your workflow better. For instance, I assumed that you'd have a blank document when the CadQuery code is executed as a macro, but can I really assume that? Should I have it add a document window if one doesn't exist?

jmwright, 2017-04-15:
@hackscribble The bug you reported should be fixed in master now too. It works properly for me. 

Thanks to both of you for reporting these problems and sticking with it to make sure they got fixed.

easyw, 2017-04-15:
@jmwright 

> I would still like to understand your workflow better. For instance, I assumed that you'd have a blank document when the CadQuery code is executed as a macro, but can I really assume that?

in fact we just import cadquery module and execute commands read from the python script itself

> Should I have it add a document window if one doesn't exist?

not necessary for our scripts... In fact if a 'script_template.py' page is open, we would like to close it during the script execution
I posted a query about that in 
https://github.com/jmwright/cadquery-freecad-module/issues/82

Thx again
Maurice

hackscribble, 2017-04-16:
@jmwright Thanks for the code update.  Tests OK here :smile:

closed 2017-04-16 by hackscribble

jmwright, 2017-04-17:
@hackscribble @easyw @dcowden I've released a new version with the latest fixes rolled in - v1.0.0.1

The version numbering is a little strange, but I wanted to show that it's a module revision that's still based on v1.0.0 of the CadQuery library. I've always held back on doing intermediate releases because of getting version numbers out of sync with the core library, but I think this numbering scheme addresses that concern.

https://github.com/jmwright/cadquery-freecad-module/releases/tag/v1.0.0.1

dcowden, 2017-04-17:
you're on fire! thanks for the super-fast iterations!