CadQuery/cadquery-freecad-workbench #82: show(result) throw an error

easyw, 2017-04-14:
Hi JM,
we have a lot of scripts using CQ FC module here
https://github.com/easyw/kicad-3d-models-in-freecad
everything is fine with CQ 0.52...
I'm trying CQ 1.0 and I get this error running the scripts:
File "C:\Users\mau\AppData\Roaming\FreeCAD\Mod\CadQuery\Helpers.py", line 23, in show
    tempFile.write(cqCodePane.toPlainText().encode('utf-8'))
<type 'exceptions.AttributeError'>: 'NoneType' object has no attribute 'toPlainText'

here a very simple example generating this error (in Win 10 64b FC 0.16, 0.16 & 0.17):
> import cadquery
> from Helpers import show
> result = cadquery.Workplane("XY").box(2,1,1)
> show(result)

Thx
Maurice

jmwright, 2017-04-14:
Can you post your FreeCAD version and operating system  please?

jmwright, 2017-04-14:
Sorry, missed the last line of your comment.

jmwright, 2017-04-15:
I've tried the sample code on both Windows and Linux, and it works without error for me.

Here's my Python information in both Windows and Linux.

 Linux:
```
Python 2.7.12 (default, Nov 19 2016, 06:48:10) 
[GCC 5.4.0 20160609] on linux2
```

Windows:
```
Python 2.7.8 (default, Nov 17 2014, 20:37:05) [MSC v.1800 64 bit (AMD64)] on win32
```

Here's the info on FreeCAD 0.16 in my Windows 10 VM.

[image]

Are there any differences between my setups and yours? Do you have anything custom in your FreeCAD or Python installations?

Also, what language do you use in your OS? I'm wondering if we have a strange character encoding issue going on since we're dealing with Python 2.x.

easyw, 2017-04-15:
OK, I found the problem...
from the error reported
> File "C:\Users\mau\AppData\Roaming\FreeCAD\Mod\CadQuery\Helpers.py", line 23, in show
tempFile.write(cqCodePane.toPlainText().encode('utf-8'))
<type 'exceptions.AttributeError'>: 'NoneType' object has no attribute 'toPlainText'

your new Helpers.py assumes you have the cqCodePane open in your FC GUI...
this was not necessary in your old FC cadquery plugin ...
If I open a CQ cqCodePane the  script runs fine...
this is the way I solved the error generation:

>     try:
>         # Grab our code editor so we can interact with it
>         cqCodePane = Shared.getActiveCodePane()
>     
>         # Clear the old render before re-rendering
>         Shared.clearActiveDocument()
>     
>         # Save our code to a tempfile and render it
>         tempFile = tempfile.NamedTemporaryFile(delete=False)
>         tempFile.write(cqCodePane.toPlainText().encode('utf-8'))
>         tempFile.close()
>     
>         docname = os.path.splitext(os.path.basename(cqCodePane.file.path))[0]
>     
>         # Make sure we replace any troublesome characters
>         for ch in ['&', '#', '.', '-', '$', '%', ',', ' ']:
>             if ch in docname:
>                 docname = docname.replace(ch, "")
>         # If the matching 3D view has been closed, we need to open a new one
>         try:
>             FreeCAD.getDocument(docname)
>         except:
>             FreeCAD.newDocument(docname)
>     except:
>         pass

Still having some probs whit the scripts but I have to investigate further
Thx
Maurice

jmwright, 2017-04-15:
@easyw How and why do you run the script (call Helpers.show) without having a code editor window open? Are you running the CadQuery script as a macro?

jmwright, 2017-04-15:
@easyw Please check latest commit to master. It should fix your issue. I also asked a follow-up question in the other issue.

easyw, 2017-04-15:
@jmwright 
now all the issues are gone... thx for the prompt solution 😄 
I just noticed that if I switch from i.e. Part WB to CQ WB a 'script_template.py' page is open...
and if I switch back to an other WB i.e. PartDesign and back to CQ WB an other 'script_template.py' is open again...
is it intended? Is there a way to close this 'script_template.py' by a python call?
thx again
Maurice

jmwright, 2017-04-15:
That behavior is intended for new users but I'm guessing it's getting in the way in your case. I can add an option in the settings file to disable that. Will that help?

easyw, 2017-04-15:
@jmwright 
more than a setting, which would depend on user settings, it would be nice if the script_template.py could be closed by a python command when the script is running

jmwright, 2017-04-15:
If you're wanting to close it from your scripts, here's how I close the active document in the module. I can walk you through what is doing if you need. https://github.com/jmwright/cadquery-freecad-module/blob/master/CadQuery/Shared.py#L6

Or are you asking for something built in to the CadQuery module to do this?

jmwright, 2017-04-15:
I'm tempted to take that out anyway. I can't think of any other modules or workbenches that open a new file automatically,  and I'm sure it gets annoying to users after they're familiar with CadQuery. They can always get that template by clicking File->New anyway.

easyw, 2017-04-15:
> I'm tempted to take that out anyway. I can't think of any other modules or workbenches that open a new file automatically, and I'm sure it gets annoying to users after they're familiar with CadQuery. They can always get that template by clicking File->New anyway.

+1 👍  from my side

jmwright, 2017-04-15:
@easyw Try the latest commit in master. It doesn't open the template anymore. I've had one or two other users mention that being an annoyance before. I want to the workbench to fit into the FreeCAD workflow, and always popping up a template breaks that flow.

I need to create a new Getting Started video for the new tabbed interface anyway, so I can just include a mention of File->New in that for people when they want to start writing a script.

easyw, 2017-04-15:
@jmwright that's perfect for me! Thx so much... I made a large use of CQ module and your WB helped a lot KiCAD in having a mechanical library of great STEP models...
https://github.com/KiCad/packages3D
and the library is still growing up 😸

jmwright, 2017-04-15:
@dcowden and I were just talking about adding a section in the documentation about cool things people are doing with CadQuery. We would like to add your KiCAD part generator work to that section, and we are interested in having links between our two projects. Thoughts?

easyw, 2017-04-15:
That would be fine for me :smile: ... I like so much the integration among FreeCAD CadQuery and KiCAD

jmwright, 2017-04-15:
@easyw - @dcowden added your protect to the readme. https://github.com/dcowden/cadquery/blob/master/README.md#projects-using-cadquery

closed 2017-04-15 by jmwright

easyw, 2017-04-19:
#jmwright I added a CQ link to the KiCad official repo README here
https://github.com/KiCad/packages3D
:Smile:

jmwright, 2017-04-19:
Awesome, thank you!