CadQuery/cadquery-freecad-workbench #83: Path to file in same folder as script?

hackscribble, 2017-04-27:
I am trying to read a data file in the same folder as my CQ script when running the script from the CQ workbench in FreeCAD.

In this example, I have the program file, a second file `answers.py` containing a class and a text file `config.yaml` all in the same folder `/home/ray/test/`.

```
import cadquery as cq
from Helpers import show

from answers import Answers

a = Answers()

if "module" in __name__:
    size = a.EVERYTHING
    box = cq.Workplane("XY").box(size, size, size)
    show(box)
    
    file = open('config.yaml', 'r') 
    # file = open('/home/ray/test/config.yaml', 'r') 
    print(file.read())
```

Running this in the workbench, the program finds `answer.py` OK and generates the box correctly.  

However, I get a "file not found" error when it tries to open the config file.  If I use the full path to the config file (see commented out line), it works.

Is there a way to find the path to the original folder when running from the workbench?  I would like to avoid having to use the full path.

Thanks.

jmwright, 2017-04-27:
Have a look at the module_locator code in the FreeCAD module.

https://github.com/jmwright/cadquery-freecad-module/blob/master/module_locator.py

Let me know if that helps.

hackscribble, 2017-04-27:
Added the code from that function to my test program ...

```
import sys
import os
import cadquery as cq
from Helpers import show
from answers import Answers

a = Answers()

if "module" in __name__:
    size = a.EVERYTHING
    box = cq.Workplane("XY").box(size, size, size)
    show(box)

    encoding = sys.getfilesystemencoding()
    print(os.path.dirname(unicode(sys.executable, encoding)))
    print(os.path.dirname(unicode(__file__, encoding)))

    file = open('/home/ray/test/config.yaml', 'r')
    # file = open('config.yaml', 'r')
    print(file.read())
```

The print statements output ...

```
/usr/bin
/tmp
```

Looks like the program file is being copied to the `/tmp` folder.  But then I'm confused as to how it finds the class file but not the config file.

hackscribble, 2017-04-27:
Just spotted that, in the report view after the output from the program, it prints ...

```
Executed /home/ray/test/program.py
```

So I searched through the source code and found this in `Command.py` ...

```
        msg = QtGui.QApplication.translate(
            "cqCodeWidget",
            "Executed ",
            None,
            QtGui.QApplication.UnicodeUTF8)
        FreeCAD.Console.PrintMessage(msg + cqCodePane.file.path + "\r\n")
```
I guess there's no way of the program being executed getting access to that variable?

jmwright, 2017-04-28:
I don't think I can make `cqCodePane.file.path` available to the running script. The program gets written to tmp and executed from there so that you don't have to save your changes before each execution (in case you want to undo), but it's not ideal sometimes. Maybe I can make a temp file in the same directory as the original script if the user has write privileges to the dir.

I'll poke around with this in the morning to see if I can come up with a solution. It's definitely something that needs to be possible though.

As a work-around, can you copy the yaml file to /tmp at the top of your program before using it? Will that work? That's a hackey solution that's only temporary, but it might get you by.

hackscribble, 2017-04-28:
I've tried copying the config file to `/tmp` but same result.

I tried it "as is" after `sudo cp` with owner and group as root, and after changing to my owner and group as per the temp files that are copied.

And I tried `"config.yaml"` and `"./config.yaml"` in the `open` statement.

jmwright, 2017-04-28:
@hackscribble If the module set environment variables that you could access for the tmp file name and the directory you're executing your script from, would that work for you? I'm still trying to figure out the best way to fix this.

```
import os
os.environ.get("MYPROJECT_DIR")
os.environ.get("TMP_FILE")
```

hackscribble, 2017-04-28:
Being able to get `MYPROJECT_DIR` would be great, thanks.

jmwright, 2017-04-28:
@hackscribble Try the latest commit in master. I added two environment variables.

```python
print(os.environ.get("MYSCRIPT_FULL_PATH"))  # Includes full path and filename
print(os.environ.get("MYSCRIPT_DIR")) # Full path excluding the filename
```

However, I still feel that executing a script should work intuitively, without having to add boilerplate code. I've got most of a plan in my head to fix this, but these variables should get you going and might be useful to people in the future.

hackscribble, 2017-04-28:
Brilliant @jmwright thanks for the quick turnround on this :smile: 

Modified test program works as expected:

```
import os
import cadquery as cq
from Helpers import show
from answers import Answers

a = Answers()

if "module" in __name__:

    size = a.EVERYTHING
    box = cq.Workplane("XY").box(size, size, size)
    show(box)

    config = '{path:s}/config.yaml'.format(path=os.environ.get("MYSCRIPT_DIR"))
    file = open(config, 'r')
    print(file.read())
```

closed 2017-04-28 by hackscribble

jmwright, 2017-04-30:
I went ahead and documented these environment variables along with the settings that can be changed for this module.

https://github.com/jmwright/cadquery-freecad-module/wiki/Advanced-Usage

hackscribble, 2017-04-30:
Many than ks @jmwright :smile:

jmwright, 2017-04-30:
No problem. Thanks for using CadQuery.