CadQuery/cadquery-freecad-workbench #84: Execute Scripts in Their Home Directory

jmwright, 2017-04-28:
Related to issue #83 

Because of the need to execute a script without saving it, the text in the editor is currently written to the /tmp directory and then executed from there. This still has to be done when a script has never been saved to any file, but the script should be executed in its home directory when possible. This allows users to include things like yaml configuration files and still access them. The module can check to see if the editor is dirty (has unsaved changes). If the editor is not dirty, the script can be executed directly in its home directory. If it is dirty, a temp file should be created in the script's home directory and executed from there. The only danger here is that if FreeCAD crashes (which still happens occasionally), that temp file won't get cleaned up. This happens with vim quite a bit when using it over a network. When the connection drops and vim is killed, it doesn't get to clean up its temp files. I think this feature is important enough to take that chance though. As projects get more complex, executing from /tmp is going to get more problematic.

jmwright, 2017-08-27:
I'm not sure this is going to be practical with the way the CQGI implementation works. I believe that everything is handled in memory with CQGI so there's no temporary script file to access. @dcowden can you confirm this?

Worst case, users will still have access to the old method (pre-CQGI) of executing their scripts.

dcowden, 2017-08-27:
@jmwright correct, cqgi does not use a temp file for execution.  It would be possible but not easy to generate one.  Cqgi builds an in memory AST, and then uses that structure to inject parameter values and function definitions.  Creating a script would be possible by re-serializing the ast to a file.

jmwright, 2017-08-27:
I'm going to close this issue. Executing scripts in their home directory is not practical with the direction we're headed. However, with CQGI we do need to provide a way to access data in a script's home directory, but I've already added a MYPROJECT_DIR environment variable that addresses this. If that doesn't meet the needs of users over the long-term we'll find another solution.

closed 2017-08-27 by jmwright