CadQuery/cadquery-freecad-workbench #90: Ribbon - a helper module for CadQuery and FreeCAD

luzpaz, 2017-07-17:
@jmwright have you seen this?  
https://github.com/hackscribble/ribbon

jmwright, 2017-07-17:
@luzpaz Thanks for the link, I hadn't seen that.

@hackscribble That's very cool. Would you be interested in having your Ribbon helper included the wiki? I might create a "Helpers" section for things like this. Also, if you're ever interested in integrating your core Ribbon operation into the main CadQuery library, we accept pull requests.

hackscribble, 2017-07-18:
@luzpaz Thanks for the referral :smile:

@jmwright Happy to be mentioned in the wiki.  If you need more examples, images, etc than are in my repo, let me know.  And I'll take a look at the main library to see if I can work out how to add it there.

luzpaz, 2017-07-18:
😄 👍

jmwright, 2017-07-18:
I've added that helper doc to the wiki. @hackscribble I think the images you had are good. I linked them into the doc, and then created a link to your Ribbon repo.

https://github.com/jmwright/cadquery-freecad-module/wiki/Helpers

Thanks to you both!

closed 2017-07-18 by jmwright