CadQuery/cadquery-freecad-workbench #94: 0.17 install problem

alnibel, 2017-08-18:
coss-posted from [https://forum.freecadweb.org/viewtopic.php?f=4&t=23953&sid=332bebf51c0df45dfec746eed74a4fc5](url) :

I've been trying to install the cadquery module without success.

I'm running the freecad daily build on ubuntu 16.04.

I've got the CadQuery folder under ~/.FreeCAD/Mod

The install fails because CadQuery's InitGui.py cannot import the module Command.py.
Here's the log message:

> 
    Log: Traceback (most recent call last):
    File "<string>", line 120, in InitApplications
    File "<string>", line 8, in <module>
    ImportError: No module named Command 

The thing I can't figure out is that Command.py is right where it is supposed to be, at

~/.FreeCAD/Mod/CadQuery/Gui/Command.py

I've also tried installing cadquery via the Addon Manager. Same results.

Btw, the exact same setup _did_ work with FreeCAD 0.16.

jmwright, 2017-08-18:
Interesting, I haven't had that problem with the daily build on 16.04. Are you using master, or one of the releases of this module?

alnibel, 2017-08-18:
I downloaded the zip using the Download button. Says it's cadquery-freecad-module-1.0.0.1.tar.gz

Also, I tried the automated FreeCAD Addon Manager. So whatever version that grabs.

It's weird. The error comes from where InitGui imports Command. For the life of me I can't see that there's anything wrong with the import statements or with the path to Command.py. Yet it fails on the try and then fails fatally on the except.

jmwright, 2017-08-18:
What happens if you open the Python console, import the os module, and then insert the module's root directory path in the Python path before selecting the CadQuery workbench?

https://github.com/jmwright/cadquery-freecad-module/blob/ed0707b150abf4dac19e507acd06a49971844916/Init.py#L14

I haven't used FreeCAD daily since Monday. It's possible that they've introduced a breaking change that causes Command.py to be invalid.

jmwright, 2017-08-18:
Sorry, you want to import the sys module, not the os module.

alnibel, 2017-08-18:
I'm afraid I'm not that up to speed on that sort of under-the-hood python stuff. I'll need more explicit step by step.

You want something like this?
import sys
libs_dir_path = ????  # for example  ~/.FreeCAD/Mod/CadQuery/GUI
sys.path.insert(0, libs_dir_path)

[and then what?]

However that may be, the initialization never gets so far as to make CadQuery available in the Workbench selector. It finds its way into InitGui but not any further than the import statement.

jmwright, 2017-08-19:
This is a tough one to solve since I can't reproduce this problem with FreeCAD daily on my Ubuntu 16.04 system. Since it sounds like FreeCAD is throwing the error on startup, doing the path modification steps I was suggesting won't help you. Please try adding the following line to `CadQuery/Init.py` in the location highlighted in the screenshot.

```
sys.path.insert(0, module_base_path)
```

[image]

Here's my freecad-daily version info.

[image]

alnibel, 2017-08-19:
My FeeCAD version is identical to yours.

I made the edit you suggested. No change in the result. Error in the log file reads the same as I posted above; InitGui can't import Command.

I checked the permissions on Command.py and the Gui folder. They're ok.

alnibel, 2017-08-19:
In the python console, I did this

> import sys
 print(sys.path)

The relevant output was

> u'/home/me/.FreeCAD/Mod/CadQuery', 
u'/home/me/.FreeCAD/Mod/CadQuery/Libs/cadquery-lib', 
u'/home/me/.FreeCAD/Mod/CadQuery/Libs', 
'/home/me/.FreeCAD/Mod/CadQuery', 

That was with your edit of Init.py. If I remove that edit I get

> u'/home/me/.FreeCAD/Mod/CadQuery/Libs', 
u'/home/me/.FreeCAD/Mod/CadQuery/Libs/cadquery-lib', 
'/home/me/.FreeCAD/Mod/CadQuery', 

As an aside, no other workbench shows more than its head entry. For example, Raytracing is only

> u'/usr/lib/freecad-daily/Mod/Raytracing', 

and assembly2 is only

> '/home/e/.FreeCAD/Mod/assembly2',

jmwright, 2017-08-19:
> As an aside, no other workbench shows more than its head entry.

Yeah, the CadQuery module embeds it's dependencies to make it easier to get up and running. That requires me to inject the library paths.

Is there anything non-standard about your Python 2.7 setup?

alnibel, 2017-08-19:
standard python 2.7.12

I just did a purge and re-install of freecad-daily. No change.

jmwright, 2017-08-19:
I'm starting to run out of ideas, but here are a few things to try.

1. Try the latest master: https://github.com/jmwright/cadquery-freecad-module/archive/master.zip
2. Try the previous release: https://github.com/jmwright/cadquery-freecad-module/archive/v1.0.0.zip
3. Try installing in `/usr/lib/freecad/Mod` instead of the Mod directory in your /home
4. Try using the name `cadquery-freecad-module` for the root directory instead of `CadQuery` before copying it into the Mod directory

I'll keep thinking. I'm at a big disadvantage here though since I can't reproduce the problem.

alnibel, 2017-08-19:
Ok. A combination of #1 and #4 worked.

I put the master from your link in 1 above in `~/.FreeCAD/Mod/` as `CadQuery` and it did not work. I tried it again as `cadquery-freecad-module` and it worked.

Thanks for all your help.

Just as an after action recap...

The Installation section of your wiki probably ought to be updated. Likewise, the FreeCAD Sandbox entry. It was references to the `CadQuery` directory in those that led me astray.

Whatever the FreeCAD Addon Manager is doing with cadquery ought to be looked at. It also mis-installed the module.

Thanks again, both for the module and the help.

jmwright, 2017-08-19:
Great, glad it's working for you.

Thanks for the feedback on the docs and on the installer script. I'll work to get those straightened out.

jmwright, 2017-08-20:
I've fixed the documentation and have made a note to evaluate the FreeCAD-AddOns install script as part of the upgrade to full FreeCAD 0.17 compatibility. I'm going to close this issue since the problem is solved and the remaining action item has been captured elsewhere.

closed 2017-08-20 by jmwright