CadQuery/cadquery-freecad-workbench #95: Simple eachpoint example

galou, 2017-08-23:
I tried to understand how eachpoint works and wrote a small example script:
```Python
import cadquery as cq
from Helpers import show

def makeEach(pnt):
    return cq.Workplane('XY').transformed(offset=pnt.toTuple()) \
        .circle(5) \
        .extrude(40)

result = cq.Workplane('XY').box(30, 30, 5) \
    .faces('>Z').workplane().rect(20, 20, forConstruction=True) \
    .vertices().eachpoint(makeEach)

# Render the solid
show(result)
```

This script doesn't work however. I get the error
```
"/home/gael/.FreeCAD/Mod/cadquery_module/Helpers.py", line 40, in show
    if not cqObject.val().label:

'Workplane' object has no attribute 'label'
```

So I changed line 40:
```diff
-    if not cqObject.val().label:
+    if not hasattr(cqObject.val(), 'label') or not cqObject.val().label:
```

but I just get an error somewhere else:
```
"/home/gael/.FreeCAD/Mod/cadquery_module/Libs/cadquery-lib/cadquery/cq.py", line 263, in toFreecad
    return self.objects[0].wrapped

'Workplane' object has no attribute 'wrapped'
```

My conclusion is that I didn't understand how eachpoint works. Could you add an example script for it? Thanks!
```

jmwright, 2017-08-23:
@galou I won't be able to do an example until later, but there's sort of an example of how to use eachpoint embedded in the test suite for CadQuery.

https://github.com/dcowden/cadquery/blob/32c39a60ca51aeaca1b650918e53349ce7669e53/tests/TestCadObjects.py#L72

galou, 2017-08-23:
Thanks for the tip! I made some progresses. Here is a very simple example.
```Python
import cadquery as cq
from Helpers import show

def makeEach(pnt):
    return cq.Solid.makeCylinder(5, 40, pnt)

result = cq.Workplane('XY').box(30, 30, 5) \
    .faces('>Z').workplane().rect(20, 20, forConstruction=True) \
    .vertices().eachpoint(makeEach, useLocalCoordinates=True) \
    .combineSolids()

# Render the solid
show(result)
```

However, I'm clueless how to remove material instead of adding and how to add other material on top of the generated cylinders. Thanks for further hints!

jmwright, 2017-08-24:
I played around a little bit with it and the solution ends up not being very clean and FreeCAD doesn't handle the resulting solid well.

Not sure if this will work for you, but it would be a more idiomatic way to create the same solid.

```python
import cadquery as cq
from Helpers import show

result = cq.Workplane('XY').box(30, 30, 5)\
    .faces(">Z").workplane().rarray(20.0, 20.0, 2, 2, True)\
    .circle(2.0).cutThruAll()

show(result)
```

dcowden, 2017-08-24:
@galou, if you're not already aware of them,  might also  find the official cadquery examples helpful  http://dcowden.github.io/cadquery/examples.html

jmwright, 2017-08-24:
@galou I just realized I didn't recreate your example very closely. I had to move the points in slightly from the edge because FreeCAD was doing some strange things with the holes in the model if I didn't.

```python
import cadquery as cq
from Helpers import show

result = cq.Workplane('XY').box(30, 30, 5)\
    .faces(">Z").workplane().rarray(19.99, 19.99, 2, 2, True)\
    .circle(5.0).cutThruAll()

show(result)
```

jmwright, 2017-08-27:
@galou Any luck on this?

galou, 2017-08-27:
Just to be clear, the example I gave is not what I wanted to achieve. I actually have no project right now that I would like to program with CadQuery. I just wanted to study the possibility of `eachPoint` (or `each`).

I managed to have a more complex shape by copying the way it's done in `cBore`:
```Python
from __future__ import division

import cadquery as cq
from Helpers import show

def make_each(pnt):
    """Make a big hole surrounded by 4 holes"""
    big_diameter = 6
    small_diameter = 3
    small_holes_distance = (big_diameter + small_diameter) / 2 + 2
    # We want a through-all but we cannot know the depth of the object, let's
    # write a big number.
    depth = 10000
    hole_dir = cq.Vector(0, 0, -1)

    # First make the hole.
    cylinders = cq.Solid.makeCylinder(
        big_diameter / 2, depth, pnt, hole_dir)

    # First make the 4 small holes
    pts = [
        (small_holes_distance, 0),
        (0, small_holes_distance),
        (-small_holes_distance, 0),
        (0, -small_holes_distance),
    ]
    for p in pts:
        pt = cq.Vector(p[0], p[1], 0) + pnt
        cyl = cq.Solid.makeCylinder(
            small_diameter / 2, depth, pt, hole_dir)

        # Fuse the cylinders
        cylinders = cylinders.fuse(cyl)
    return cylinders

box_with_holes = cq.Workplane('XY').box(60, 60, 5) \
    .faces('>Z').workplane().rect(30, 30, forConstruction=True) \
    .vertices().cutEach(make_each)

# Render the solid
show(box_with_holes)
```

Unfortunately, I can just remove some material. I guess adding some material would require a function similar to `cutEach` but with `s = s.fuse(cb)` instead of `s = s.cut(cb)`, in cq.py l.75. Ideally I would like to have access to the underlying solid so that I can do both fuse and cut operations inside the callback function.

I guess the reason that `each` doesn't work is because it's not implemented for anything else than vectors https://github.com/jmwright/cadquery-freecad-module/blob/ae27f2d8f3b8a6ed0f0d6a2740fb36eccabb77af/Libs/cadquery-lib/cadquery/cq.py#L1528.

jmwright, 2017-08-27:
@galou Sorry, it usually takes me awhile to understand the core of what someone else is trying to do. I'm probably still not going to address what you're wanting to learn correctly, but I'll give it a shot.

We could add an `extrudeEach` function that I think would do what you're suggesting (adding material the same way cutEach takes it away). For that matter, each major operation could have an *Each corollary (sweepEach, revolveEach, etc) that handles all the underlying merging of objects correctly. There would need to be a discussion on how this fits with the current CadQuery design first, but it could be done.

With the existing CadQuery implementation, this (I think) does what you're asking for with adding material rather than taking it away. Notice the use of combine, eachpoint and union.

```python
from __future__ import division

import cadquery as cq
from Helpers import show

def make_each(pnt):
    """Make a big hole surrounded by 4 holes"""
    big_diameter = 6
    small_diameter = 3
    small_holes_distance = (big_diameter + small_diameter) / 2 + 2
    # We want a through-all but we cannot know the depth of the object, let's
    # write a big number.
    depth = 10000
    hole_dir = cq.Vector(0, 0, 1)

    # First make the hole.
    cylinders = cq.Solid.makeCylinder(
        big_diameter / 2, depth, pnt, hole_dir)

    # First make the 4 small holes
    pts = [
        (small_holes_distance, 0),
        (0, small_holes_distance),
        (-small_holes_distance, 0),
        (0, -small_holes_distance),
    ]
    for p in pts:
        pt = cq.Vector(p[0], p[1], 0) + pnt
        cyl = cq.Solid.makeCylinder(
            small_diameter / 2, depth, pt, hole_dir)

        # Fuse the cylinders
        cylinders = cylinders.fuse(cyl)
    return cylinders

box = cq.Workplane('XY').box(60, 60, 5)#

holes = box.faces('>Z').workplane().rect(30, 30, forConstruction=True) \
    .vertices().eachpoint(make_each).combine() #.cutEach(make_each)

box_with_protrusions = box.union(holes)

# Render the solid
show(box_with_protrusions)
```

A few other things to note if you want to dig deeper into the layers of CadQuery to accomplish things it may not do currently:

1. object.val() gives you access to the underlying CadQuery shape object
2. object.val().wrapped gives you access to the core FreeCAD shape object
3. If you're using the CadQuery module for FreeCAD, you're not limited to chained CadQuery scripting. The module will allow standard FreeCAD scripting and also plain Python to be executed.
4. I'm not sure if this would help in your case, but we have a discussion going on about a direct API for CadQuery that would expose a lower-level layer of CadQuery to help fill in the gaps that the fluent API misses. https://github.com/dcowden/cadquery/issues/167#issuecomment-325197170

Not sure how much all that helps in your studies of the possibilities of eachpoint though.

galou, 2017-08-29:
What I want to achieve is (done with FreeCAD):
[image]

One of my attempts, which doesn't work:
```Python
from __future__ import division

import cadquery as cq
from Helpers import show

def make_each(box, workplane, point):
    extrude_diameter = 10
    extrude_distance = 6
    hole_diameter = 6

    offset = cq.Vector(point[0], point[1], workplane.plane.origin.z)
    
    # First make the extrusion.
    box = box.transformed(offset=offset) \
        .circle(extrude_diameter / 2).extrude(extrude_distance)

    # Then make the hole.
    box = box.transformed(offset=offset) \
        .circle(hole_diameter / 2).cutThruAll()
    return box

box = cq.Workplane('XY').box(100, 100, 5)
workplane = box.faces('>Z').workplane()

side = 30
pts = [
    (side, side),
    (-side, side),
    (-side, -side),
    (side, -side),
]
for p in pts:
    box = make_each(box, workplane, p)

# Render the solid
show(box)
```
An example that could be close to this is `Ex030_Panel_with_Various_Holes_for_Connector_Installation.py` but in this example `workplane(offset=1, centerOption='CenterOfBoundBox')` is used and this works only if the bounding box effectively doesn't change. But still, there is no added material in this example.

jmwright, 2017-08-29:
This is one way to accomplish what I think I see in your screenshot (not done the way you're looking for). I hard coded values to get the scale somewhat right.

```python
import cadquery as cq
from Helpers import show

result = cq.Workplane('XY').box(50, 50, 5)\
    .faces(">Z").workplane().rect(35, 35, forConstruction=True)\
    .vertices().circle(3.0).extrude(2.5)\
    .faces(">Z").workplane().rect(35, 35, forConstruction=True).vertices()\
    .circle(2.0).cutThruAll()
    
show(result)
```

[image]

However, it doesn't rely on the functional approach (creating a new operation) that you seem to be going for.

The following code gets us close to the new operation, but getting the holes through the plate is the hard part. We almost need a new type of boolean operation to combine the solids the right way.

```python
import cadquery as cq
from Helpers import show

def makeStuds(pnt):
    holes = cq.Solid.makeCylinder(4, 15, pnt)
    return cq.Solid.makeCylinder(5, 10, pnt).cut(holes)

result = cq.Workplane('XY').box(50, 50, 5)\
    .faces("<Z").workplane(invert=True).rect(35, 35, forConstruction=True)\
    .vertices().eachpoint(makeStuds, useLocalCoordinates=True).combineSolids()
    
show(result)
```

I talked to Dave about this a little bit, and there doesn't seem to be a solution readily available with the current implementation of CadQuery. Creating the boss and the hole in one operation is beyond my expertise at this point, but I'll keep thinking about it.

galou, 2017-09-06:
Thanks again for having studied the problem a bit deeper. I'd be glad to participate in the development of such a function but I unfortunately don't have much free time. I guess that anyway it makes little sense to develop this in CadQuery 1.0 and that it's better to wait for CadQuery 2.0.

closed 2017-09-06 by galou

jmwright, 2017-09-07:
The direct API discussion that's going on will probably make this idea a moot point eventually, but I've been thinking about what you're trying to do. There could be an extrudeEach corollary to cutEach, but there could also maybe be a compoundEach function that would do both extrusion and cutting. 

You would be able to create something like a makeBoss function that would pass the function that creates the objects to cut first, and also the function to create the objects to extrude [here](https://github.com/dcowden/cadquery/blob/e3d8bacf9519fc575f9b00daa99e75c62b7d0db4/cadquery/cq.py#L1821). The call might look something like this:

```python
return self.compoundEach(_cutObj, _extrudeObj, True, clean)
```

The results object [here](https://github.com/dcowden/cadquery/blob/e3d8bacf9519fc575f9b00daa99e75c62b7d0db4/cadquery/cq.py#L1762 ) could be iterated over twice. Once for the cuts and once for the extrusions.

It's not as clean as allowing custom operations, but I think it might work for this use case.