CadQuery/cadquery-freecad-workbench #96: Implement Custom Code Editor

jmwright, 2017-09-02:
The code editor that CadQuery relies on (pyCode) is not working properly with all versions of Qt5/Pyside2 and has been put into maintenance mode by it's author. While I could fork and maintain the repos myself, at this point I think we're better off in the long-term to implement our own editor. The custom editor will be tailored to CadQuery Python syntax, and will give us more flexibility when implementing new features in the future. It will also be lighter weight than pyQode.

Here are some features that I think the editor needs to have starting out.

* Line numbers
* Syntax highlighting
* Proper tabbing
* Syntax checking (check for errors and PEP8 compliance)
* Automatic code completion (that finally works properly)

People are welcome to request other features on this thread or in the CadQuery Google group.

adam-urbanczyk, 2017-09-08:
Take a look at Spyder (MIT license), e.g. here:

https://github.com/spyder-ide/spyder/blob/master/spyder/widgets/editor.py#L2781

jmwright, 2017-09-08:
That's cool, thanks. I didn't even think about just using the editor from Spyder. I'll experiment with that to see if it will work for us.

jmwright, 2018-05-02:
The PyQode editor isn't supported anymore and is archived, so I made a copy of it in this module's `Libs` directory and have started maintaining it separately. Creating an editor from scratch is a huge task, and one I don't think will ever get done with the resources this project has. I think we're better off maintaining PyQode locally ourselves.

I'm closing this since it's more of a wontfix now.

closed 2018-05-02 by jmwright