CadQuery/cadquery-freecad-workbench #97: Implement CQ 2.0 Support

jmwright, 2017-09-07:
@adam-urbanczyk has a PythonOCC-based CadQuery 2.0 implementation in progress.

https://github.com/adam-urbanczyk/cadquery

If possible, this module needs to support CQ 2.0 and allow interchange between FreeCAD and PythonOCC objects to maximize the usefulness of CQ 2.0 and make it easy for people to start working with it.

adam-urbanczyk, 2017-09-08:
Yes I fully agree. I see two options:

1. Export to BREP and load in FreeCAD 
2. Use the  __fromPythonOCC__ method in FreeCAD, but then we need to care about OCC/OCE version compatibility in PythonOCC and FreeCAD

jmwright, 2017-09-08:
@adam-urbanczyk Do you know how the BREP import-export process would work between PythonOCC and FreeCAD? I've never tried that before.

adam-urbanczyk, 2017-09-09:
I just implemented exportBrep in my branch. You can try the following code:
```
box = Workplane('XY').box(1,1,1).val().exportBrep('tmp.brep')
```

and on the FreeCAD side:

```
s = Part.Shape()
s.importBrep('tmp.brep')
Part.show(s)
```

jmwright, 2017-09-09:
Excellent, thanks.

luzpaz, 2018-05-04:
Any progress on this ?

dcowden, 2018-05-04:
I'm not sure about the progress. But @jmwright  and I discussed this the other day. We're currently thinking it may be best to try to move CQ 1.x to OCC, which could be done slowly vs in a 'big bang'

closed 2018-12-29 by jmwright