CadQuery/cadquery-freecad-workbench #98: Can't use Arrow keys in PyCodeEdit while modifier other than shift is held down.

rakete, 2017-09-25:
I have a very minor problem which is also very specific to my workflow. I have Capslock bound to AltGr, and then make my own keyboard layout where I bind AltGr+j,k,l,i to the arrow keys so that I can navigate around without leaving the home row.

This works fine everywhere, but not in in the PyCodeEdit panel opened by this module. The problem is probably quite simply that when I use my custom keybindings (e.g. I press Capslock + i together to send the 'up' keystroke) that those register as arrow key together with a modifier key, and since there is no keybinding that handles those two together, nothing happens.

I have worked around this issue by making a custom mode which I add to PyCodeEdit, but I assume that there is probably a 'better' way which I have not figured out. The code for this workaround is here: https://github.com/rakete/cadquery-freecad-module/blob/c15df3c31099ace72fddf28f923148d213abc8d7/Gui/ImportCQ.py#L13

I've already seen that you are planning on replacing PyCodeEdit with something else. So this problem will probably be solved then and further work on this is not necessary. I still wanted to add this issue since, well, it was an issue for me... feel free to close it if you want.

jmwright, 2017-09-25:
@rakete Thanks for the feedback. I had to look up what an AltGr key is, but I think I understand what you're suggesting well enough now to ask follow-up questions.

When you say that your key setup "works fine everywhere", does that include other Qt based apps? For instance, will your key setup work in the macro editor in FreeCAD? If so, maybe it's just a matter of fixing modifier key support in this module's editor. Could you go to `Macro->Macros...` and then click the `Create` button in the macro dialog and try editing a script using your keyboard layout and modifier key?

As a side note, PyQode modes are not likely to be supported in the in the new editor. Even so, I will do my best not to limit users in the way they use this editor.

rakete, 2017-09-26:
With "works fine everywhere" I meant: works fine in other applications. I tried my keybindings in QtCreator where they worked fine, but I don't really use that many Qt applications so there might be others that have the same problem.

I also just tried creating a macro, and I have the same problem within the freecad macro editor. With one subtle difference: there was one keybinding that worked in the cadquery editor (AltGr+Home), which does not work in the macro editor.

My workaround works well enough for the moment and that it won't be supported when you replace PyCodeEdit is not a problem. 

When you replace PyCodeEdit with something else I can test again and report back if the problem is still present. Although if it does not work in the macro editor then that probably means the issue is probably caused by freecad and the Qt "around" the cadquery-freecad-module (like maybe an event filter in the mdi panel into which PyCodeEdit is inserted)?

jmwright, 2017-09-26:
> I also just tried creating a macro, and I have the same problem within the freecad macro editor.

That does probably mean that the keybindings aren't working correctly because of FreeCAD.

If you wanted to turn your workaround into something that could be turned on and off using settings, I'd be happy to accept a PR. I'd prefer for it to default to "off" though to lessen the risk of changing the current behavior of the module.

https://github.com/rakete/cadquery-freecad-module/blob/c15df3c31099ace72fddf28f923148d213abc8d7/Settings.py

It will probably be 2 to 3 months before I get the new editor done, and one of the development guides I use will be that Settings.py file. That will help keep me from forgetting useful settings like alternative keybinding support. I can also do a minor version release in the meantime to get the new keybinding setting out to users.

jmwright, 2018-12-24:
@rakete If you're still using this workbench, I'm working on implementing a custom code editor on the [kill-pyqode branch](https://github.com/jmwright/cadquery-freecad-module/tree/kill-pyqode). If you're interested, please try that editor out to see if your custom keymapping works. If it doesn't I'll look at trying to include your custom code in the new editor.

closed 2018-12-29 by jmwright