CadQuery/cadquery-freecad-workbench #100: Wrong script is executed when there are duplicate filenames in MDI list

kontrollerX, 2017-10-03:
I have 2 tabs (documents) in my FreeCAD MDI window.
The filenames are both default "sample_template.py" script files.
On the first document I only have the python statement 
     print "1"
On the second,
    print "2"
When I execute the first script as the active tab (document), "2" is printed in the report view. And vice versa.

jmwright, 2017-10-03:
HI @kontrollerX thanks for reporting this. Can you tell me a little bit about the use case associated with this? Is it something that your noticed, or do you need to use this FreeCAD module that way? My first instinct was to try to find an ID for each MDISubWindow instead of using the titles, but that information isn't available. I'll have to figure out another way to uniquely identify the subwindows.

kontrollerX, 2017-10-03:
I  was experimenting with CadQuery under the default filename and I just clicked CadQuery "New Script" to create a second MDI window to compare almost identical code ( instead of modifying code in the first window), when this issue came up for me.

The quick and simple fix for the me, subsequently, was to rename a new file immediately after CadQuery created it. It's not a major issue but I was confused about my code for a few moments until i realised what was going on.

The simplest fix in the cadquery-freecad-module code is to generate new filenames with an incrementing numeric suffix like "sample_template_1.py" etc. as CadQuery creates them.... kinda like Notepad++ does.

But then again "sample_template_1.py" implies there are different types of sample templates. 

Perhaps put a warning message in the template to rename and save the file upon creation ?

jmwright, 2017-10-03:
> Perhaps put a warning message in the template to rename and save the file upon creation ?

That should work. You should already get a message if you try to save over the template, so adding a message after opening it asking you to save the template under another name shouldn't be a problem.

jmwright, 2017-10-04:
This has been addressed in commit 98da3e9c9bc8ee8395dcf3efa0018c17e096956a

When the user creates a new script they're asked to save it under another name before moving on.

closed 2017-10-04 by jmwright