CadQuery/cadquery-freecad-workbench #102: Imported modules aren't re-imported when script is re-run

fragmuffin, 2017-10-08:
When importing a module, FreeCAD needs to be re-started to successfully re-import

**Steps to re-create**
1) Create a local file, like `~/temp/my_module/__init__.py` with contents...
```python
import cadquery
from Helpers import show

def foo():
    box = cadquery.Workplane('XY').box(10, 10, 10)
    show(box)
```

2) start open FreeCAD, and a new cadquery script, copy this in

```python
import sys
sys.path.append('/home/nymphii/temp')
from my_module import foo
foo()
```

3) run it, you should be shown a cube (10x10x10)
4) keeping FreeCAD open, change the dimensions of the box in `my_module/__init__.py` file
5) re-run the script (hotkey `F2`) in FreeCAD

**Expectation**
The box should show the _new_ dimensions

**Actual Results**
The box is still in the _old_ dimensions

**Work-around**
Close FreeCAD, then re-open it

jmwright, 2017-10-08:
@fragmuffin Can you try the latest master, using `show_object()` instead of `show()`? You'll have to do a recursive clone because I've made the cadquery lib a git submodule now.

show_object() uses the new CadQuery Gateway Interface (CQGI). I'm curious if it handles the imports correctly.

fragmuffin, 2017-10-08:
@jmwright done, and the same issue.
The problem is unrelated to `cadquery`.
If, instead, `my_module` doesn't do anything CAD related, the problem persists.

for example, if `~/temp/my_module/__init__.py` instead contains...

```python
import os

def foo():
    locally_scoped_str = 'testing 123'
    with open(os.path.join(os.environ['HOME'], 'temp', 'test.txt')) as fh:
        fh.write(locally_scoped_str)

```

the same error can be observed when changing the value of `locally_scoped_str`, saving `__init__.py`, then re-running with `F2`

jmwright, 2017-10-08:
CQGI does the execution environment much differently than the module does. I was hoping that the issue was in the environment setup and tear down. I'll look into this issue. Thanks for reporting it.

fragmuffin, 2017-10-08:
@jmwright this may be the culprit
https://github.com/jmwright/cadquery-freecad-module/blob/98da3e9c9bc8ee8395dcf3efa0018c17e096956a/Gui/Command.py#L200:L201
```python
# We import this way because using execfile() causes non-standard script execution in some situations
imp.load_source('temp_module', tempFile.name)
```
where `tempFile` contains the `cadquery` script being run.

----

yep, I just used that method manually, same issue.
So the question is how to "unload" a loaded module.

jmwright, 2017-10-08:
That section of code is bypassed when you use show_object() though. Did you try show_object after a fresh start of FreeCAD? CQGI sets up a separate environment each time it runs so I had expected that would fix this issue. If you ran the script with show and then immediately with show_object, the section of code you linked might still cause problems.

fragmuffin, 2017-10-08:
oh!, I see how that works now, so sorry.
I changed `foo` to return the `box`... then used `show_object` instead of `show.
same issue.

fragmuffin, 2017-10-08:
@jmwright 

with the `imp.load_source` code at fault, a solution could be.

```python
from contextlib import contextmanager
import sys

@contextmanager
def revert_sys_modules():
    modules_before = set(sys.modules.keys())
    yield
    for mod_name in sys.modules.keys():
        if mod_name not in modules_before:
            del sys.modules[mod_name]

with revert_sys_modules():
    imp.load_source('temp_module', tempFile.name)
```

I haven't tested it yet, but I've reeealy got to get to sleep right now :zzz:
I'll be busy for the next few days, but when I have time I can test it out and make a PR if it works

jmwright, 2017-10-08:
> I'll be busy for the next few days, but when I have time I can test it out and make a PR if it works

That would be awesome, thanks for all your work on this.

fragmuffin, 2017-10-09:
My pleasure @jmwright !
I'm enjoying `cadquery` a lot, it's a CAD solution I've been looking for for a very long time, so I'm happy to contribute... and I'm very happy that I _can_ contribute :)

jmwright, 2017-10-09:
@fragmuffin We're really glad that you've gotten involved. You've brought a lot to the project, and your enthusiasm is greatly appreciated. @dcowden and I are really busy with day jobs, so if we're slow to respond or miss things once in awhile, please don't take it as a sign of disinterest. We're very excited about where CadQuery is headed, and it's very encouraging when other people believe in it too and jump in with us.

dcowden, 2017-10-09:
@fragmuffin @jmwright not to just ditto Jeremy, but.. ditto!

It's so cool to have other people helping.

jmwright, 2017-10-13:
I haven't closed this because there might have been a bug introduced with this code change.

https://github.com/jmwright/cadquery-freecad-module/pull/103#issuecomment-335433577

fragmuffin, 2017-10-14:
@jmwright I've created #104 with details on how to reproduce the bug, so this issue may be closed.

closed 2017-10-14 by jmwright