CadQuery/cadquery-freecad-workbench #105: File names with '-' cause problems

Grawp, 2017-10-20:
Create file named e.g. `Ex001-Simple-Block.py`. Open it.
- When you execute the script while the tab with code is opened, new shape is added to existing one(s) instead of replacing the original.
- When you execute the script while the tab with model is opened, exception is risen:
> Running the Python command 'CadQueryExecuteScript' failed:
> Traceback (most recent call last):
>   File "/home/michal/.FreeCAD/Mod/cadquery-freecad-module/Gui/Command.py", line 153, in Activated
>     scriptText = cqCodePane.toPlainText().encode('utf-8')
> 
> 'NoneType' object has no attribute 'toPlainText'

jmwright, 2017-10-20:
@Grawp Thanks for reporting this.

This problem stems from the fact that this module has to stay away from certain characters in FreeCAD document names for the 3D view. See this issue [here](https://github.com/jmwright/cadquery-freecad-module/issues/75).

I think the only way to fix this is to translate characters like '-' into other allowed characters (that would never be in a filename), and then swap those characters back when executing. I can also maybe add a message when a file is opened that it's filename contains disallowed characters that will cause problems.

jmwright, 2017-10-21:
I've addressed this in commit 760780475bd6b150b2abc803401b2872fef53743

The best I can do is replace the dash characters in the 3D view's document name. I chose two underscores '__'  since I don't expect anybody to directly try to execute an \_\_init\_\_.py file. This way the filename can contain dashes, and the 3D view will still work, albeit with double underscores in the name instead of dashes.

closed 2017-10-21 by jmwright