CadQuery/cadquery-freecad-workbench #107: Workbench not apearing in FC 0.16

Detzi88, 2017-10-25:
Hi,
i just tried to install the ''cadquery_module'' with the ''adons_installer.FCMarcro'' from within
OS: Windows 7
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.16.6712 (Git)
Build type: Release
Branch: releases/FreeCAD-0-16
Hash: da2d364457257a7a8c6fb2137cea12c45becd71a
Python version: 2.7.8
Qt version: 4.8.7
Coin version: 4.0.0a
OCC version: 6.8.0.oce-0.17

It says installed but after a restart the workbench does not appear in the Dropdown Menue. Tried several times, ohther workbenches are installing fine. The Module Data are there also, but the workbench is not apearing.

Detzi

jmwright, 2017-10-25:
@Detzi88 Sorry for the inconvenience. I've switched the CadQuery library to being a git submodule, and it's causing problems with the add-ons installer. I've got a PR open on the FreeCAD-addons repo to fix this.

https://github.com/FreeCAD/FreeCAD-addons/pull/52

Until then, please install the workbench manually by downloading the zip file from GitHub.

jmwright, 2017-10-26:
@Detzi88 Were you able to try the manual install and did it work?

Detzi88, 2017-10-27:
@jmwright 
Hi i tried it and it did not work either.

jmwright, 2017-10-27:
Ok, there are a couple options then.

1. You can install and use my version of the macro from the PR on FreeCAD-addons. https://github.com/FreeCAD/FreeCAD-addons/pull/52/files#diff-63d64650154b573102f30319d1f519d2
2. I can create a release zip file that has everything in it.

jmwright, 2017-10-28:
@Detzi88 Please try [this release](https://github.com/jmwright/cadquery-freecad-module/releases/tag/v1.0.0.2) and reopen this issue if you still have problems.

closed 2017-10-28 by jmwright