CadQuery/cadquery-freecad-workbench #108: submodule cadquery not loaded by addons macro

danwarren, 2017-11-08:
I'm running the 0.15 FreeCAD from the repo of Ubuntu 16.04.2LTS on arm64.

After watching the errors on a terminal I was able to manually clone the cadquery repository as referenced on the git page, but I would expect it to be handled by the installer.

The module failed with a nearly invisible error on the splash screen, and was otherwise completely missing.

jmwright, 2017-11-08:
@danwarren Thanks for the information. Assuming the installer you're referencing is the FreeCAD-addons macro, I've got a pull request active to fix this. In the meantime, a recursive clone or manually installing the latest release is recommended.

https://github.com/jmwright/cadquery-freecad-module/releases/tag/v1.0.0.2

jmwright, 2017-11-08:
@danwarren As a side note, there are PPAs for newer versions of FreeCAD. I run both stable and daily on my 16.04 system.

Stable (0.16): https://launchpad.net/~freecad-maintainers/+archive/ubuntu/freecad-stable

Daily (0.17): https://launchpad.net/~freecad-maintainers/+archive/ubuntu/freecad-daily

jmwright, 2017-11-08:
Sorry, missed the ARM64 qualifier. I don't know if those PPAs will work for that platform.

amca01, 2017-11-08:
I'm also having difficulties: I've tried downloading cadquery both from the archlinux repo, and by using the addon macro from with FreeCAD itself.  And each time when I restart FreeCAD I get the message:

`During initialization the error No module named cadquery occurred in /home/me/.FreeCAD/Mod/cadquery_module/InitGui.py`

or  

`During initialization the error No module named cadquery occurred in /usr/lib/freecad/Mod/cadquery_module/InitGui.py`

What do I need to do to convince FreeCAD that the cadquery module is there and can be used?

jmwright, 2017-11-08:
@amca01 Are you installing this from AUR? https://aur.archlinux.org/packages/freecad-cadquery-git/

jmwright, 2017-11-08:
@danwarren @amca01 Please try the AUR package and the FreeCAD-addons installer again and let me know if it works.

danwarren, 2017-11-10:
@jmwright, looks like they are shipping nightlys for arm64
https://launchpad.net/~freecad-maintainers/+archive/ubuntu/freecad-daily/+build/13703477

Since the issue was encountered while using the macro installer, would upgrading to daily freecad fix this issue, or otherwise improve the integration with the module ? ( I only installed freecad to run cadquery)

How about modifying the module readme to reference the working pull request version of the macro installer instead of directing us to the currently non-working release version? (I totally get the bad vibes/unnecessary fork aspect of this, but as it stands following the instructions on the module github, at least in some environments, appears to silently fail)

jmwright, 2017-11-10:
@danwarren 
>Since the issue was encountered while using the macro installer, would upgrading to daily freecad fix this issue, or otherwise improve the integration with the module ?

No, the change that's needed hasn't been merged for even the nightly builds yet. 

The reversion that I did should put the module back in line with the existing documentation.  I may try the breaking submodule change again once the PR is merged and makes it into the 0.17 stable release. That will be a long time to wait though. For now, master should be working with the stock addons macro and the AUR build script again.

danwarren, 2017-11-10:
Ok, just uninstalled/reinstalled with the macro, working fine, thanks.

closed 2017-11-10 by danwarren

jmwright, 2017-11-10:
@danwarren Glad to hear that it's working. Sorry for the inconvenience. I was trying to make this module easier to maintain, but it caused a lot more trouble than expected for users.