CadQuery/cadquery-freecad-workbench #111: Examples don't run as intended

linuxcf, 2017-12-13:
I installed CADQuery on my FreeCAD17, running on stock 64-bit Ubuntu 16.04.3LTS with the latest 4.10 Ubuntu kernel, with the FreeCAD add-on manager.  When I ran examples 4 and 6, the resulting solids were the complements of the intended ones.  I have attached pertinent files.  

Also, in Example 8, the regular hexagons are only cut through half the thickness of the slab, no matter what I do.  I even used 'CutThruAll()' and 'CutBlind()' with explicit arguments, but the end result was still the same.

All the examples I checked had 'from Helpers import show' missing.

[Ex004.pdf](https://github.com/jmwright/cadquery-freecad-module/files/1554427/Ex004.pdf)
[Ex006.pdf](https://github.com/jmwright/cadquery-freecad-module/files/1554428/Ex006.pdf)

jmwright, 2017-12-13:
I'm getting the same results running 0.17 from the FreeCAD daily ppa in Ubuntu 17.10.
```
OS: Ubuntu 17.10
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.17.12802 (Git)
Build type: None
Branch: master
Hash: 91bb7ed0c51ba47f011199af7bc0a3a2964cf5be
Python version: 2.7.14
Qt version: 4.8.7
Coin version: 4.0.0a
OCC version: 7.1.0
Locale: English/UnitedStates (en_US)
```
I wonder if this is related to the wire sorting problem in CadQuery issues [173](https://github.com/dcowden/cadquery/issues/173#issuecomment-349772375) and [83](https://github.com/dcowden/cadquery/issues/83). @RustyVermeer do you have any insight on the two examples above (4 and 6) based on your current work on wire/edge sorting for DXF import?

@linuxcf  On example 8, the reason the hexagons will only cut from the center of the box is that the box call centers the box in the origin workplane. Then when the hexagons are drawn, they're drawn on the origin workplane too. You can see where the origin workplane is by clicking View->Toggle axis cross. That will put the origin marker in the view so that you can see where your part is in relation to the origin and origin workplane.

In order to get the behavior you want there are several options, but here are a couple.
```python
# Selects the face in the minimum Z direction, establishes a workplane, and puts the hexagons on that.
result = cq.Workplane("front").box(width, height, thickness) \
                              .faces('<Z').workplane() \
                              .pushPoints([(0, 0.75), (0, -0.75)]) \
                              .polygon(polygon_sides, polygon_dia) \
                              .cutThruAll()
```
```python
# Extrude a rectangle instead of using a box so that the resulting solid sits flat on the origin workplane instead of the workplane being in the middle of the box.
result = cq.Workplane("front").rect(width, height).extrude(thickness) \
                              .faces('<Z').workplane() \
                              .pushPoints([(0, 0.75), (0, -0.75)]) \
                              .polygon(polygon_sides, polygon_dia) \
                              .cutThruAll()
```
There are several other ways to do it too including a [centered](https://github.com/dcowden/cadquery/blob/152da2007707861736795deaf7d5c549707ba769/tests/TestCQSelectors.py#L219) argument for the box function that will control whether the box sits on "top" of the origin workplane, or is embedded down in it.

> All the examples I checked had 'from Helpers import show' missing.

The latest version of this workbench switched over to the CadQuery Gateway Interface (CQGI). It's a way for us to make scripts execute in a standard way across CadQuery environments. You can still use `Helpers.show` if you want, but to make your scripts portable you should use `show_object`. In the end, `show_object` calls `Helpers.show`, so you're getting the same end result. `show_object` just does some other things to add features (dynamic variable editing) and make the script portable. If the documentation is still referencing `Helpers.show` anywhere please let me know. Those need to be updated.

RustyVermeer, 2017-12-15:
Ok, so updating to FreeCAD 0.17 actually broke my DXF stuff, sadly.
However, I started poking around with example 4 in FreeCAD 0.17. I reproduced the same error as is shown above, and started breaking things down a bit. 

The code of interest in CadQuery is [here](https://github.com/dcowden/cadquery/blob/c1d3a3deef418ac0491458768774424f5c5d02e2/cadquery/freecad_impl/shapes.py#L878) in the **linearExtrude** function

I wrote some code to mimic that function and find out what's going on.
In FreeCAD 0.16, the Face built from the 2 wires is shown to be valid. 
In FreeCAD 0.17, this is not true.
However, Part.Face has a method called **validate** that does ... _something_ and then causes the face to show as valid.

The code below can be pasted into the FreeCAD 0.17 CadQuery editor and _should_ work.
If you comment out the line

`f.validate()`

you should see the incorrect behavior, like before.

If this works on other machines, and not just my own, I think we're on the right track for a solution.

```
# This example is meant to be used from within the CadQuery module of FreeCAD.
import cadquery as cq
import Part
from Helpers import show

from cadquery.freecad_impl.shapes import Face

# The dimensions of the model. These can be modified rather than changing the
# object's code directly.
circle_radius = 30.0
rectangle_width = 13.0
rectangle_length = 19.0
thickness = 13.0

# Extrude a cylindrical plate with a rectangular hole in the middle of it
result = (cq.Workplane("front")
                                    .circle(circle_radius) 
                                    .rect(rectangle_width, rectangle_length)
                                    # .extrude(thickness)
         )

eDir = result.plane.zDir.multiply(thickness)
wireSets = result.ctx.pendingWires
fwire = []
for wire in wireSets:
    w = wire.wrapped
    fwire.append(w)
    
f = Part.Face(fwire)
print "face is valid? " + str(f.isValid())
f.validate()
print "face is valid? " + str(f.isValid())
thisObj = f.extrude(eDir.wrapped)
thisObjcast = Shape.cast(thisObj)
r = cq.CQ(Compound.makeCompound([thisObjcast]))

print '\n'
# Render the solid
show(r)
```

dcowden, 2017-12-15:
I Definitely don't have a problem with that, I'm sure validate can't hurt

jmwright, 2017-12-15:
@RustyVermeer Thanks a lot for digging into this. Your solution looks good to me too.

BTW - I had to add the following line to get the example to run on my system.
```python
from cadquery import Shape, Compound
```

RustyVermeer, 2017-12-15:
sorry, yea. I did too, actually. I 'cleaned up' the code before pasting here. Must've gone too far :)
I edited my comment above to make that change

RustyVermeer, 2017-12-15:
Where is the appropriate place to make this change, in @dcowden 's CadQuery repo or here?

jmwright, 2017-12-15:
@RustyVermeer The main [CadQuery repo](https://github.com/dcowden/cadquery) is the correct place to make this change. It's a long story, but I'm currently not able to have the CadQuery lib set up so that I can push changes back to that repo from here.

RustyVermeer, 2017-12-15:
Made and merged that change.

I hope that was all done correctly

jmwright, 2017-12-15:
If it causes problems we could try to add an isValid() check before calling validate(), but I don't see any problems so far. This change has been integrated into the workbench. Thanks @RustyVermeer !

jmwright, 2017-12-15:
@linuxcf Please do an update through the add-on manager. Feel free to reopen this issue if you have any other problems.

closed 2017-12-15 by jmwright

linuxcf, 2017-12-15:
**@jmwright**  Thank you.  I can confirm that after the update, CADquery's examples 4 and 6 now work as intended on my FC17.

dcowden, 2017-12-16:
Thanks for making this change guys! It's been a nagging issue for a while, it's great to finally have it fixed.