CadQuery/cadquery-freecad-workbench #112: Actually Implement CQGI's show_object Funtion

jmwright, 2018-01-05:
Right now I just check to see if `show_object` is present in the text of a script to determine if it's CQGI-compliant or not. @RustyVermeer has a [better implementation](https://github.com/RustyVermeer/cqnb/blob/f0763b353586fe809ce5a3c9049254161ce4c80f/cqjupyter.py#L166) in cqnb (CadQuery Jupyter Notebook plugin) that I should switch to when I get a chance.

RustyVermeer, 2018-01-05:
@jmwright , It's good to have an implementation enhancement, of course. I'd caution using my code as a strict example, because I am not 100% sure that I've implemented show_object as intended.

What I mean is that since I simply defined my own show_object function and have it imported directly into jupyter's namespace, I circumvent [cadquery's show object code](https://github.com/dcowden/cadquery/blob/bfb380ac20e5286a4daa576c15d85becbe6bccb5/cadquery/cqgi.py#L301)

If functionality is added to the cqgi code there in future, my implementation does not account for that. So, I'm just unsure if what I've done is technically correct.

jmwright, 2018-01-05:
@RustyVermeer According to the CQGI specification, what you've done is correct. There's no `show_object` function defined anywhere in the CadQuery codebase, it's left up to the environment to do that. This workbench accounts for debug objects as well, so my implementation will most likely be an amalgamation of yours and the code I have now.

@dcowden Am I correct about what the specification says about `show _object`?

dcowden, 2018-01-05:
The intended  use for show_object is that the script calls it, but the executing environment does not need to define it. Here's the 'normative' way to execute cq scripts using cqgi:

`from cadquery import cqgi

user_script = ...
build_result = cqgi.parse(user_script).build(build_parameters={ 'param': 2 }, build_options={} )

objects = build_result.results # a list of shapeResult objects
debugObjects = build_result.debugObjects #a list of shaperesult objects

display_a_shape_on_the_screen( objects[0].shape)
send_results_to_client(cq.exportShape(objects[0].shape)
`

In short, show_object is designed to be controlled by cqgi, and always available to calling scripts when run via cqgi.  The executing environment need not define this method-- they should retreive the results as above.

ShapeResult objects have a shape property and the user parameters used to build it-- at that point, the executing environment should use these shapes to display them, probably using cadquery export or whatever

There is more documentation [here](https://dcowden.github.io/cadquery/cqgi.html)

dcowden, 2018-01-05:
..as a side note, the reason you probably didnt find show_object is because its not defined classically. I'm using AST processing in CQGI to find and allow mutating variable values that are hard-coded in the script.  The show_object method is defined [here](https://github.com/dcowden/cadquery/blob/bfb380ac20e5286a4daa576c15d85becbe6bccb5/cadquery/cqgi.py#L301)

and then linked into the executing environment's namespace [here](https://github.com/dcowden/cadquery/blob/bfb380ac20e5286a4daa576c15d85becbe6bccb5/cadquery/cqgi.py#L104)

jmwright, 2018-01-05:
Yeah, it seems like I misunderstood the specification and the code. Thanks for the clarification.

jmwright, 2018-06-12:
There are changes coming to CQGI that will render this issue moot. Closing as #wontfix

closed 2018-06-12 by jmwright