CadQuery/cadquery-freecad-workbench #114: CQ is not working on FC 0.17 13362

easyw, 2018-03-09:
Hi, because 
PyQt4 has been abandoned, CQ doesn't load correctly...

```
Log: Init:      Initializing C:\Users\mau\AppData\Roaming\FreeCAD\Mod\cadquery_module... failed
Log: ----------------------------------------------------------------------------------------------------
Log: Traceback (most recent call last):
  File "<string>", line 120, in InitApplications
  File "<string>", line 9, in <module>
  File "C:\Users\mau\AppData\Roaming\FreeCAD\Mod\cadquery_module\CadQuery_rc.py", line 9, in <module>
    from PyQt4 import QtCore
ImportError: No module named PyQt4

```
this can solve the issue

`from PySide import QtCore`

but it shoould be used an earlier Resource Compiler for Qt

jmwright, 2018-03-10:
> but it shoould be used an earlier Resource Compiler for Qt

@easyw Could you clarify what you mean by that? Does the resource file have to be compiled with an earlier version of the Qt compiler to work with FreeCAD 0.17?

> Just an other minor issue... it would be possible to change the shortcut to run CQ script from F2 to an other one?

You can change it manually on [this line](https://github.com/jmwright/cadquery-freecad-module/blob/master/CQGui/Command.py#L139) and then restart FreeCAD. Then the new shortcut key that you set should be in effect. Let me know if that doesn't work for you. I can also add a setting for it as well.

easyw, 2018-03-10:
@jmwright 

> Does the resource file have to be compiled with an earlier version of the Qt compiler to work with FreeCAD 0.17?

The issue could re-appear if the resource would be re-compiled, because the compiler will automatically add this line:
`from PyQt4 import QtCore`

```
# Resource object code
#
# Created by: The Resource Compiler for PyQt4 (Qt v4.8.7)
#
# WARNING! All changes made in this file will be lost!

```
You probably would find useful to switch to 'pyside-rcc.exe' to build the resources
This can be found in:
FreeCADLibs_11.5.3_x64_VC12\bin
https://github.com/sgrogan/FreeCAD/releases/tag/0.17-med-test

> You can change it manually on this line and then restart FreeCAD. Then the new shortcut key that you set should be in effect. Let me know if that doesn't work for you. I can also add a setting for it as well.

Having the shortcut configurable would be very nice! 😄 
Maurice

jmwright, 2018-03-11:
@easyw Please try the latest commit and let me know if it works for you. c23e237bade857ba69a8130d8bdc6d2400e780c6

I also added an `execute_keybinding` item to Settings.py so you could change the key to execute the script from the default of F2.

easyw, 2018-03-11:
@jmwright 
this is working fine now! 
Thx a lot also for keybinding 😺

closed 2018-03-11 by easyw