CadQuery/cadquery-freecad-workbench #115: Add blurb and latest features to FC 0.17 release page

luzpaz, 2018-03-12:
https://www.freecadweb.org/wiki/Release_notes_017#Additional_Modules 
this way people are informed about this 3rd party workbench?

jmwright, 2018-03-12:
@luzpaz do you know what the process is for submitting a description for that page?

luzpaz, 2018-03-12:
@jmwright do you have a wiki account ? That's all it takes ;) If not I can arrange something.

jmwright, 2018-03-12:
I don't have a wiki account. I only have one for the forum.

luzpaz, 2018-04-16:
Did this get sorted out for you ?

jmwright, 2018-04-18:
We had a couple of conversations come up in the community that distracted me and I haven't gotten back to this yet. I haven't forgotten though. Thanks for reminding me.

luzpaz, 2018-05-06:
I'll add it  for you. Thanks!

jmwright, 2018-05-06:
Thanks for doing that!

closed 2018-05-06 by jmwright