CadQuery/cadquery-freecad-workbench #116: Option to use external editor with the cadquery module

gntech, 2018-05-01:
I am using atom as my primary code editor and would like to use that to edit my cadquery models instead of the internal code editor in freecad. Currently there are some obstacles to this workflow that I think we can solve.

This is the current workflow:

1. Open script in cadquery-module.
2. Open the same script in atom.
3. Do edits in atom and hit save.
4. Cadquery module senses the file has been changed and a popup ask me to reload the file.
5. I reload the file
6. Then I can press F2 to build the model.

This is cumbersome since I need to press "yes" on the reload pop-up for every change to the model as well as hit F2 for build.

This is what I would like to be able to do:

1. Open script in cadquery-module.
2. Enable the option to "autoreload and autobuild"
3. Open the same script in atom.
4. Do edits in atom and hit save.
5. Cadquery module senses the file has been changed and automatically reload the file and does a rebuild.

This way I can have atom open on one screen and freecad open on the other screen. I can do my edits in atom and the model in freecad is instantly updated for every change.

jmwright, 2018-05-01:
Have a look at the use_external_editor and execute_on_save options in Settings.py.

https://github.com/jmwright/cadquery-freecad-module/wiki/Advanced-Usage

gntech, 2018-05-01:
Nice! This was exactly what I was looking for :+1:

closed 2018-05-01 by gntech

dbhaig, 2019-05-27:
Is this still an option? I can't find Settings.py and the CadQuery->Settings menu does not appear to have an external editor option. Am I missing something?

Thanks.

jmwright, 2019-05-27:
@dbhaig Unfortunately, I forgot to update the docs for this. By default, this module should work with an external editor now, and the "Allow Reload" option in the new Settings dialog controls that. Some editors handle altering the file on disk different ways though, so the behavior may vary a little bit depending on the editor you're using.

dbhaig, 2019-05-27:
Got it. Thank you for the prompt response.