CadQuery/cadquery-freecad-workbench #118: Integrate cqparts

jmwright, 2018-05-02:
### Background
cqparts adds a layer on top of CadQuery to better support parts and assemblies, with additional support for things like fasteners with threads. This is one area where CadQuery is sorely lacking compared to something like OpenSCAD, and cqparts is well positioned to address that shortcoming.

Related to [this issue](https://github.com/fragmuffin/cqparts/issues/66)

### Implementation
In keeping with the "batteries included" philosophy that I've adopted with this module, cqparts and the libraries that it depends on should be embedded. That's sufficiently complex that a script will need to be created to download the cqparts packages from PyPi, extract them, and then put them in the appropriate places, updating them as needed.

gntech, 2018-05-04:
cqparts is brilliant, I am very much in favor of including it in the freecad-module. As mentioned, It brings a whole other level of functionality to cadquery.

jmwright, 2018-05-05:
Done in commit 6104b0a403c64663bea09dfc16d25c9b34a04b91

I would still like to include some cqparts examples in the CadQuery > Examples submenu, but that can come later.

closed 2018-05-05 by jmwright