CadQuery/cadquery-freecad-workbench #120: Importing cqparts fails on Windows due to embedded binary of numpy for Linux

zwn, 2018-05-14:
I have installed the module into freecad using the addons manager.

OS: Windows 10
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.17.13519 (Git)
Build type: Release
Branch: releases/FreeCAD-0-17
Hash: 1a8b868018f45ea486c0023fdbfeb06febc1fb89
Python version: 2.7.14
Qt version: 4.8.7
Coin version: 4.0.0a
OCC version: 7.2.0
Locale: English/UnitedKingdom (en_GB)

When running `import cqparts` I get the following error:

```
>>> import cqparts
Traceback (most recent call last):
  File "<input>", line 1, in <module>
  File "C:\Users\zbynek\AppData\Roaming\FreeCAD\Mod\cadquery_module\ThirdParty\cqparts\__init__.py", line 53, in <module>
    from . import codec
  File "C:\Users\zbynek\AppData\Roaming\FreeCAD\Mod\cadquery_module\ThirdParty\cqparts\codec\__init__.py", line 191, in <module>
    from .gltf import GLTFExporter
  File "C:\Users\zbynek\AppData\Roaming\FreeCAD\Mod\cadquery_module\ThirdParty\cqparts\codec\gltf.py", line 9, in <module>
    import numpy
  File "C:\Users\zbynek\AppData\Roaming\FreeCAD\Mod\cadquery_module\Libs\numpy\__init__.py", line 142, in <module>
    from . import add_newdocs
  File "C:\Users\zbynek\AppData\Roaming\FreeCAD\Mod\cadquery_module\Libs\numpy\add_newdocs.py", line 13, in <module>
    from numpy.lib import add_newdoc
  File "C:\Users\zbynek\AppData\Roaming\FreeCAD\Mod\cadquery_module\Libs\numpy\lib\__init__.py", line 8, in <module>
    from .type_check import *
  File "C:\Users\zbynek\AppData\Roaming\FreeCAD\Mod\cadquery_module\Libs\numpy\lib\type_check.py", line 11, in <module>
    import numpy.core.numeric as _nx
  File "C:\Users\zbynek\AppData\Roaming\FreeCAD\Mod\cadquery_module\Libs\numpy\core\__init__.py", line 26, in <module>
    raise ImportError(msg)
ImportError: 
Importing the multiarray numpy extension module failed.  Most
likely you are trying to import a failed build of numpy.
If you're working with a numpy git repo, try `git clean -xdf` (removes all
files not under version control).  Otherwise reinstall numpy.

Original error was: cannot import name multiarray
```

Further checking the problem I see that multiarray is a .so file and I am on Windows. I am not sure if Freecad plugin mechanism can deal with this. Is there a different way of installing that would allow me to pip-install the required libraries?

zwn, 2018-05-14:
I am able to run pip from the embedded freecad python environment:

```
$ ./python.exe -m pip list
matplotlib (1.4.2)
pip (9.0.2)
Pivy (0.5.0)
pycollada (0.4.1)
pyparsing (2.0.3)
python-dateutil (2.2)
PyYAML (3.11)
setuptools (5.4.1)
six (1.8.0)
wheel (0.30.0)
You are using pip version 9.0.2, however version 10.0.1 is available.
You should consider upgrading via the 'python -m pip install --upgrade pip' command.
```
So using it to install any binary module seems possible. I was able to upgrade pip to version 10.0.1 as suggested and install numpy running:
```
$ ./python.exe -m pip install numpy
Collecting numpy
  Downloading https://files.pythonhosted.org/packages/22/be/6865fecd80834a242b32f722e4fde27335f60de6b6b1430626fe7a84ce40/numpy-1.14.3-cp27-none-win_amd64.whl (13.3MB)
    100% |################################| 13.3MB 616kB/s
matplotlib 1.4.2 requires pytz, which is not installed.
pycollada 0.4.1 has requirement python-dateutil==1.5, but you'll have python-dateutil 2.2 which is incompatible.
Installing collected packages: numpy
Successfully installed numpy-1.14.3
```

However the installed `numpy` fails to import with very similar message 😟. Oh, the reason for that is most likely the fact, that python included in freecad is compiled with microsoft visual studio 2013 while the stock python 2.7 uses 2008. That means nothing from pypi will install correctly 😢.

jmwright, 2018-05-14:
@zwn I have been able to confirm the same problem on Windows 10. The module embeds numpy in a pretty naive way, and that's probably where the problem is coming from. multiarray being a .so file makes things more challenging.

@fragmuffin Have you used cqparts on Windows? Specifically on version 10? How do you install numpy on Windows?

fragmuffin, 2018-05-14:
@jmwright no, I haven't tried it on Windows.
I've got a Windows 10 box, but haven't turned it on in months.

I'm going to have difficulty supporting Windows because I just don't know how to setup automated testing without having a dedicated box / license for it.

But I should probably at least try to get it working once :)

zwn, 2018-05-14:
[Appveyor](https://www.appveyor.com/) provides free CI windows environment.

adam-urbanczyk, 2018-05-14:
@fragmuffin you can use this as an example setup: https://github.com/dcowden/cadquery/blob/master/appveyor.yml

fragmuffin, 2018-05-15:
@zwn @adam-urbanczyk 
Oh!, I see... I looked up appveyor when @adam-urbanczyk originally set it up for `cadquery`.
But at the time I thought it was $29 a month... I didn't notice that it's free for Open Source projects.

To be continued... fragmuffin/cqparts#88

jmwright, 2018-05-18:
Just making sure I tie this issue to the thread on the Google Group.

https://groups.google.com/forum/m/#!msg/cadquery/Uf5vWtmPTLI/6NT6KE8bAwAJ

easyw, 2018-06-11:
I have the same issue ...
with this version 
[Latest commit 19994ff on May 3](https://github.com/jmwright/cadquery-freecad-module/tree/19994ffd488224d0eb9540e083a7cbeab99d4db7) everything is fine

jmwright, 2018-06-12:
@easyw I don't understand why cqparts would cause a problem with your scripts since you don't reference it. Can you please post more detail about the exact error you're getting?

easyw, 2018-06-12:
@jmwright 
I get the errors just on loading FC, without launching any script...

> During initialization the error 
> Importing the multiarray numpy extension module failed.  Most
> likely you are trying to import a failed build of numpy.
> If you're working with a numpy git repo, try `git clean -xdf` (removes all
> files not under version control).  Otherwise reinstall numpy.
> 
> Original error was: cannot import name multiarray
>  occurred in C:\Users\userC\AppData\Roaming\FreeCAD\Mod\GDT\InitGui.py
> Please look into the log file for further information
> loading Base
> Loading Parameters
> Loading Compatibility
> loading Gui
> loading Features
> loading Utils
> loading Containers
> loading LinkTools
> loading TempoVis
> loading Observer
> loading AACommand
> loading Control
> loading Tools
> loading Tools
> loading Icons
> Loading CommandCollection1
> loading LeaveEnter
> loading Tip
> loading TransferObject
> loading Part
> loading Module
> loading ShapeGroup
> loading ShapeBinder
> loading Ghost
> loading Exporter
> loading Features.PartDesign
> loading Features
> loading PDShapeFeature
> loading MuxAssembly
> Loading GlobalToolbar
> loading View
> Loading OnOff
> loading ListUsages
> has custom!
> has!
> has!
> During initialization the error 
> Importing the multiarray numpy extension module failed.  Most
> likely you are trying to import a failed build of numpy.
> If you're working with a numpy git repo, try `git clean -xdf` (removes all
> files not under version control).  Otherwise reinstall numpy.
> Original error was: cannot import name multiarray
>  occurred in C:\Users\userC\AppData\Roaming\FreeCAD\Mod\FCGear\InitGui.py
> Please look into the log file for further information
> During initialization the error 
> Importing the multiarray numpy extension module failed.  Most
> likely you are trying to import a failed build of numpy.
> If you're working with a numpy git repo, try `git clean -xdf` (removes all
> files not under version control).  Otherwise reinstall numpy.
> 
> Original error was: cannot import name multiarray
>  occurred in C:\Users\userC\AppData\Roaming\FreeCAD\Mod\assembly2\InitGui.py
> Please look into the log file for further information
> 

it seems related to other WBs which are calling/using numpy

easyw, 2018-06-12:
Note that bumpy is already bundled with FC on win

jmwright, 2018-06-12:
@easyw Thanks for the info. That was very helpful.

@easyw @zwn Please try the latest master of this module. If you've installed it through the add-ons manager, you should be able to just click the Install/Update button.

easyw, 2018-06-12:
it works  fine now... thx 😄

jmwright, 2018-06-12:
@zwn Feel free to reopen, but this seems to be working for both @easyw and I.

closed 2018-06-12 by jmwright

fragmuffin, 2018-07-13:
Finally got AppVeyor up and running.
FreeCAD 0.17, python 3.6 (x64)
fragmuffin/cqparts#119

I had a lot of trouble working _outside_ a `conda` environment; `conda install` bad, `conda create` good.
But no `numpy` issues :+1: