CadQuery/cadquery-freecad-workbench #121: Python3 support

adam-urbanczyk, 2018-05-18:
I cannot get CQ module working windows and FC 0.17

It looks like the module is written for py2 but with newest FC versions py3 is bundled.

```python
Traceback (most recent call last):
  File "<string>", line 136, in InitApplications
  File "<string>", line 12, in <module>
  File "C:\Users\x\Anaconda2\envs\cq-fc\Library\Mod\cadquery-freecad-module\module_locator.py", line 12, in module_path
    return os.path.dirname(unicode(__file__, encoding))
NameError: name 'unicode' is not defined
```

adam-urbanczyk, 2018-05-18:
@zwn maybe this is interesting for you as well

zwn, 2018-05-18:
Thanks @adam-urbanczyk. Other (easily solvable) issues involve supporting proper relative imports.

I've stopped trying somewhere along Pyside -> Pyside2 conversion which I unfortunately know nothing about (some method signature changes mostly I guess). There might be more after those.

jmwright, 2018-05-30:
@adam-urbanczyk When you worked on the CadQuery Python 2/3 conversion, how did you handle converting the unicode calls to something that would work in both versions? Just changing `unicode` to `str` isn't the correct way to handle it, is it?

zwn, 2018-05-30:
Maybe this [question on stackoverflow](https://stackoverflow.com/questions/38697037/how-to-convert-python-2-unicode-function-into-correct-python-3-x-syntax) can help.

jmwright, 2018-05-30:
@zwn @adam-urbanczyk Do you have special setups on your Windows systems? I don't get any errors running the CadQuery workbench on the latest stable version of FreeCAD I see for download on their site. Below is my FreeCAD version info.
```
OS: Windows 10
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.17.13522 (Git)
Build type: Release
Branch: releases/FreeCAD-0-17
Hash: 3bb5ff4e70c0c526f2d9dd69b1004155b2f527f2
Python version: 2.7.14
Qt version: 4.8.7
Coin version: 4.0.0a
OCC version: 7.2.0
Locale: English/UnitedStates (en_US)
```

I made some Python 3 updates to the module, but I'm guessing they won't be enough. I need to be able to replicate the problem you're having before I can fix the rest of the errors.

adam-urbanczyk, 2018-05-31:
@jmwright  not sure right now. I remember I used e.g. `from __future__ import unicode_literals`

I need to take a look why the unicode() is there in the first place to be able to tell more.

adam-urbanczyk, 2018-05-31:
@zwn @jmwright  this is my env btw (based on appveyor setup of cadquery):

``` conda create --quiet --name cqtest -c cadquery -c conda-forge python=%PYTHON_VERSION% freecad=0.17 python=3.6 pyparsing```

adam-urbanczyk, 2018-06-02:
@jmwright  I think this will do the trick for both versions:
```python
from __future__ import unicode_literals
os.path.dirname('{}'.format(__file__))
```

jmwright, 2018-06-11:
I tried using the conda environment in Ubuntu 17.10, but I get this error which seems to be the same thing that was reported in #120 on Windows.
```
Importing the multiarray numpy extension module failed.  Most
likely you are trying to import a failed build of numpy.
If you're working with a numpy git repo, try `git clean -xdf` (removes all
files not under version control).  Otherwise reinstall numpy.
```
I tried installing numpy using pip and conda, and I get the message that it's already installed.

jmwright, 2018-06-12:
The numpy error was related to another workbench I had installed (Assembly 2). I have gotten around that on startup now.

jmwright, 2018-06-12:
It took a fair amount of work, but the latest commit to master works for me in the Anaconda FreeCAD 0.17 / Python 3 environment. Use with caution though. There may still be issues lurking as different features of PyQode are triggered, and I had multiple segfaults running FreeCAD from that environment. I haven't seen any segfaults in awhile, but I want people to be aware. Save your work often.

jmwright, 2018-06-14:
I've been using the Python 3 environment pretty exclusively over the past couple of days. I've fixed a few bugs that came up, but everything seems to be working well now. I'll keep using Python 3 and fixing things as I notice them.

closed 2018-06-14 by jmwright