CadQuery/cadquery-freecad-workbench #122: Can't sweep a path along a wire

jhol, 2018-05-19:
Hi - I'm trying to make a screw thread in the CadQuery work-bench in FreeCad 0.17. Here is some code that is analogous to the code I'm trying to have:

```
import cadquery as cq
thread_profile = cq.Workplane("XZ").circle(1)
thread_helix = cq.Wire.makeHelix(4, 12, 20)
r = thread_profile.sweep(thread_helix)
show_object(r)
```
When I run this script I get the following error message in the log:
```
Executing CQGI-compliant script.
Error executing CQGI-compliant script. 'Wire' object has no attribute 'wire'
Executed /home/joel/workspace/opentechlab/cadquery-example/nozzle/cap.py
```

jmwright, 2018-05-19:
[Example 31](https://github.com/dcowden/cadquery/blob/master/examples/FreeCAD/Ex031_Sweep.py) involves sweeping a profile along a wire. That may help you.

Depending on what you're doing, you might also want to have a look at the [cqparts library](https://fragmuffin.github.io/cqparts/doc/cqparts_fasteners/index.html) too. @fragmuffin has done a lot with threads in that.

fragmuffin, 2018-05-21:
Hi @jhol 

Yes, I had a lot of fun with threads, and my journey isn't over yet.
I managed to build threads from a tooth profile by [creating a cross-section](https://fragmuffin.github.io/cqparts/doc/api/cqparts_fasteners.solidtypes.threads.html#cqparts_fasteners.solidtypes.threads.base.profile_to_cross_section
) by polar-plotting the profile.
The code to create the helix, and sweep it is [here](https://github.com/fragmuffin/cqparts/blob/master/src/cqparts_fasteners/solidtypes/threads/base.py#L332-L339). As you can see from that code, immediately after it is a messy try-catch block to break when the sweep doesn't create a watertight mesh, which sadly happens about 30% of the time.
There's a lot of detail on this in fragmuffin/cqparts#1, I don't have a solution yet.

Examples of profiles and the resulting threads are pictured here: [`cqparts_fasteners.solidtypes.threads`](https://fragmuffin.github.io/cqparts/doc/api/cqparts_fasteners.solidtypes.threads.html)

jmwright, 2018-05-23:
@jhol Feel free to reopen if needed.

closed 2018-05-23 by jmwright

jhol, 2018-05-23:
@jmwright this is a bug though, right? I agree there's a workaround, but my example shouldn't fail with an error, right?

jmwright, 2018-05-23:
> this is a bug though, right?

I don't think so. When you call `cq.Wire.makeHelix` you're getting back a lower level object (Wire and not a Workplane), so you have to repackage the helix before you can use it with sweep (higher level operation). The following code works for me.
```python
import cadquery as cq
thread_profile = cq.Workplane("XZ").circle(1)
thread_helix = cq.Wire.makeHelix(4, 12, 20)
shape = cq.Wire.combine([thread_helix])
path = cq.Workplane("XY").newObject([shape])
r = thread_profile.sweep(path)

show_object(r)
```
This is the way it's done in [cqparts](https://github.com/fragmuffin/cqparts/blob/ece76ae0e7ae0c09ff717397eb85ad7a285e6939/src/cqparts_fasteners/solidtypes/threads/base.py#L207) as well.

It is probably a little counter-intuitive because you're mixing layers of the abstraction. There may be a way to implement makeHelix at the top layer of abstraction (Workplane level), but I'm having trouble picturing how that would work at the moment.

jmwright, 2018-05-24:
@jhol If you're still having trouble getting it to work or still feel that there's a bug, feel free to reopen or continue the discussion.

nopria, 2020-06-22:
I don't understand how [the workaround](https://github.com/jmwright/cadquery-freecad-module/issues/122#issuecomment-391391751) works, in the following snippet the circle looks swept along a path very different from the defined one.

```
import cadquery as cq
thread_profile = cq.Workplane("XZ").circle(5)
thread_helix = cq.Wire.makeHelix(50,50,100)
shape = cq.Wire.combine([thread_helix])
path = cq.Workplane("XY").newObject([shape])
r = thread_profile.sweep(path)

show_object(path)
show_object(r)
```
[image]

I run the above snippet in cq-editor, maybe in FreeCAD the result is different?

jmwright, 2020-06-22:
@nopria The FreeCAD workbench does not support CadQuery 2.0, so I would stick with CQ-editor. I think that there are multiple issues with sweep that have been fixed on CadQuery master (via the update to OCCT 7.4.x), but I don't think there's a release of CQ-editor that has those changes in it.

adam-urbanczyk, 2020-06-23:
@nopria this is not related to FreeCAD workbench. I answered here https://github.com/dcowden/cadquery/issues/309 , but please use the new CQ repo for future questions/issues.