CadQuery/cadquery-freecad-workbench #123: Cadquery in freecad AppImage

yohcop, 2018-06-02:
Hi,
I can't get the CQ freecad module working in Freecad, or I'm missing something.

I downloaded FreeCad AppImage binary. Run it, and installed this module from the tools->addon manager. The install seems fine, the files are present in ~/.FreeCad/Mod/cadquery_module/.

However on starting FreeCad, I get the following error:
`During initialization the error No module named cadquery occurred in /home/y/.FreeCAD/Mod/cadquery_module/InitGui.py`

Below is the full log, but one relevant part:

```
Log: Init:      Initializing /home/y/.FreeCAD/Mod/cadquery_module... failed
Log: ----------------------------------------------------------------------------------------------------
Log: Traceback (most recent call last):
  File "<string>", line 120, in InitApplications
  File "<string>", line 8, in <module>
  File "/home/y/.FreeCAD/Mod/cadquery_module/CQGui/Command.py", line 14, in <module>
    from cadquery import cqgi
ImportError: No module named cadquery
```

`cadquery` is present (this file exists, but maybe it's not in the path?) `~/.FreeCAD/Mod/cadquery_module/Libs/cadquery/cadquery/__init__.py `

Thanks a lot,

And here is the full FreeCad.log:
```
﻿Msg: FreeCAD 0.17, Libs: 0.17R13522 (Git)
© Juergen Riegel, Werner Mayer, Yorik van Havre 2001-2018
  #####                 ####  ###   ####  
  #                    #      # #   #   # 
  #     ##  #### ####  #     #   #  #   # 
  ####  # # #  # #  #  #     #####  #   # 
  #     #   #### ####  #    #     # #   # 
  #     #   #    #     #    #     # #   #  ##  ##  ##
  #     #   #### ####   ### #     # ####   ##  ##  ##

Log: Time = Fri Jun  1 21:08:36 2018
Log: AppDataSkipVendor = true
Log: AppHomePath = /tmp/.mount_FreeCA7TfoMN/usr/lib/freecad/
Log: AppIcon = freecad
Log: AppTempPath = /tmp/
Log: BinPath = /tmp/.mount_FreeCA7TfoMN/usr/lib/freecad/bin/
Log: BuildRepositoryURL = git://github.com/FreeCAD/FreeCAD.git releases/FreeCAD-0-17
Log: BuildRevision = 13522 (Git)
Log: BuildRevisionBranch = releases/FreeCAD-0-17
Log: BuildRevisionDate = 2018/05/08 13:25:00
Log: BuildRevisionHash = 3bb5ff4e70c0c526f2d9dd69b1004155b2f527f2
Log: BuildVersionMajor = 0
Log: BuildVersionMinor = 17
Log: Console = 0
Log: CopyrightInfo = © Juergen Riegel, Werner Mayer, Yorik van Havre 2001-2018
  #####                 ####  ###   ####  
  #                    #      # #   #   # 
  #     ##  #### ####  #     #   #  #   # 
  ####  # # #  # #  #  #     #####  #   # 
  #     #   #### ####  #    #     # #   # 
  #     #   #    #     #    #     # #   #  ##  ##  ##
  #     #   #### ####   ### #     # ####   ##  ##  ##

Log: Debug = 0
Log: DocPath = /tmp/.mount_FreeCA7TfoMN/usr/lib/freecad/doc/
Log: ExeName = FreeCAD
Log: ExeVendor = FreeCAD
Log: ExeVersion = 0.17
Log: LoggingFile = 1
Log: LoggingFileName = /home/y/.FreeCAD/FreeCAD.log
Log: MaintainerUrl = http://www.freecadweb.org/wiki/Main_Page
Log: PATH = /tmp/.mount_FreeCA7TfoMN/usr/bin/:/tmp/.mount_FreeCA7TfoMN/usr/sbin/:/tmp/.mount_FreeCA7TfoMN/usr/games/:/tmp/.mount_FreeCA7TfoMN/bin/:/tmp/.mount_FreeCA7TfoMN/sbin/:/usr/local/epath/mucks:/usr/local/epath/ibazel:/usr/local/epath/chromedriver:/usr/local/epath/buildifier:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin:/usr/local/epath/go/bin
Log: PYTHONHOME = /tmp/.mount_FreeCA7TfoMN/usr/
[... 148 lines cut ...]
Log: Init:      Initializing /tmp/.mount_FreeCA7TfoMN/usr/lib/freecad/Mod/Material... done
Log: Init:      Initializing /tmp/.mount_FreeCA7TfoMN/usr/lib/freecad/Mod/Raytracing... done
Log: Init:      Initializing /tmp/.mount_FreeCA7TfoMN/usr/lib/freecad/Mod/Mesh... done
Log: Init:      Initializing /tmp/.mount_FreeCA7TfoMN/usr/lib/freecad/Mod/Path... done
Log: Init:      Initializing /tmp/.mount_FreeCA7TfoMN/usr/lib/freecad/Mod/Part... done
Log: Init:      Initializing /tmp/.mount_FreeCA7TfoMN/usr/lib/freecad/Mod/Arch... done
Log: Init:      Initializing /tmp/.mount_FreeCA7TfoMN/usr/lib/freecad/Mod/Robot... done
Log: Init:      Initializing /tmp/.mount_FreeCA7TfoMN/usr/lib/freecad/Mod/AddonManager... done
Log: Init:      Initializing /tmp/.mount_FreeCA7TfoMN/usr/lib/freecad/Mod/Idf(InitGui.py not found)... ignore
Log: Init:      Initializing /tmp/.mount_FreeCA7TfoMN/usr/lib/freecad/Mod/Inspection... done
Log: Init:      Initializing /tmp/.mount_FreeCA7TfoMN/usr/lib/freecad/Mod/Drawing... done
Log: Init: Loading FreeCAD GUI
Log: Init: Running FreeCADGuiInit.py start script... done
Log: Init: Activating default workbench StartWorkbench
Log: Loading GUI of Web module... done
Log: Loading GUI of Start module... done
Log: Loading Start module... done
Log: Init: Showing main window
Log: Main window restored
Log: Show main window
Log: Toolbars restored
Log: Couldn't connect to spacenav daemon
Log: Init: Entering event loop
Log: Init: Processing command line files
```

jmwright, 2018-06-02:
@yohcop There's an issue with this module being run from within an appimage. The module needs access to files external to the appimage itself. See [this comment](https://github.com/jmwright/cadquery-freecad-module/issues/43#issuecomment-377794293) for a potential solution.

yohcop, 2018-06-02:
Thanks a lot for the very quick reply! I'll try the ppa version instead.

yohcop, 2018-06-02:
Unfortunately, I get exactly the same error with the ppa version of freecad (same log, except the paths are now linking to my system /usr/lib/freecad/...)

yohcop, 2018-06-02:
For what it's worth, I installed the plugin manually from the last version of May 10 (https://github.com/jmwright/cadquery-freecad-module/tree/6734927c1dd9df776b735b9ba6be65fb0bd4ce1d), and it works. So probably something in the last 2 changes is the issue.

jmwright, 2018-06-02:
@yohcop Can you try to install again from the addons manager? For some reason I'm not able to reproduce your problem, but it's either the thing I just reverted, or one other change that's the issue.

jmwright, 2018-06-02:
If that doesn't fix it, could you post the result of running these commands in FreeCAD's python console?
```python
import sys
print(sys.path)
```

yohcop, 2018-06-02:
Yup, seems to be working now. I wrote a longer email with an error I got when I first tried, but now I can't reproduce it anyway (related to the editor files apparently).

Thanks a lot for the incredibly fast turnaround!
I'll close this now.

closed 2018-06-02 by yohcop