CadQuery/cadquery-freecad-workbench #124: 'show_object' is not defined based on text in comments

Xaphiosis, 2018-06-10:
This code:

```
import cadquery as cq

# New method to render script results using the CadQuery Gateway Interface
# Use the following to render your model with grey RGB and no transparency
# show_object(result, options={"rgba":(204, 204, 204, 0.0)})
# These can be modified rather than hardcoding values for each dimension.
length = 80.0               # Length of the block
height = 60.0               # Height of the block
thickness = 10.0            # Thickness of the block
center_hole_dia = 22.0      # Diameter of center hole in block

result = cq.Workplane("XY").box(length, height, thickness) \
    .faces(">Z").workplane().hole(center_hole_dia)

show_object(result)
```

results in:

```
Running the Python command 'CadQueryExecuteScript' failed:
Traceback (most recent call last):
  File "/home/x/.FreeCAD/Mod/cadquery_module/CQGui/Command.py", line 226, in Activated
    imp.load_source('temp_module', tempFile.name)
  File "/tmp/tmpIUCiyy", line 15, in <module>
    show_object(result)

name 'show_object' is not defined
```

Whereas the same code as above minus the comment:

```
import cadquery as cq

length = 80.0               # Length of the block
height = 60.0               # Height of the block
thickness = 10.0            # Thickness of the block
center_hole_dia = 22.0      # Diameter of center hole in block

result = cq.Workplane("XY").box(length, height, thickness) \
    .faces(">Z").workplane().hole(center_hole_dia)

show_object(result)
```

Results in:

```
Executed /tmp/blah.py
```

This is a bit of a major head-scratcher to a first time user of cadquery! I'm guessing something is scanning the python script to see whether "show" or "show_object" are used?

I'm using cadquery 1.2.0 via pip (updated today), latest FreeCAD 0.17 stable from ppa:freecad-maintainers/freecad-stable, running on Xubuntu 16.04.

jmwright, 2018-06-10:
@Xaphiosis Thanks for reporting this. This is due to the clumsy way the FreeCAD module detects whether the script is CadQuery Gateway Interface (CQGI) compliant. If you remove just the `# show_object` from the comment block, it should solve the problem.

There's a reason the detection works the way it does, but obviously I need to find a different solution. I'll put this on the to-do list.

dcowden, 2018-06-10:
@Xaphiosis this is also my fault-- the current cqgi implementation is kind of dumb-- [this issue](https://github.com/dcowden/cadquery/issues/270) is open to fix it

Xaphiosis, 2018-06-10:
I see. Perhaps a half-way solution for now would be to adjust the template to contain a reminder for the user to remove the comment?

jmwright, 2018-06-10:
[This](https://github.com/jmwright/cadquery-freecad-module/blob/master/CQGui/Command.py#L156) is the offending line. I wanted to make sure that show_object was actually active and not just commented out before running the script. I either need to take out the comment checks, or let the test pass if a commented show_object call is accompanied by an uncommented one.

jmwright, 2018-06-11:
@Xaphiosis This should be fixed in the latest commit to master.

closed 2018-06-12 by jmwright