CadQuery/cadquery-freecad-workbench #128: can't open a script file

rowanG077, 2018-07-27:
Hello,

I have been a user of FreeCad + CadQuery for a while now but it was always on windows. Recently I switched over to ubuntu 18.04 and installed freecad using the default install as is described here: https://www.freecadweb.org/wiki/Installing.

Now I installed the cadquery plugin via the addon manager. Now when I open any script I get the following error message:
```

Running the Python command 'CadQueryOpenScript' failed:
Traceback (most recent call last):
File "/home/rowang/.FreeCAD/Mod/cadquery_module/CQGui/Command.py", line 298, in Activated
ImportCQ.open(filename[0])
File "/home/rowang/.FreeCAD/Mod/cadquery_module/CQGui/ImportCQ.py", line 51, in open
from pyqode.core.modes import FileWatcherMode
File "/home/rowang/.FreeCAD/Mod/cadquery_module/Libs/pyqode/core/modes/__init__.py", line 6, in <module>
from .autocomplete import AutoCompleteMode
File "/home/rowang/.FreeCAD/Mod/cadquery_module/Libs/pyqode/core/modes/autocomplete.py", line 5, in <module>
from pyqode.core.api import TextHelper
File "/home/rowang/.FreeCAD/Mod/cadquery_module/Libs/pyqode/core/api/__init__.py", line 6, in <module>
from .code_edit import CodeEdit
File "/home/rowang/.FreeCAD/Mod/cadquery_module/Libs/pyqode/core/api/code_edit.py", line 18, in <module>
from pyqode.core.managers import BackendManager
File "/home/rowang/.FreeCAD/Mod/cadquery_module/Libs/pyqode/core/managers/__init__.py", line 15, in <module>
from .backend import BackendManager
File "/home/rowang/.FreeCAD/Mod/cadquery_module/Libs/pyqode/core/managers/backend.py", line 9, in <module>
from pyqode.core.api.client import JsonTcpClient, BackendProcess
File "/home/rowang/.FreeCAD/Mod/cadquery_module/Libs/pyqode/core/api/client.py", line 15, in <module>
from pyqode.qt import QtCore, QtNetwork
File "/home/rowang/.FreeCAD/Mod/cadquery_module/Libs/pyqode/qt/QtNetwork.py", line 18, in <module>
from PySide2.QtNetwork import *

No module named PySide2.QtNetwork
```

I'm not sure what I am doing wrong. Am I missing some dependancies?

jmwright, 2018-07-27:
I've been hesitant to upgrade to 18.04, so I haven't encountered this. You might need to install the PySide2 equivalent of the QtNetwork installation step here. https://github.com/jmwright/cadquery-freecad-module/blob/master/docs/installation.md#troubleshooting

jmwright, 2018-07-31:
@rowanG077 Any progress on this?

rowanG077, 2018-08-01:
I couldn't manage to install PySide2 to work with cadquery. So I just use the AppImage version FreeCad that includes PySide2.

jmwright, 2018-09-25:
@rowanG077 I finally upgraded to 18.04 and am not able to reproduce this problem. I always install FreeCAD from the stable PPA. I don't use the version in the stock Ubuntu repo because it always feels stale.

If you're still interested in a solution, could you try installing from the stable PPA to see if that solves this issue for you?

closed 2018-09-30 by jmwright