CadQuery/cadquery-freecad-workbench #129: Updating cadquery-freecad-module WB to be compatible with the upcoming 0.18 release (req. py3 and Qt5 compatibility)

luzpaz, 2018-09-03:
ref: https://github.com/FreeCAD/FreeCAD-addons/issues/93

jmwright, 2018-09-03:
@luzpaz Thanks for organizing this. Is there a release date for 0.18 yet?

luzpaz, 2018-09-03:
@jmwright np
There is a soft agreement for the end of 2018

jmwright, 2018-12-29:
The updates in #136 have been tested against FreeCAD 0.18 with Qt5 and Python 3. Updating via the FreeCAD add-on manager should pull the new version. Be aware though that there are significant changes in this version of the workbench, the most disruptive probably being the new code editor. Please see the [changelog](https://github.com/jmwright/cadquery-freecad-module/blob/master/changes.md#v130-unreleased) for details.

closed 2018-12-29 by jmwright