CadQuery/cadquery-freecad-workbench #130: F2 key is hijacked

ceremcem, 2018-09-06:
F2 key is used for both renaming an object and running a cadQuery script when cadQuery WB is opened. At the moment, F2 tries to run some scripts (which ends up with exception thrown) while we are not in cadQuery WB.

## Proposal

Maybe some other key key could be selected as a shortcut for running scripts.

jmwright, 2018-09-07:
The `execute_binding` setting in the Settings.py file allows you to change the execute key.

https://github.com/jmwright/cadquery-freecad-module/blob/master/Settings.py#L5

ceremcem, 2018-09-09:
Do you think this setting should stay as is by default?

jmwright, 2018-09-10:
It's been F2 since I created the workbench. If I change it now, I'll need to add something like a deprecation warning for several months so that long-time users don't have problems.

Do you have a key that you suggest changing it to? I can't promise it will change, but I'll think it through.

ceremcem, 2018-09-10:
I would suggest a key but I'm very busy with Asm3 workbench at the moment. I don't know if it conflicts with another key but Code::Blocks uses `F9` to build and run and Geany uses `F9` to build and `F5` to run. So I would vote for `F9`.

Also, is it possible to disable this key while the WB is not in use? That would make conflicts a lot less occurred and even current `F2` shortcut would survive.

easyw, 2018-09-10:
@jmwright I shouldn't use F5 because it is already mapped on FC for refresh.
A good option would be to write the map binding in the FC parameter variable, so it wouldn't be over-written by a CQ update.
i.e. 
FreeCAD.ParamGet("User parameter:BaseApp/Preferences/Mod/CQ")

jmwright, 2018-09-12:
@ceremcem Asm3 sounds great. I've been using Assembly2 for quite awhile, so I'll give your workbench a look.

I think I'll try what @easyw is suggesting to keep from overriding the F2 key for the rest of FreeCAD. Thanks for the suggestion @easyw

jmwright, 2018-09-14:
@ceremcem Thanks for the builder boilerplate suggestion. I'll check that out.

I haven't totally decided whether to change F2 to something else yet, but whatever I do, my main goal is for my workbench to play well inside of FreeCAD. I want it to work well with the normal workflow, and not cause conflicts. I think the solution @easy is suggesting may take care of that.

jmwright, 2018-12-29:
In #136 I've changed the default "execute script" key to F9, but it is still configurable. There's a settings dialog under CadQuery->Settings to change this. Updating via the FreeCAD add-on manager should pull the new version. Be aware though that there are significant changes in this version of the workbench, the most disruptive probably being the new code editor. Please see the [changelog](https://github.com/jmwright/cadquery-freecad-module/blob/master/changes.md#v130-unreleased) for details.

closed 2018-12-29 by jmwright