CadQuery/cadquery-freecad-workbench #131: Install cadquery module on freecad crashs it while starting

KammutierSpule, 2018-10-29:
I would like to cross link this issue:
https://www.freecadweb.org/tracker/view.php?id=3661

cadquery module is crashing FreeCad.

It happens on any 0.16 0.17 0.18 version

jmwright, 2018-10-30:
Things like this are usually a bad interaction with another add-on workbench. Do you have any other extra workbenches installed?

KammutierSpule, 2018-10-30:
I do have, kicadStepup module. I will try latter removing that other add-on.
but if I delete the cadquery module folder it works ok.

KammutierSpule, 2018-10-30:
Hi @easyw 
I was looking to solve my issues before contact you, but .. here you are ;)

I am very new to this so I am still trying to understand what is what.
So I learned that for FreeCad, cadquery project is this one: cadquery-freecad-module

I am running it from command line, but if I open it via cadquery-freecad-module user interface and press execute I also the [other issue](https://github.com/jmwright/cadquery-freecad-module/issues/132)

I get freecad crash when I install cadquery-freecad-module from package manager or if I clone this repository.
But if I checkout the tagged version v1.2.0 it do not crash.

When it crash, I tested with FreeCad 0.16 0.17 0.18

easyw, 2018-10-30:
@KammutierSpule @jmwright 
it seems that [our cq scripts](https://github.com/easyw/kicad-3d-models-in-freecad/tree/master/cadquery/FCAD_script_generator) are running fine, instead a simple example not...
The error on running the script seems an issue of sys.path ...

On the other side I cannot reproduce your crash installing CQ from the Addon Manager
> cadquery module is crashing FreeCad.

In which way did you get the CQ different from the one collected by the Addon manager?

KammutierSpule, 2018-10-30:
> In which way did you get the CQ different from the one collected by the Addon manager?

Maybe there is a difference, I remember there was some error related with gitpython? and I installed it.
So it may be getting directly from git repository now

easyw, 2018-10-30:
I get the same CQ version using gitpython

> CadQuery 1.2.0
CadQuery is a parametric scripting API for creating and traversing CAD models
Author: David Cowden
License: Apache-2.0
Website: https://github.com/dcowden/cadquery

And I get no crash...

Sometimes it is useful to reset the FC profile...
Does it crash also with only CQ installed?

KammutierSpule, 2018-10-30:
I runned on a different Linux Mint (version 18) machine with similar setup and it is not crashing. Could it be something related with latest LinuxMint version?

jmwright, 2018-10-30:
> you need to have an empty document open to run the script... other-ways CQ raises an error 

@easyw Can you post the steps to reproduce please? I'm not sure what order you'd be doing things in.

jmwright, 2018-10-30:
Is test.py from one of your repos?

I never run CadQuery from a macro, so this is new territory for me.

KammutierSpule, 2018-10-30:
@jmwright The test.py is related with the other issue #132
that is not related with the crash I reported.

jmwright, 2018-11-10:
@KammutierSpule In the FreeCAD issue wmayer asked what versions of Python and Qt your FreeCAD was built against. To find that information so you can answer, go to Help -> About FreeCAD and click on the Libraries tab.

KammutierSpule, 2018-11-13:
According to FreeCad guys, this is related with versions libs. I managed to make it work without crash with the FreeCAD built against QT5.6.2 Py:3.6.6

Not sure if you can do anything about it.

https://www.freecadweb.org/tracker/view.php?id=3661

jmwright, 2018-11-13:
@KammutierSpule Ok, thanks. I'm going to close this issue for now. Feel free to reopen if the issue resurfaces.

closed 2018-11-13 by jmwright