CadQuery/cadquery-freecad-workbench #132: 'NoneType' object has no attribute 'addObject'

KammutierSpule, 2018-10-30:
On version v1.2.0
I am getting this error while trying to run any simple example:

```
import cadquery as cq
from Helpers import show
result = cq.Workplane("front").box(2.0, 2.0, 0.5)
show(result)
```
I got this error:
```
Traceback (most recent call last):
  File "/home/kammutierspule/test.py", line 6, in <module>
    show(result)
  File "/home/kammutierspule/.FreeCAD/Mod/cadquery_module/Helpers.py", line 55, in show
    newFeature = ad.addObject("Part::Feature", newName)
<type 'exceptions.AttributeError'>: 'NoneType' object has no attribute 'addObject'
```

KammutierSpule, 2018-10-30:
I tried to run it via
CadQuery -> OpenScript and got this error

```
Running the Python command 'CadQueryOpenScript' failed:
Traceback (most recent call last):
  File "/home/kammutierspule/.FreeCAD/Mod/cadquery_module/CQGui/Command.py", line 298, in Activated
    ImportCQ.open(filename[0])
  File "/home/kammutierspule/.FreeCAD/Mod/cadquery_module/CQGui/ImportCQ.py", line 51, in open
    from pyqode.core.modes import FileWatcherMode
  File "/home/kammutierspule/.FreeCAD/Mod/cadquery_module/Libs/pyqode/core/modes/__init__.py", line 6, in <module>
    from .autocomplete import AutoCompleteMode
  File "/home/kammutierspule/.FreeCAD/Mod/cadquery_module/Libs/pyqode/core/modes/autocomplete.py", line 5, in <module>
    from pyqode.core.api import TextHelper
  File "/home/kammutierspule/.FreeCAD/Mod/cadquery_module/Libs/pyqode/core/api/__init__.py", line 6, in <module>
    from .code_edit import CodeEdit
  File "/home/kammutierspule/.FreeCAD/Mod/cadquery_module/Libs/pyqode/core/api/code_edit.py", line 18, in <module>
    from pyqode.core.managers import BackendManager
  File "/home/kammutierspule/.FreeCAD/Mod/cadquery_module/Libs/pyqode/core/managers/__init__.py", line 15, in <module>
    from .backend import BackendManager
  File "/home/kammutierspule/.FreeCAD/Mod/cadquery_module/Libs/pyqode/core/managers/backend.py", line 9, in <module>
    from pyqode.core.api.client import JsonTcpClient, BackendProcess
  File "/home/kammutierspule/.FreeCAD/Mod/cadquery_module/Libs/pyqode/core/api/client.py", line 15, in <module>
    from pyqode.qt import QtCore, QtNetwork
  File "/home/kammutierspule/.FreeCAD/Mod/cadquery_module/Libs/pyqode/qt/QtNetwork.py", line 18, in <module>
    from PySide2.QtNetwork import *

No module named PySide2.QtNetwork
```

jmwright, 2018-10-30:
For the QtNetwork error, see the documentation [here](https://github.com/jmwright/cadquery-freecad-module/blob/master/docs/installation.md#troubleshooting).

You mentioned Linux Mint in the other issue. How are you installing FreeCAD? Is it an AppImage, using one of the FreeCAD PPAs, or some other method?

Please be aware that's there's not full support for FreeCAD 0.18 yet. There's a separate issue for that [here](https://github.com/jmwright/cadquery-freecad-module/issues/129).

jmwright, 2018-10-30:
Ok, that's the way I install FreeCAD on Ubuntu 18.04.

KammutierSpule, 2018-10-30:
I solved the issue with QtNetwork and I am now able to run it over the "CadQuery -> OpenScript -> F2" on booth versions!
Thanks! Sorry I didn't noticed that in documentation...

I still get the issue when I run via command line:
`Exception while processing file: ./test.py ['NoneType' object has no attribute 'addObject']`

jmwright, 2018-10-30:
When you have `show()` in your script, it's expecting to be run from within the FreeCAD GUI. It's trying to access the active document, so it won't run correctly from the command line.

KammutierSpule, 2018-10-30:
Makes sense, but I am starting FreeCad to go to GUI (not to run headless).
Also I can reproduce in GUI by opening the script using the other GUI window tool ( the icon while on the CadQuery mode ) by selecting the script and press run it.

jmwright, 2018-10-30:
When you say "command line" do you mean the Python console within FreeCAD, or the Linux terminal? I want to make sure we're thinking of the same thing here.

easyw, 2018-10-30:
@jmwright 
[image]

KammutierSpule, 2018-10-30:
> When you say "command line" do you mean the Python console within FreeCAD, or the Linux terminal? I want to make sure we're thinking of the same thing here.

I mean Linux terminal:
$freecad ./test.py

jmwright, 2018-10-30:
@KammutierSpule @easyw Thanks for the information.

You're both running the CadQuery script in a way that I never anticipated. I didn't know that the scripts would work properly from within the macro editor. It's interesting that it worked in the tagged v1.2.0 commit, but not in the current commit on master.

@easyw Are you running it from the macro editor as part of the KiCAD model generation framework your team created?

KammutierSpule, 2018-10-30:
Actually I am trying to follow something similar how KiCad model generation is doing to create shapes.
I found that running FreeCad on terminal to run a script it takes a lot to start, I will look if I can run script in a headless mode, so I hopping that cadquery still work that way.
Otherwise, I will run scripts manually on the gui.

jmwright, 2018-10-30:
@KammutierSpule Depending on what you want, you can run a CadQuery script without the `show()` call and just do an export to file directly from the script. The file types supported are listed [here](https://github.com/dcowden/cadquery/blob/master/cadquery/freecad_impl/exporters.py#L19). Not sure if that's what you're looking for or not, but that would allow the scripts to be run from the command line without the FreeCAD GUI.

easyw, 2018-10-30:
> @easyw Are you running it from the macro editor as part of the KiCAD model generation framework your team created?

yes, we launch our scripts by command line to be able to script all the process.

> It's interesting that it worked in the tagged v1.2.0 commit, but not in the current commit on master.

I'm not sure if this worked before... to let this work it is enough to create an empty FC document and then run the code...
We do that in our scripts and you probably do it in your CQ private editor, when you run a script

KammutierSpule, 2018-10-30:
> @KammutierSpule Depending on what you want, you can run a CadQuery script without the `show()` call and just do an export to file directly from the script. The file types supported are listed [...]

I would like to export it to a mesh file, giving some "quality options", maybe to proprietary format.
I have some difficult at moment to distinguish what are CadQuery, CadQuery-freecad-module and FreeCad scripting, so I am not sure about this dependencies and what can I do with which one.

easyw, 2018-10-30:
@KammutierSpule 
you can simply run the script inside FC and CQ FC module.
All the scripts can run through FC standard console... for your script to run you miss only the need to create a new doc, if not existing one in FC already, and then apply your CQ and FC python code...
You don't need to start it from command line as we do in our scripts, you can simply open your .py file in FC and then Run it through the standard 'Execute Macro' green button command.

[test.py.txt](https://github.com/jmwright/cadquery-freecad-module/files/2530052/test.py.txt)  (rename it to test.py)

jmwright, 2018-10-30:
@KammutierSpule 

- cadquery = Core library that provides the Python CAD scripting API. It doesn't provide a GUI, it's just designed to be used from the command line and/or in Python scripts.
- cadquery-freecad-module = Workbench/plugin for FreeCAD that wraps the core cadquery library to provide a CadQuery development GUI inside FreeCAD.
- FreeCAD scripting (macros) = Can be mixed with CadQuery scripting, but is a separate API. FreeCAD macros are often used for data conversion and GUI manipulation (document tabs). [Here](https://github.com/dcowden/cadquery/blob/master/examples/FreeCAD/Ex024_Using_FreeCAD_Solids_as_CQ_Objects.py) is an example of mixing FreeCAD scripting with CadQuery scripting. The CadQuery objects have to be at a lower level, so there's more code than it takes to do the same in pure CadQuery scripting.

jmwright, 2018-12-29:
In #136 I've added a work-around for showing CadQuery objects from the macro editor. Updating via the FreeCAD add-on manager should pull the new version. Be aware though that there are significant changes in this version of the workbench, the most disruptive probably being the new code editor. Please see the [changelog](https://github.com/jmwright/cadquery-freecad-module/blob/master/changes.md#v130-unreleased) for details.

closed 2018-12-29 by jmwright