CadQuery/cadquery-freecad-workbench #133: save dialog on exit

tolot27, 2018-12-16:
Currently, FreeCad (v18 build 15417) does not show a warning on exit if a CadQuery script is changed and unsafed. Instead, it closes FreeCad immediately or asks at least, if the generated 3D model should be saved.

Goal: Show a warning on exit with option to save the script.

jmwright, 2018-12-17:
Thanks for reporting this. This workbench still isn't fully compatible with FreeCAD 0.18, and especially not with the Python 3/Qt 5 builds. I'm hoping to have some time over the US holidays to work through the compatibility issues like this one.

easyw, 2018-12-17:
@jmwright 
thanks for having a look at CQ compatibility with FC 0.18 ..
please consider that FC 0.18 is going to be released as the new official release on this winter...
Please have alsop a look at Py3/Qt5 compatibility, which will be the next FC target :smile:
PS I tested some of our CQ 3D package scripting in FC0.18 and they works just fine 😉 (but we use a subset of the great CQ WB).

jmwright, 2018-12-29:
@tolot27 If you close the script using CadQuery->Close Script you will get the warning, but not if you click the X on the document in the MDI area. I have not found a way to make FreeCAD aware that there are changes to the document in that context.

jmwright, 2024-12-04:
#158 is an attempt to update this workbench for CQ 2.x and FreeCAD 1.0. Results look promising, but the updated workbench will rely on FreeCAD's built-in macro editor and external IDEs rather than including its own editor. I'm going to close this issue because of that.

closed 2024-12-04 by jmwright