CadQuery/cadquery-freecad-workbench #134: Soliciting Community Feedback: Do You Use the Built-In Code Editor?

jmwright, 2018-12-17:
Out of the users of this workbench, how many use the built-in code editor? I ask because I'm thinking about re-opening isue #96 . Longer description of why I'm asking below.

This workbench currently uses a code editor widget called Pyqode, which hasn't been supported by the dev for a long time. It relies quite a bit on code that breaks in FreeCAD 0.18/Python 3/ QT 5. The Pyqode codebase is pretty substantial and can be hard to fix, but to replicate the exact same functionality quickly is probably out of the question with the amount of time I have. The reason I have never just switched to implementing a built-in editor is that things like code completion would get dropped (which has not always worked in the past anyway), at least for the forseeable future. If most users rely on external editors, this might not be a big problem. But if everyone uses the built-in editor, this might be a very disruptive change.

The decision may be forced anyway to be able to upgrade to FreeCAD 0.18 compatibility, but I'd be interested in hearing from the community.

easyw, 2018-12-17:
Hi @jmwright 
personally I use an external editor.
May be @poeschlr or @Misca1234 or @Shackmeister could give you an other feed back

Misca1234, 2018-12-17:
I have never used the built in editor, I do not even use the cadquery editor in FreeCad, only external.

PS
If you are looking for a new hobby project, I would suggest a spice simulator ;)
DS

poeschlr, 2018-12-17:
I nearly always used the internal editor. (at least at some stages of the development of my scripts.)

I however do not really need many features of it. (At least none that an external editor could not supply)
As long as there is some way to quickly build a model from a file i am fully ok with using an external editor. Ideally there would also be some botton to update from a changed source file or even better a way to automatically update the model if the source file changes.

jmwright, 2018-12-17:
@poeschlr There's an `execute_on_save` [setting in Settings.py](https://github.com/jmwright/cadquery-freecad-module/blob/master/docs/usage.md#settings) that will save the script automatically every time you save file in the external editor. 

I had done a simple code editor in Qt awhile back for another project. I've started [a branch](https://github.com/jmwright/cadquery-freecad-module/tree/kill-pyqode) where I've swapped out PyQode for that custom editor. It lacks a lot of features, but I think it's usable. Beta feedback would be appreciated. Keep in mind though that you're only evaluating the built-in editor. The external editor functionality is broken in that branch because I'm looking to separate it out. I think that if you use an external editor, it would be nice if the workbench ignored the built-in editor and just displayed the 3D model. Currently the file watcher is tied in with PyQode, so you have to have the text open in PyQode to get `execute_on_save` functionality.

tolot27, 2018-12-18:
I've started using the internal editor because I could not find an external one which shows the created shapes without any error.
In general, I use Visual Studio Code for development but did not try the `execute_on_save` setting yet.

tolot27, 2018-12-18:
What I would really appreciate is if the 3D view would be part of/attached to/side-by-side to the internal editor window. That would allow switching between different tabs of the editor much easier. I don't like the window tiling, if I have more than one cadquery script open.

jmwright, 2018-12-19:
@tolot27 That's an interesting idea. I'm not sure there's a Python binding for the 3D view in FreeCAD, but I'll look into it. I can add it as a setting if it's a possibility. I would probably use that side-by-side view a lot.

jmwright, 2018-12-19:
Posting this related link here for my own information.

https://www.freecadweb.org/wiki/Embedding_FreeCADGui

jmwright, 2018-12-24:
Thanks for the feedback everyone. I've taken your comments into account as I'm working towards FreeCAD 0.18 compatibility. Feel free to continue commenting, but I'm going to close this for now.

closed 2018-12-24 by jmwright